OpenEtherCATsociety/SOEM #789: No socket connection form issue

nehapagare17, 2024-03-06:
[image]
[image]
I am getting this issue while running slaveinfo please help me with this.what is going wrong in this.I am using linux

nehapagare17, 2024-03-07:
[image]
after not using bracket i am getting this output that no slave is connected though i have connected another linux pc as slave using RJ45 .what i need to do in that case?

ArthurKetels, 2024-03-07:
It is very unlikely your "other" linux PC is an EtherCAT slave. So this SOEM behaviour is normal. Connect one or more proper EtherCAT and it will work. I suggest to read first about how EtherCAT actually works.
www.ethercat.org

nehapagare17, 2024-03-07:
that means i cannot make PC as slave? and i will have to connect some EtherCAT slave like LAN-9252? if yes then i will have to connect it to that PC where i have configured SOES or to same PC where I have configured master?

nehapagare17, 2024-03-07:
> External Email
>
> Yes, you need some special EtherCAT hardware chip like the LAN-9252 (many other options) to build a slave. Google "how to develop ethercat slave". Only the master is a software only solution and runs on many generic platforms.

I am bit confused about SOES what is it? is it not EtherCAT software?

nehapagare17, 2024-03-07:
I mean EtherCAT slave software?