OpenEtherCATsociety/SOEM #791: SOEM on visual studio code (windows)

Bipin9179, 2024-03-08:
Hello Everyone,
please tell me the step how to install SOEM and create master using visual studio code on windows as I m using SOEM first time and while installing SOEM I m getting errors so please guide me step wise .

ArthurKetels, 2024-03-08:
The README.md is very clear. If you follow the instructions it works. Setting up the prerequisites is most likely where you fail. But setting up the compiler and cmake is outside the scope of the SOEM project.

I assume you meant VisualStudio instead of Visual Studio Code? The first is a C, C++, C# compiler with IDE, the second only a glorified editor.

Bipin9179, 2024-03-08:
> The README.md is very clear. If you follow the instructions it works. Setting up the prerequisites is most likely where you fail. But setting up the compiler and cmake is outside the scope of the [...]

No I am using visual studio code different one

Bipin9179, 2024-03-08:
PS C:\Users\Anzo\Downloads\SOEM\SOEM-master\build> make
Scanning dependencies of target soem
[  4%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.obj
gcc.exe: error: /D: No such file or directory
gcc.exe: error: _CRT_SECURE_NO_WARNINGS: No such file or directory
gcc.exe: error: /WX: No such file or directory
make[2]: *** [CMakeFiles/soem.dir/soem/ethercatbase.c.obj] Error 1
make[1]: *** [CMakeFiles/soem.dir/all] Error 2
make: *** [all] Error 2

This is error

ArthurKetels, 2024-03-08:
You can not use make in windows for this project. Read the instructions:

> Start a Visual Studio command prompt then:
> 
>     mkdir build
>     cd build
>     cmake .. -G "NMake Makefiles"
>     nmake
> 

Of course you first have to install Visual Studio and CMake.

Bipin9179, 2024-03-08:
So it means I have to use Microsoft visual studio instead of visual studio code ?