OpenEtherCATsociety/SOEM #792: DC start time diff

Edphoenixmax, 2024-03-14:
I wish all you good. I have a question about ecx_dcsync functions. As we can see ,at the first of this function ,it read dc systemtime to calculate the start time. It add system time and cycletime and so on ,but evert time I use this function it will read the slave's own system time , it has a little difference between first slave and the last slave , so that the starttime will be  a little bit different. Is that acceptable or I use this function wrong?

Edphoenixmax, 2024-03-14:
Thanks for your answer!

Edphoenixmax, 2024-03-14:
Another question, if I use this function to set DC start time,if I set cycletime as 1.0,if the slave1 systemtime is 4.0,while the slave2 systemtime is 4.1,it will start at 5.0 and 5.1 . As time goes on,systemtime of slave2 will get close to slave1,if both slaves' systemtime  is 8.0, the sync signal will both trigger at 9.0 or slave2 will still trigger at 9.1?

ArthurKetels, 2024-03-17:
When you read local time of a slave it can be either synchronized or not. When during start-up of SOEM ecx_configdc() is used it set-up the cyclic PDO transfer to also distribute the clock among all slaves. When this had time to stabilize all local clocks of all slaves are running at the same speed and have the same absolute time (give or take 200ns). So a slave does no longer need to know the time of other slaves to do things synchronously. It knows it's local time is exactly the same as any other slave.

So when you read the local clock of any slave there should be very little difference other than your time of packet transfer and propagation delay.  

The ecx_dcsync0() function set-up a slave to fire pulses starting at an absolute time at a specific rate. If you give the same parameters to two different slaves they will fire at exactly the same times. You can test this with an oscilloscope attached to the sync output of those two slaves. You will see the pulses match perfectly.

From the comment in the ecx_sync0() function:
```
/* Calculate first trigger time, always a whole multiple of CyclTime rounded up
   plus the shifttime (can be negative)
   This insures best synchronization between slaves, slaves with the same CyclTime
   will sync at the same moment (you can use CyclShift to shift the sync) */
```
Remember that the first sync pulse will happen at a defined absolute time. So make sure this time is in the future and it is rounded to the cycle time. Then it does not matter  much when you invoke the function for each slave. Because you always have a time difference between calling the sync function for the first and the last slave.

For very large slave count configurations it makes sense to tweak the function a bit to wait a second or so before the first pulse. Then there is enough time to prepare all slaves.

Edphoenixmax, 2024-03-18:
I may get your point. So in the ecx_configdc function , it writes the delaytime value in the register 0x0928, so the slave will use this value automatically,right? So the time of packet transfer and propagation delay can be ignored
Thanks for your reply!

closed 2024-03-26 by Edphoenixmax