OpenEtherCATsociety/SOEM #796: PDO Adresse do not match between output of slaveinfo -map and in ethercat-sdk

LiuyanHuang12, 2024-03-22:
Hi,

i am wondering why my PDO Adresse of the 2nd Slave(BK1120) do not match between the output of slaveinfo -map and in the Software ethercat-sdk. 

Could anyone please tell me why or guide me how to debug it? 

wenn the end Position of Slave1 (AKD) 17 is, then plus 8 bit, we should get the beginning Position of Slave2 as 18, right? But why in Ethercat-sdk it is 20?

if anyother data is needed, please tell and i will post it.
many many thanks!
[image]

ArthurKetels, 2024-03-22:
I am not sure what your question is? SOEM seems to generate a reasonable mapping. Does it work?
What is the software ethercat-sdk? If that piece of software gives the wrong answer ask them for a solution.

You have to remember there are many ways of mapping the slave FMMU's to the logical address space. And one is not better or worse than the other. It also depends on your optimization goal. If you look in the source code of SOEM you will find the algorithm is very simple. It is also not mandatory to use the slave mapping functions of SOEM, you can use your own. As long as you fill in the the result in the SOEM slave table it all works just fine.

LiuyanHuang12, 2024-03-26:
Thank you for the really fast reply. 
ethercat-sdk is a software from the RT-Labs, we can also use it to control the slave.

ArthurKetels, 2024-03-26:
OK, but I still do not know what your question is. Two different software solutions, two different slave mappings. I do not see any problem. If your questions is answered, please close the issue. If not, rephrase the question.

LiuyanHuang12, 2024-03-27:
Thank you, i will close the issur.

closed 2024-03-27 by LiuyanHuang12

nakarlsson, 2024-03-29:
@LiuyanHuang12 , 
EtherCAT SDK is a GUI ontop of SOEM. Default EtherCAT SDK use SOEM overlapping IO config to support TI ESCs out of the box.

So, as Artur pointed out, just a different way of setting up things.

LiuyanHuang12, 2024-04-03:
Thank you, I have read more and have a better understanding now. :D