OpenEtherCATsociety/SOEM #799: SDORead fails after transitioning from OPERATIONAL to SAFE_OP

topgun128, 2024-04-10:
Good day to you sirs
I am trying to control a motor with the synapticon circulo and the SOEM library.
I can move from INIT to PRE_OP, define my own PDO map, then continue to SAFE_OP and OPERATIONAL.
I can read values using SDOread in PRE_OP, SAFE_OP and OPERATIONAL.

However after I move from OPERATIONAL back to SAFE_OP, the SDOread function always fails.
It returns 0 and does not change the value of the passed pointer.

To transition back I use:
```
context->slavelist[0].state = EC_STATE_SAFE_OP;
ecx_writestate(context, 0);
ecx_statecheck(context, 0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE);
```

Only after switching back to INIT and reinitializing everything, the SDOs are readable again.
I have tested my code with the Elmo Twitter, which does not experience the issue.
Synapticon support sais that they do not experience the issue with the IgH Master master and dont have any experience with the SOEM library.

Any idea if there is something I am missing? Is there something I need to do after switching to SAFE_OP before using SDOread?

Thank you in advance,
Lucas

ArthurKetels, 2024-04-10:
If you can post a wireshark .pcap that shows the issue, that would help.

topgun128, 2024-04-11:
Thank you for the quick answer. I will have access to the hardware on Monday. I will pot the wireshark data then.

topgun128, 2024-04-15:
I got the .pcap. Here is what was captured:
[sdo_issue.zip](https://github.com/OpenEtherCATsociety/SOEM/files/14975519/sdo_issue.zip)

- I switch from SAFE_OP to OPERATIONAL
- I read position sdo (0x6064) twice. It works.
- I switch from OPERATIONAL to SAFE_OP
- I attempt to read position sdo (0x6064) twice. It fails.

To my untrained eyes i looks as if the controller just does not give any response.
Do you see some kind of error?
I use a 700ms timeout (EC_TIMEOUTRXM). Why does it not time out, but return immediately?

ArthurKetels, 2024-04-15:
The problem is that in SAFE_OP soem tries to send a mailbox to the slave, but it can not write to the mailbox buffer (work counter is zero on return of the packet). Then soem tries again with the same result. The probable reason is that the slave has deactivated the mailbox-in SyncManager. 

To check this you could add the code below, just before the SDO read function. It will show up in the wireshark capture and read the complete SM0.

```
uint8_t buf[8];
ec_FPRD(0x1001, ECT_REG_SM0, sizeof(buf), &buf, EC_TIMEOUTRET);
```

topgun128, 2024-04-15:
Thank you for your input.
I have added your line to my code. Note that I used ecx_FPRD as I work with a context.
Here is the result. The SyncManager answer indeed looks different.
I have also added another SDO read before switching from SAFE_OP to OPERATIONAL, where SDO reading still works.
[sdo_fprd.zip](https://github.com/OpenEtherCATsociety/SOEM/files/14978075/sdo_fprd.zip)

Is there a way to manually activate the SyncManager?
Is this a bug on the slave side or a configuration issue?

ArthurKetels, 2024-04-15:
Yes, the input mailbox gets disabled by the slave after going to safe-OP.  Therefore further mailbox communication is inhibited.

After looking through the issues list here I found #500 , this is exactly the same problem. So I see Synapticon did not solve this issue for many years.

Because this issue does not happen with Twincat and IgH, it is worthwhile to figure out what is different.

Some things you could also try:
- Instead of issuing state safe-OP try to send safe-OP+ACK. When a slave goes to safe-OP+error it disables SM2 or SM3. perhaps the slave has some internal state confusion and the ACK will trigger a reset.
- Send some LRW during the state transition from OP to safe-OP.
- As a last resort re-enable the SM from the master side (conceptually this is very wrong)

topgun128, 2024-04-15:
Thank you for your help. I will contact their support.
I tried sending SAFE_OP + ACK (0x14). It did not help.
I can't figure out what you mean with send some LRW (excuse my inexperience). I would try it if it is any value to you.
Otherwise I would consider this issue answered / closed as this is clearly not a bug of yours.

closed 2024-04-15 by ArthurKetels