OpenEtherCATsociety/SOEM #800: Unable to get power supply and drives to SAFE_OP (AX8000)

lshoek, 2024-04-11:
Hi all,

I am currently working a custom application built on SOEM to commission a Beckhoff AX8000 series power supply (AX8620) and two drives (AX8108). Unfortunately I have been unable to advance these devices' states to `SAFE_OP`. 

This issue board has been incredibly helpful in providing information and tips to deal with all sorts of drives, but now I have hit a wall.

My configuration is validated and tested with TwinCAT 3 (TC3), which is able to advance to OP. To mirror the extensive configuration set, I have exported the TC3 startup lists of all devices to XML and wrote a parser that reads and transmits all entries correctly using `ecx_SDOwrite` calls. Workcounters report successful writes and `ecx_readState` calls to individual devices show no errors. The DC sync is also adopted from the TC3 settings. Yet, I never succeed in getting to `SAFE_OP`. Am I overlooking any important steps in the configuration process? It is probably worth mentioning that I can successfully configure and operate the (supposedly simpler) [Beckhoff EL72x1](https://infosys.beckhoff.com/english.php?content=../content/1033/el72x1-901x/index.html&id=) drive with my current SOEM implementation.

TC3's commissioning procedure seems very complicated judging from the capture, and I have trouble finding the cause of this problem with the available [documentation ](https://infosys.beckhoff.com/english.php?content=../content/1033/ax8xxx_object/11606773387.html&id=3362361772709106230) of the AX8000 series.

Here is a log snippet from my application. I have reached a point where I can now consistently get all SDO calls to succeed after introducing delay.
```
[info] 3 slaves found and configured
[info] EthercatMaster: all slaves reached pre-operational state
[info] EthercatMaster: mapping slaves...
[info] Slave 1: Setup AX8620-0000-0000 (8 SDO entries)
[info] Slave 1: Setup WKC: 8
[info] Slave 2: Setup AX8108-0000-0000 (133 SDO entries)
[info] Slave 2: Setup WKC: 133
[info] Slave 3: Setup AX8108-0000-0000 (133 SDO entries)
[info] Slave 3: Setup WKC: 133
[info] EthercatMaster: all slaves mapped
[warn] EthercatMaster: not all slaves reached safe-operational state

Slave 1 State=0x12 StatusCode=0x001e : Invalid input configuration
Slave 2 State=0x12 StatusCode=0x0003 : Unknown
Slave 3 State=0x12 StatusCode=0x0003 : Unknown
```

This is the log of a clean run of `slaveinfo.exe` after a power cycle:
```
slaveinfo.exe \Device\NPF_{EBDDDE56-0286-4E99-8381-A21C09C09760} -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{EBDDDE56-0286-4E99-8381-A21C09C09760} succeeded.
Time:1712763282.425 PACKET slave:1 index:1c00.03 error:1
Time:1712763282.501 PACKET slave:1 index:1c13.00 error:1
3 slaves found and configured.
Calculated workcounter 6
^C
C:\dev_2\thirdparty\soem\msvc\x86_64\bin>slaveinfo.exe \Device\NPF_{EBDDDE56-0286-4E99-8381-A21C09C09760} -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{EBDDDE56-0286-4E99-8381-A21C09C09760} succeeded.
3 slaves found and configured.
Calculated workcounter 7
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration
Slave 2 State=12 StatusCode=  30 : Invalid DC SYNC configuration
Slave 3 State=12 StatusCode=  30 : Invalid DC SYNC configuration

Slave:1
 Name:? M:00000002 I:21ac6012
 Output size: 0bits
 Input size: 16bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 21ac6012 Rev: 00690000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1100 L: 128 F:00010022 Type:2
 FMMU0 Ls:00000018 Ll:   2 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0018.0] 0x6000:0x01 0x01 BOOLEAN      Non-generative brake order_n
  [0x0018.1] 0x6000:0x02 0x01 BOOLEAN      Ready to operate
  [0x0018.2] 0x6000:0x03 0x01 BOOLEAN      TorqueOFF order_n
  [0x0018.3] 0x6000:0x04 0x01 BOOLEAN      Generative brake order_n
  [0x0018.4] 0x6000:0x05 0x01 BOOLEAN      Limit dc link current_order_n
  [0x0018.5] 0x6000:0x06 0x01 BOOLEAN      NC handling_order_n
  [0x0018.6] 0x0000:0x00 0x02
  [0x0019.0] 0x0000:0x00 0x03
  [0x0019.3] 0x6000:0x0C 0x05 BIT5         Counter

Slave:2
 Name:AX8108-0000-0104
 Output size: 96bits
 Input size: 88bits
 State: 18
 Delay: 175[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 1fac6012 Rev: 00680000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:2a00 L:  11 F:00010020 Type:4
 SM4 A:3c00 L:   0 F:00000024 Type:5
 SM5 A:4e00 L:   0 F:00000020 Type:6
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001a Ll:  11 Lsb:0 Leb:7 Ps:2a00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:5
 MBX length wr: 1024 rd: 1024 MBX protocols : 0d
 CoE details: 2f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x2C00:0x01 0x10 UNSIGNED16   Power supply controlword
  [0x0002.0] 0x6040:0x00 0x10 UNSIGNED16   Ch A Controlword
  [0x0004.0] 0x607A:0x00 0x20 INTEGER32    Ch A Target position
  [0x0008.0] 0x60FF:0x00 0x20 INTEGER32    Ch A Target velocity
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x001A.0] 0x0000:0x00 0x01
  [0x001A.1] 0x603E:0x02 0x01 BOOLEAN      Position actual value invalid
  [0x001A.2] 0x603E:0x03 0x01 BOOLEAN      1st Additional position actual value inv
  [0x001A.3] 0x0000:0x00 0x05
  [0x001B.0] 0x6041:0x00 0x10 UNSIGNED16   Ch A Statusword
  [0x001D.0] 0x6064:0x00 0x20 INTEGER32    Ch A Position actual value
  [0x0021.0] 0x60F4:0x00 0x20 INTEGER32    Ch A Following error actual value

Slave:3
 Name:AX8108-0000-0104
 Output size: 96bits
 Input size: 88bits
 State: 18
 Delay: 400[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1003
 Man: 00000002 ID: 1fac6012 Rev: 00680000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:2a00 L:  11 F:00010020 Type:4
 SM4 A:3c00 L:   0 F:00000024 Type:5
 SM5 A:4e00 L:   0 F:00000020 Type:6
 FMMU0 Ls:0000000c Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000025 Ll:  11 Lsb:0 Leb:7 Ps:2a00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:5
 MBX length wr: 1024 rd: 1024 MBX protocols : 0d
 CoE details: 2f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x000C.0] 0x2C00:0x01 0x10 UNSIGNED16   Power supply controlword
  [0x000E.0] 0x6040:0x00 0x10 UNSIGNED16   Ch A Controlword
  [0x0010.0] 0x607A:0x00 0x20 INTEGER32    Ch A Target position
  [0x0014.0] 0x60FF:0x00 0x20 INTEGER32    Ch A Target velocity
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0025.0] 0x0000:0x00 0x01
  [0x0025.1] 0x603E:0x02 0x01 BOOLEAN      Position actual value invalid
  [0x0025.2] 0x603E:0x03 0x01 BOOLEAN      1st Additional position actual value inv
  [0x0025.3] 0x0000:0x00 0x05
  [0x0026.0] 0x6041:0x00 0x10 UNSIGNED16   Ch A Statusword
  [0x0028.0] 0x6064:0x00 0x20 INTEGER32    Ch A Position actual value
  [0x002C.0] 0x60F4:0x00 0x20 INTEGER32    Ch A Following error actual value
End slaveinfo, close socket
End program
```

Here are captures of the startup procedure using TwinCAT 3 and my custom application:
[captures.zip](https://github.com/OpenEtherCATsociety/SOEM/files/14948910/captures.zip)

Has anyone been able to setup these devices successfully using SOEM? Please reach out! I can provide more information/context if needed. Thanks!

lshoek, 2024-04-16:
Problem solved! Will briefly describe how I came to the solution and hopefully help out others working with these drives.

Inspecting the TC3 Wireshark trace for important data packets that were absent in my own startup procedure (during the transition from PREOP to SAFEOP) turned up the missing pieces. Unlike the previous drives we worked with, it seems that it was required to overwrite the configuration for `FMMU` and `SyncManager`, as also stated in the [EtherCAT device protocol](https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf).

Filtering for `FPWR` packets (`ecat_mailbox.coe.type == 2` ) and the specified power supply `ecat.sub1.adp == x` made it easier to comb through the DC and mapping configuration. These entries were missing from my startup:
```
SyncManager (0x0810 = SM2): 0012000004000000
SyncManager (0x0818 = SM3): 0019020020000100
FMMU (0x0600 = FMMU0): 00080001020000070019000101000000
```

Which correspond to the appropriate members in the SOEM information struct:
```
ec_slavet& s = context->slavelist[index];

std::vector<uint8> data = { 0x00, 0x12, 0x00, 0x00, 0x04, 0x00, 0x00, 0x00 };
std::memcpy(&s.SM[2], data.data(), data.size());

data = { 0x00, 0x19, 0x02, 0x00, 0x20, 0x00, 0x01, 0x00 };
std::memcpy(&s.SM[3], data.data(), data.size());
```
Or as such:
```
s.FMMU[0].LogStart 	= 0x00080001;
s.FMMU[0].LogLength 	= 0x0200;
s.FMMU[0].LogStartbit 	= 0x00;
s.FMMU[0].LogEndbit 	= 0x07;
s.FMMU[0].PhysStart 	= 0x0019;
s.FMMU[0].PhysStartBit 	= 0x00;
s.FMMU[0].FMMUtype 	= 0x01;
s.FMMU[0].FMMUactive = 0x01;
```
Setting these after the SDO startup configuration (in the `PO2SOconfigx`) got the devices into `SAFEOP`!

I suppose this was a far too involved way to find this solution. I'm not an EtherCAT dev first so I'm still rather fuzzy on some of the concepts. Any comments on proper commissioning with this drive are still welcome. Anyway thank you very much for this great library!

closed 2024-04-16 by lshoek

ArthurKetels, 2024-04-16:
Thanks for the feedback!  I notice that complex slaves from Beckhoff are very complicated to get going without special knowledge. Standard EtherCAT behaviour certainly does not cut it any longer. Sorry I was not able to help you dissect the issue, too much other work.

Guistac, 2025-12-05:
Hello @lshoek
I'm currently working on a master that controls these types of drive as well, and indeed they are super tricky.

I managed to get my all devices to OP, I also set up this confusing PowerSupplyStatusWord-to-Drive communication by having the master mirror process data from one to the other. I parse and upload twincat startup lists, I did manual SM watchdog overrides, managed to clear STO through FSOE.
But still I cannot get the drives to enable their power stage, I'm always faced with communication errors between the power supply and drive.

Did you manage to make use of your drives? 
I hope this reaches you!