OpenEtherCATsociety/SOEM #804: DC config issues

esgasaria, 2024-05-06:
Hello!

I'm currently working with a system that uses EtherCAT using these Beckhoff modules:

- EK1100
- EL3104
- EL4132

I'm unable to run TwinCAT because I don't have a physical PLC master, so I'm using my PC (linux) as master and SOEM to code them.
I ran the slaveinfo code and was able to get an overview of my modules:

From there I was able to see that the AO module has the DC set to 0.
[SlaveInfo_terminal.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/15219621/SlaveInfo_terminal.pdf)

I think this is why I'm not able to control the output module.

With the AI module I have no issues whatsoever, and I'm able to read the values using the same library in c++ as shown in the simpletest.zip attachment. In this code I'm just trying to control the output sent b the AO module based on the potentiometer reading from the AI module.
The AI readings are working great as can be seen in the AI_readings.pdf file.
[AI_readings.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/15219638/AI_readings.pdf)

I apologise in advance if the code is still a bit messy, I've tried different things to understand how to access the AO module, and maybe try to set a DC for it with this low level method.

So, after this explanation, my question is how can I set a DC for my analog output module so I can control my outputs? I tried to SDO write them but I couldn't manage to get it running (you'll see some of my attempts commented out).
[simpletest.zip](https://github.com/OpenEtherCATsociety/SOEM/files/15219633/simpletest.zip)
[SlaveInfo_terminal.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/15219636/SlaveInfo_terminal.pdf)
[AI_readings.pdf](https://github.com/OpenEtherCATsociety/SOEM/files/15219640/AI_readings.pdf)

ArthurKetels, 2024-05-06:
Your EL4132 is an extremely old type. Perhaps even a beta type. Still with a FPGA instead of the ASIC. These early models did not have any distributed clocks at all. It also is missing quite a bit of other ESC functionality.

Nevertheless it still should be able to output analog values (if there are no bugs in this slave). The logical mapping shows the FMMU expects the output values at address 0 and 2 of IOmap. So first thing is to check if you actually copy your values to the right spot. 

I agree you code is not very nice to look at, But it should function for this very simple case. Unfortunately you commented out all code that was intended to set the analog output values. So I can not check if this was done correctly.

I also recommend to read the documentation going with SOEM. The examples are very clear. It is recommended to use packed structures to map your slaves to the IOmap.

esgasaria, 2024-05-08:
Hello!
I was able to fix it, thank you very much!

ArthurKetels, 2024-05-08:
It would have been nice for other readers if you had posted the solution.

esgasaria, 2024-08-19:
Yes, the lack of DC in the Analog Output was creating issues, so I took it away from the system and it already improved a lot. The endiannes was also reversed, I fixed it together with packed structures like you suggested and then it worked :). This was a while back but this is what I remember from the solution

closed 2024-08-19 by ArthurKetels