OpenEtherCATsociety/SOEM #806: cannot changeinto preop state

Edphoenixmax, 2024-05-09:
I want toknow why SOEM use  ecx_FPWRw(context->port,configadr,ECT_REG_ALCTL,htoes(EC_STATE_PRE_OP | EC_STATE_ACK), EC_TIMEOUTRET3) this function to change slaves state into preop state. What is the meaning of write this htoes(EC_STATE_PRE_OP | EC_STATE_ACK) into ECT_REG_ALCTL register.
Thanks for your answer!

ArthurKetels, 2024-05-09:
This is standard EtherCAT slave state machine handing. The master writes the state it wants the slave to go to into the AL control register. Then the slaves handles this request, transitions to the state and updates the state AL status register. If for a reason the slave can not comply it will not transition the state and returns an error (set the error register and the EC_STATE_ERROR bit in the AL status register. The same happens when the slave is in OP mode and some critical error occurred (f.e. watchdog timeout). The slave goes back to safe-OP and sets the error bit.

If the EC_STATE_ERROR bit is set then the only way for the master to clear this bit is to set the EC_STATE_ACK bit in the ECT_REG_ALCTL register. Even when there is no error bit set it does not hurt or change anything. That is why SOEM always sets the ACK bit in the safe-OP transition.

Edphoenixmax, 2024-05-10:
> This is standard EtherCAT slave state machine handing. The master writes the state it wants the slave to go to into the AL control register. Then the slaves handles this request, transitions to the [...]
Thanks for your answer, I would appreciate it very much! I want to know why slaves cannot change into operational state, is there any way to find out the answer?

ArthurKetels, 2024-05-10:
Do a ecx_readstate() and then retrieve context->slavelist[slave].ALstatuscode. It will tell you the reason the slave is in error state. You have to look up the status code  with ec_ALstatuscode2string() or the slave manual if it is a vendor specific error.

Now solving the error is a whole different matter.

PhamTuTai, 2024-06-05:
> Do a ecx_readstate() and then retrieve context->slavelist[slave].ALstatuscode. It will tell you the reason the slave is in error state. You have to look up the status code with [...]
I am a new member of soem.
The problem I'm having is similar to the above. However after I call the function ecx_config_map_group() and execute "do input mapping of slave and program FMMUs" the soem state cannot change to SAFE_OP state.
I read the error code ALstatuscode as 0x28: "Synchronization not supported".
Even though I tried changing SAFE_OP to INIT the servo allowed it and didn't report any error..
I want to ask what are the conditions for the servo to change the SAFE_OP state?
I am using Servo Omron R88D-KN01H-ECT.
Thank.

PhamTuTai, 2024-06-05:
I am a new member of soem.
The problem I'm having is similar to the above. However after I call the function ecx_config_map_group() and execute "do input mapping of slave and program FMMUs" the soem state cannot change to SAFE_OP state.
I read the error code ALstatuscode as 0x28: "Synchronization not supported".
Even though I tried changing SAFE_OP to INIT the servo allowed it and didn't report any error..
I want to ask what are the conditions for the servo to change the SAFE_OP state?
I am using Servo Omron R88D-KN01H-ECT.
Thank.

ArthurKetels, 2024-06-05:
You have to know from the slave manual what synchronization modes are supported. If freerun is not supported you must set-up  a syn0/1 clock pulse and sync the master with it. There are many posts her on how to do that.

Further it is not nice to hijack a post from another user. Open you own question. You can not be sure if your problem is the same as the original poster.