OpenEtherCATsociety/SOEM #814: DC Mode  Bus shift  and Master shift

toshisanro, 2024-05-28:
Hi , the  SOEM  DC  mode only support Master shift mode.  I use FPWR  to  set the first DC slave  DCSYSTIME every cycle .  The next DC time  is equal to   current  DC time  add master avg cycle time  .   Is it possible to do it this way? 
[image]

toshisanro, 2024-05-28:
Network error causes duplicate submissions. please delete it.

ArthurKetels, 2024-05-28:
The proper way is to send a BWR to ECT_REG_DCSYSTIME instead of the FRMW as in below example.
```
                  DCO = ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FRMW, idx, FALSE,
                                           context->slavelist[context->grouplist[group].DCnext].configadr,
                                           ECT_REG_DCSYSTIME, sizeof(int64), context->DCtime);
```
When as slave receives a write instruction to REG_DCSYSTIME it compares it with its internal clock. When the value is larger it will speed up its internal clock, when it is lower it will decrease its internal clock. This is a PLL. The change is that instead of the first slave providing the clock now the master has to do it. So you drive context->DCtime. 

Now, the drawback. Seldom the master clock is accurate enough to work as a stable reference. You will end up with more jitter in your distributed clock system and possibly even loose slave synchronization. 

So what is your reason to use the master clock as reference?

dantrigg, 2024-05-28:
I am also trying to use the Master as the reference clock.

ArthurKetels, 2024-05-28:
>  Is it as simple as replacing EC_CMD_FRMW with EC_CMD_BWR and fill the master clock into context->DCtime?

For the slaves yes.

> Can you elaborate on why the Master clock is seldom accurate enough to work as a stable reference?

Numerous things, the most obvious ; master clock is perfect, it need to be added to a packet, this packet traverses the network stack to be send out to the wire, this adds delay, slave receives master clock with delay, slave tries to lock on delayed time.

Plus, did you actually measure your master clock stability? You might get surprising results.

> We don't have the option of synchronizing our OS operation with an external clock.

This is the standard lame excuse. And it is mostly not true.

All clocks in the world are different, and by definition not synchronized. You can only track your clock deviations to another clock. You might try to "correct" your clock during this tracking, but is is generally a bad idea. All correction no mater how small will cause discontinuities. It is much simpler to just properly track the behavior of different clock sources and use offsets when communicating your clock source to others.

You might want to read up on how GNSS works.

dantrigg, 2024-05-28:
Thanks for your reply.

We don't have a network stack, the packet is written directly to the NIC FIFO.

I'm not sure what you mean about measuring stability. The counter/clock I am trying to use to manufacture the DC has a resolution of about 5.5nsec.  I know there will be jitter in the clock and it may not be the absolute best way of doing things. Assuming I can make it accurate enough for a maximum of 8 slaves to sync to, I have to try.

toshisanro, 2024-05-29:
> The proper way is to send a BWR to ECT_REG_DCSYSTIME instead of the FRMW as in below example.
>
> ```
>                   DCO = ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FRMW, [...]
Thanks  for your  reply.
I tested the servo driver in DC  master shift mode , the control and communication tasks in different threads but same cycle . When the servo motor speed is fast, every few seconds it  does not run smoothly,  I think it's caused by the master clock drift.

ArthurKetels, 2024-05-29:
> Thanks for your reply.
> 
> We don't have a network stack, the packet is written directly to the NIC FIFO.

This changes the equation. With direct control over the NIC hardware much better performance is possible.

> 
> I'm not sure what you mean about measuring stability. The counter/clock I am trying to use to manufacture the DC has a resolution of about 5.5nsec. I know there will be jitter in the clock and it may not be the absolute best way of doing things. Assuming I can make it accurate enough for a maximum of 8 slaves to sync to, I have to try.
> 
> It may be considered a lame excuse. I would rather not get into the reasons why. If I had the option to rewrite and re-architect everything in order to add EtherCAT, we would be having a different discussion.

;-)

> 
> In your experience, when do slaves begin to have trouble synchronizing?
> 
> I only need the slaves to be synchronized within about 100usec. I am running a 1 to 2 msec cycle time.

The standard Beckhoff IP in ESCs handles clock difference rates up to 850ppm. This is an average over multiple seconds. Individual clock updates may jitter much more. 

On bare metal implementations where the master needed to be the reference clock I manged to sync well within 1us with more than 20 slaves. On FPGA master implementations (yes, also with SOEM) with stratum 3 clock and modified slave hardware I managed to sync within 500ps. The system was so sensitive that even bending or squeezing the network cables caused a measurable sync shift.

dantrigg, 2024-05-29:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> EXTERNAL SENDER: This message was sent from an external account. Use caution when clicking on links.

You have got me rethinking my plan of trying to supply the reference clock from the master.

I will try to explain some of the considerations. Our setpoint calculations are running on a 80386 processor using a proprietary RTOS. I have to send and receive process data to and from an ARM MK66 processor where the position loop, EtherCat and Ethernet Port reside. The loop runs when the process data is sent to the ARM in a 1 msec interrupt that runs the OS. I have measured the timing of the process data on the ARM and it is received at 1ms with jitter of +-30usec. I was planning on trying to filter out the jitter, and use a calculated offset added to the ARM clock to manufacture a Reference clock.

After thinking through what you said, I think it may be less complicated, and easier to get support, if I use the SOEM DC code out of the box and try to calculate an adjustment to my 1msec timer on the 80386 processor to stay in relative sync with the DC reference clock. I am looking into how much resolution I have in the 80386 timer.  If I have trouble regulating it for some reason I think I can go to a 2 msec cycle time, which would provide some extra jitter tolerance.

Thank you,

Dan Trigg
Product Development Manager, Machine Controls

AMS Controls Inc.
T: +1 314.344.3144 ext: 119
           800.334.5213
F:+1 314.344.9996
***@***.******@***.***>
www.amscontrols.com<http://www.amscontrols.com/>

Our Mission:
AMS Controls provides world-class engineered solutions - including production management software and integrated machine controls - that optimize the operations of manufacturers in the metal forming industry and provide a fair return on investment.

ArthurKetels, 2024-05-30:
Hi Dan,

With your suggestion you are doing exactly the same as in the SOEM sample software red_test. There it is very simplistic and must be upgraded to do real world task, but the idea is there. 

When running a clock model I mostly use a 2nd order approximation. The model observes offset (0th order), drift (1st order) and change of drift (2nd order).  This allows to make very fine adjustments in timers and prevents sudden discontinuities. Then integrate the model back to timer offset values that can be used in the next timer cycle. Then even when the timer granularity is large a smooth adjustment is possible.