OpenEtherCATsociety/SOEM #816: Narrowing conversion / integer overlfow in slaveinfo.c ebus current calculation from SII

kj4tmp, 2024-05-30:
With an EK1100, the following lines in slaveinfo.c have the following effect:

```c
ec_slave[cnt].Ebuscurrent = ec_siigetbyte(cnt, ssigen + 0x0e); // assigns 48 to int16 ebuscurrent
ec_slave[cnt].Ebuscurrent += ec_siigetbyte(cnt, ssigen + 0x0f) << 8; // adds 63488 to int16 ebuscurrent, overflow!

```

Is this ok?

ArthurKetels, 2024-05-30:
The SII specification states the Ebuscurrent as int16 type. It can be either positive or negative. C integer math allows this assignment of bytes. But perhaps |= would be easier to understand than the current +=. To put it simply, C does not care about numbers, just bits.

kj4tmp, 2024-06-08:
Thank you for the explanation, when re-writing this example for python, I was unaware of the differences between C and python with this operator. Closing.

closed 2024-06-08 by kj4tmp