OpenEtherCATsociety/SOEM #817: Multiple device reading errors using SDO

Tomisacats, 2024-05-30:
First, the device name is garbled
Second, when I connect 20 devices, the seventh and tenth devices can't read data using SDO, but separate connections can
The above issues only occur in versions above V1.3.3-beta.1
I hope the author can fix it, thank you very much

ArthurKetels, 2024-05-30:
Making a wireshark capture and posting it here would allow to investigate the cause. As it stands now there is not much information to work with.

Tomisacats, 2024-06-04:
This is a screenshot. I hope it helps to understand the problem. Thank you
[image]

ArthurKetels, 2024-06-04:
Thanks for the screenshot, however this is not very helpful. I need a full .pcap file from init to OP (or as far as you get). The capture files compress very well so even large ones can be posted here. I need to see all interaction from SOEM with the slaves as well as any timing of the packets.

Tomisacats, 2024-06-05:
This is the captured communication package please check, thank you
[test.zip](https://github.com/user-attachments/files/15580294/test.zip)

Tomisacats, 2024-06-27:
Sorry, we cannot provide the communication data related to initialization. Thank you for your answer

closed 2024-06-29 by ArthurKetels