OpenEtherCATsociety/SOEM #820: PDO and SDO Testing - Motor pulses only once

keithiscool, 2024-06-05:
Hello,
I have used the simpletest(ec_write/read SDO usage) and ebox(IOMap/PDO usage) tests - the documentation is clear on the SOEM front and I think I might be having an issues with a drive, but not sure how to troubleshoot. I cannot get the drive into operation, but SOEM is in Operation.

I have been writing to a drive with this approximate sequence and I am wondering if there are any troubleshooting steps to pursue.
I am not sure if I can post the code here, but would like to see if there are things to investigate. I am not sure if I can post a Wireshark capture either due to company policy.

The motor does pulse (one click) after sending commands to the control register.
**_When reading feedback values from the drive, the drive does not enter operation at any point._**

I confirmed SOEM is in operation (state == 8) and use that portion of simpletest after the IOMap() function call

I have written to the drive using PDO (through IOMap and structs) and also ec_write() with the same results. I have tried using the manufacturer's procedures using the different modes without success.

I can read and write to all registers, along with getting values from the PDO (velocity, position, control word feedback).

The power supply voltage does not drop.
The drive boots up in a fault state, but clears when running the procedure.
Thank you for any time/assistance.

keithiscool, 2024-06-05:
I have a locked rotor now when alternating the "go SP bit" in a while loop.
Note the motor seems to be in operation state now
I basically have this:
Any ideas why a motor would ignore position/velocity commands?

setCntrlRegFunction(int cntlRegValue){
	struct->cntlReg = cntlRegValue;
	printf("debug msg");
	ec_send_processdata();
}

bias = 1001;
while(1){
	setCntrlRegFunction(cntl reg masked "do not engage SP")
	set position + bias;
	setCntrlRegFunction(cntl reg masked "engage SP")
	bias += 10;
	ec_receive_processdata(EC_TIMEOUTRET);
	printf("debug msg");
	osal_usleep(10);
}

closed 2024-06-05 by keithiscool

ArthurKetels, 2024-06-05:
If your problem is solved please notify other users how you did it. It is very frustrating when someone posts a question and closes it without any explanation.

keithiscool, 2024-07-08:
I apologize for the confusion - the ticket was switched and now I am trying to get either a prototype drive (uses simpler control word 0x06, 0x07, 0x0F, 0xFF command word bytes)
I have moved to try and get an EL7041 to function using SOEM - I was able to connect using TwinCAT to the EL7041, but I cannot get the motor to rotate.
Please note the work is being continued on the other thread - again, I apologize for the confusion.
https://github.com/OpenEtherCATsociety/SOEM/issues/822