OpenEtherCATsociety/SOEM #821: ec_slave.outputs is changed in ec_receive_processdata

alexanderivanov1112, 2024-06-07:
Hi everyone! 

I'm working with Delta servos, and globally it works fine. But sometimes there is strange behaviour - drive falls from ServoOn state without any reason. It just changes StatusWord to previous state.

While debugging this issue I found that reason is that outputs (ec_slave.outputs) are changed in ec_receive_processdata. I added simple code that compares this array before and after calling of this function. And some values (ControlWord in my case) become changed.

If I understand it right, this array should not be changed by drive, because it's data direction is from master to slave. But actually it does. Am I doing something wrong, or is it drive bug, or maybe some SOEM bug that was already fixed? I use SOEM version dated 
approximately 2019.

nakarlsson, 2024-06-09:
No bug in SOEM I’m aware of.

Sounds like something overflows, I suggest you do some deeper debugging. There is a small window in receive process data that read data back to inputs. Outputs are not written to and are in front of inputs.

Also, verify that the .outputs and .inputs pointers don’t change? They are the Source and Destination into the IOmap.

If it happens frequently it should be easy to find if doing some instrumentation.

NevolinAlexander, 2024-06-24:
I have spent some time for debugging and found that change of first byte of ec_slave[].outputs happens here: in function ecx_receive_processdata_group, in string:

`memcpy(context->idxstack->data[pos], &(context->port->rxbuf[idx][EC_HEADERSIZE]), context->DCl);`

(can't give it's number because it was changed while modifying code for debugging purposes)

It's hard for me to undestand what it's doing, can you explain it? And maybe you have some ideas about reasons of my problem.

NevolinAlexander, 2024-06-24:
Also I found that problem appears when I change PDO structure (using SDO_write as described in servo manual) AND it's size changes (for example I exclude on of parameters). If I change something and size remains the same (for example I replace on int16 parameter with another uint16 parameter), there is no problem.

ArthurKetels, 2024-06-29:
That code line you showed 2 posts above is not part of the current code base. So perhaps update to the latest version?

NevolinAlexander, 2024-06-30:
Yes, you're right. I used version of summer 2019 (approximately). In current version this is line 1950 of ethercatmain.c.

I've tried to update code (updated whole folder 'soem'), but problem remains the same.

brentonchris, 2025-01-13:
@alexanderivanov1112 Has this problem been resolved? I also encountered the same problem on SOEM-14.0.
After the ec_deceive_processdata function, the output data in the iomap will be changed.

NevolinAlexander, 2025-01-22:
Yes it has (probably). Seems like it was servo problem. I used Delta drives made in 2020. Later I tried new Delta drives with new firmware and problem has gone.

To finally approve this statement I need to update firmware on old drives and check them again. It can be done only at servicepoint, so for now I had not time to do this. Maybe later.

NevolinAlexander, 2025-04-30:
I have updated firmware on these drives. But it did not solve problem. Actually I don't have any ideas anymore, because I have some more exact same drives with same firmware and they don't have this problem. Don't want to spent time for this anymore, let it be so :) I put these drive on hardware where this problem is not critical, and all other drives work well.