OpenEtherCATsociety/SOEM #822: Struct using ec_slave[].inputs not populating, also stepper motor not moving help

keithiscool, 2024-06-13:
Hello,
I am attempting to get a stepper motor to spin, but cannot confirm values returning from the slave using the defined PDO mapping.

I tried controlling the motor using simpletest.c, but the motor would not enter operation status --> I then resorted to use PDO instead of SDO for commands.

I used the eboxtest.c example to correctly map the struct pointing to the IOMap using ec_slave[].inputs and I cannot get values back.

I am also not clear on the procedure to get a standard motor to spin. Is this the correct procedure?
configure PDO
configure 0x1C12/3 to point to modified PDO
verify PDO
change mode to 1 or 7 for bench testing
change control word (0x6040) to switched on (0x0007
change control word (0x6040) to accept next SP (0x002F)
set desired SP (0x607A)
set control word to use new SP (0x6040 = 0x007F)

I tried to follow the procedure normally, but also tried to block code in profileOperation() to test reading using PDO. - wait for the drive to send back correct bits in the control word to verify the commands were sent.

I verified the drive and SOEM cannot get into operation using the above procedure - the only way to get voltage to the motor is if I call ec_send_processdata() in a while() loop at the printf() "InCntlrd5b:".

I am getting garbage back from the struct pointer - I am using the same syntax as the tutorial.txt - the struct pointer is created globally and I use it throughout the eboxtest() and profileOperation() functions.

Attached Output with Wireshark and code in a zipped folder:
I am new to EtherCAT - it seems to be helpful with some things.
I confirmed the motor can get voltage and current - I think it is something I am doing possibly with the control word to not keep the drive in operation mode..?

<Output>
SOEM (Simple Open EtherCAT Master)
E/BOX test
Starting  IOMap PDO test
ec_init on \Device\<ValueHidden>} succeeded.
[eboxTest.zip](https://github.com/user-attachments/files/15825875/eboxTest.zip)
[eboxTest.zip](https://github.com/user-attachments/files/15825877/eboxTest.zip)

1 slaves found and configured.
found drive at index: 1
Init IOMap
_Setup_PDO_Map
Start PDO Read
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10 UNSIGNED16
  [0x0002.0] 0x607A:0x00 0x20 INTEGER32
  [0x0006.0] 0x6081:0x00 0x20 UNSIGNED32
  [0x000A.0] 0x6060:0x00 0x08 INTEGER8
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x000B.0] 0x6041:0x00 0x10 UNSIGNED16
  [0x000D.0] 0x6064:0x00 0x20 INTEGER32
  [0x0011.0] 0x606C:0x00 0x20 INTEGER32
  [0x0015.0] 0x6061:0x00 0x08 INTEGER8
End PDO Read: RetvalRead: 2
Stepper Drive found.
Slave:1 Output size: 88bits Input size: 88bits State: 4 delay:0.1 WKC:8100167
Request operational state for all slaves
InCntlrd0: 1208, Posrd: 0, Velrd: 0, Moderd: 0
InCntlrd0a: 1231, Posrd: 0, Velrd: 0, Moderd: 0
InCntlrd1: 1231, Posrd: 0, Velrd: 0, Moderd: 0
OutCntlSP1: 002F, PosSP: 13003, VelSP: 822090585, ModeSP: 18
OutCntlSP1a: 002F, PosSP: 23003, VelSP: 7001, ModeSP: 18
OutCntlSP1b: 002F, PosSP: 13003, VelSP: 7001, ModeSP: 18
InCntlrd2: 1231, Posrd: 0, Velrd: 0, Moderd: 0
inStruct2: 1231, Posrd: 0, Velrd: 0, Moderd: 0
InCntlrd3: 1231, Posrd: 0, Velrd: 0, Moderd: 0
InCntlrd4b: 1231, Posrd: 0, Velrd: 0, Moderd: 0
InCntlrd5a: 1231, Posrd: 0, Velrd: 0, Moderd: 0
[eboxTest.zip](https://github.com/user-attachments/files/15825872/eboxTest.zip)

keithiscool, 2024-06-13:
[eboxTest.zip](https://github.com/user-attachments/files/15826079/eboxTest.zip)

keithiscool, 2024-06-13:
[Output1b.txt](https://github.com/user-attachments/files/15826096/Output1b.txt)

keithiscool, 2024-06-13:
resetting the default parameters of the drive with 0x1011 seemed to remove E500 errors, but sending 0x0007, 0x002F and 0x005F to 0x6040 does not help.
I am curious why the mode 0x6060 will not accept a value - it is an int8 and I am setting it to 1 and 7 without success when I read 0x6061.

ArthurKetels, 2024-06-15:
First, you are almost there. The slave is correctly configured. It mapped correctly, all SM and FMMU are set correctly. The slave goes to safe-OP and later to OP without problem. 

What is wrong is that you totally over-complicated the code. Do not use SDO commands to control the drive, PDO will just do fine. You are in luck the slave actually accepts SDO while in OP, many slaves do not. Also remember that send_processdata and receive_processsdata always come in pairs. 

What is an oversight is that on Windows you can not simply use PACKED on a struct. Look at the definition of PACKED in osal_defs.h for win32:
```
#ifndef PACKED
#define PACKED_BEGIN __pragma(pack(push, 1))
#define PACKED
#define PACKED_END __pragma(pack(pop))
#endif

```
PACKED doesn't do anything, you have to use PACKED_BEGIN and PACKED_END around the struct definition. The test programs in the win32 section do not seem to be updated for that.

An example from ethercattype.h:
```
/** ethernet header definition */
PACKED_BEGIN
typedef struct PACKED
{
   /** destination MAC */
   uint16  da0,da1,da2;
   /** source MAC */
   uint16  sa0,sa1,sa2;
   /** ethernet type */
   uint16  etype;
} ec_etherheadert;
PACKED_END
```

The result is that due to compiler alignment of the struct you get garbage. This is both true for the input and output structs. So off course you can not control the drive.

keithiscool, 2024-06-18:
@ArthurKetels 
I am able to get reading and writing using the #pragma fix using added WIN32 guards.
I am still unable to get the drive to move - I am commanding in the profileOperation() function.

0x06,7,2F,5F correctly and the drive will not go.
I confirmed the position command using both PDO and SDO into 0x607A (values are populated as 43003 and 1234.
The motor "locks/holds" position if I use a while() loop at the end of the profileOperation() function.

1) I ordered another motor controller - have you ever seen a motor controller that does not go into operation (0x1237 expected, but keep getting 0x1233)?

keithiscool, 2024-06-18:
[Output1c.txt](https://github.com/user-attachments/files/15887959/Output1c.txt)

keithiscool, 2024-06-18:
Added ec_dcsync to be identical to here: https://github.com/OpenEtherCATsociety/SOEM/issues/218
still 

I found the exact DC synch assign in the ESI.
After making the change, the controller is still not able to get into OP
Done during PRE_OP:
ec_config_map(&IOmap);
ec_configdc();

Done if EC_STATE_OPERATIONAL
osal_usleep(10000); //wait for OS to sync on Distributed Clock (DC)
ec_dcsync0(slave_index, TRUE, SYNC0TIME, 0); // SYNC0 on slave ON
osal_usleep(10000); //wait for OS to sync on Distributed Clock (DC)

keithiscool, 2024-06-18:
[Output1d.txt](https://github.com/user-attachments/files/15891182/Output1d.txt)
[ECatTest1D.txt](https://github.com/user-attachments/files/15891183/ECatTest1D.txt)

added more debugging - still not able to get drive to go from 0x1233 (ready) to 0x1237 (operational)

keithiscool, 2024-06-18:
[ecatTest1D.zip](https://github.com/user-attachments/files/15891412/ecatTest1D.zip)
Wireshark:

keithiscool, 2024-06-18:
[ecatTest1D_Read10CommentedWhileLoop.zip](https://github.com/user-attachments/files/15891494/ecatTest1D_Read10CommentedWhileLoop.zip)

Commented While Loop at readRegistersPDO(10) to reduce Wireshark size

keithiscool, 2024-06-21:
Similar issues:
https://github.com/OpenEtherCATsociety/SOEM/issues/574
https://github.com/OpenEtherCATsociety/SOEM/issues/442
trying to operational state (0x1237), not SAFE_OP (0x1233)
attempting to set during PRE_OP
ecx_context.manualstatechange = 1;
I assume the drive is using a different generation state machine?
legacy state machine seems to auto step using the SOEM PRE_OP, SAFE_OP, OPERATIONAL states I assume from reading these tickets.

this exact issue, but answer was not updated:
https://github.com/OpenEtherCATsociety/SOEM/issues/613

I am beginning to test a second drive on Raspberry Pi to rule out potential drive and also WIN32 issues.

keithiscool, 2024-06-24:
I tried using "ecx_context.manualstatechange = 1;" without success
Any ideas why the drive state reads 0x08 (OPERATIONAL) and the drive also receives the PDO control word 0x0F and still will not go into operation (0x1237)?
I keep getting 0x1233 and the motor only steps once.

I attached source, Wireshark and also the terminal output.

I have a feeling I am doing something wrong, but I am running out of things to try to see what is wrong.
The PDO has position, velocity, control word and also mode set.

The SM2 control register is being set to 0x5F in the source file. I am not sure why it is changing on its own to 0x60. This is a command register, the status reads back 0x1233 immediately after.

[Drive1_Output.zip](https://github.com/user-attachments/files/15959774/Drive1_Output.zip)

keithiscool, 2024-06-26:
@ArthurKetels
I have been trying to get two EtherCAT drives spinning for the past month and I think I am missing something.
I tried to log progress the best I could, but I think experience with ethercat is something I am lacking.

I am testing a EL7041 I have and I am not able to get the drive into enable (ready, ready EN stay zero no matter what I set enable 0x7010. Hopefully I can compare with the other drive if this one can spin the motor.

I looked at the first motor drive and this one as well. It seems they are cutting int and short variables to smaller chunks.
I tried using bit arrays and :4, :10 bit sizes in the struct definitions without success:

I am running this code on a raspberry pi 5 (Raspbian OS) and I am not able to set the enable bit and the velocity using the PDO. I was able to write and read the Voltage and Current maximums for the drive and set them to the correct values. I confirmed the values using the slaveinfo project.

ECatTestLinux2 - main entry (using simple_test as a base project)
I went into the simpletest() function where the InOp flag is set; it calls the ECATOperation::Drive2profileOperation() method.

**### TLDR - begin here::**
//above struct values (struct located in Drive2.h) - note the PDO element is 4 bits that are blank????
The output of slaveInfo is included to show the EEPROM for the PDO.
//I cannot find EL7041 registers at 0x000E
bool empty5[4]; //   [0x000E.7] 0x0000:0x00 0x04
//more struct values

I have tried to use the cut-down int variables in the PDO structs without success. I still am not able to set values using the PDO.
The first drive I was able to set values, but the control word was a cut-down 10bits , not a full uint16.
uint8 empty5 : 4;

Question: 
1) Do you have a working test with the EL7031/EL7041 that I can reference?
I am not sure if the 4/10bit numbers in the PDO of the other drive along with the 4,2,1 bit spacers in this drive PDO are what I am missing. I cannot get the wireshark logs from the RPi, but the Win32 tests with the first drive are above.
[RPiEL7041Project.zip](https://github.com/user-attachments/files/15995003/RPiEL7041Project.zip)

keithiscool, 2024-06-27:
Wireshark from the RPi here:
[ECatTestLinux2RPi20240626d.zip](https://github.com/user-attachments/files/16006733/ECatTestLinux2RPi20240626d.zip)

keithiscool, 2024-07-02:
Looking here for getting the handle on using TwinCAT setup and PDO mapping with EL7031/EL7041:
https://github.com/OpenEtherCATsociety/SOEM/issues/223
I think the PDO has gaps in the 1601/02/03 default PDO mapping (not sure if this is effecting control.
I will try setting the default PDO 1601/02/03 using SDOwrite() and the above method (used two NULL bytes in PDO assignment for some unknown reason).

keithiscool, 2024-07-08:
I am attempting to get the drive to spin after confirming the drive functions with TwinCAT3 - the default setup on TwinCAT uses 0x1600, 1602, 1604. I am using the exact same initialization as the WIN32 examples with the EL7031 (https://github.com/OpenEtherCATsociety/SOEM/blob/master/test/win32/simple_test/simple_test.c). I found more info here and am continuing to read more into the EK1100/EL7041 docs. I was not able to write the 0x1600 register to make them not have empty bits in the Sync Manager (SM) - https://github.com/OpenEtherCATsociety/SOEM/issues/822#issuecomment-2192562644
output is in that .txt file

I am attempting to read the 0x1C12, 0x1C13, but am getting errors with the ec_SDOread - trying to confirm the values that are used in the above WIN32 example.

keithiscool, 2024-07-08:
I verified the correct PDO mapping is completed without errors - I am in the process of comparing the Wireshark from TwinCAT3 with the Wireshark from the SOEM project I made.
I tried manual state change, changing the PDO map using presets (seems PDO mapping writes are not permitting on EL7041 as the 0x1601- 1A03 registers are RO permissions. The PDO mapping is on page 169 of the EL7041 datasheet.
I confirmed I was able to read and query the 1C12 and 1C13 registers for the mailbox setup.
Things look ok- not sure why the drive does not send back "READY" or "READY ENABLED" - they are both 0.
I keep resetting, enabling, setting velocity, and Reduce Torque without success. simply setting those bits worked in TwinCAT. TwinCAT PC is not connected to the drive or RPi for the SOEM tests.

[ECatTestLinux2.zip](https://github.com/user-attachments/files/16133926/ECatTestLinux2.zip)

keithiscool, 2024-07-09:
Working through comparing the EL7041 using TwinCAT and the expected SOEM setup, mode switch, and packed struct
I attached Wireshark captures of TwinCAT3 paired with the EL7041 drive (Should handle PDO assign using predefined PDO mappings, operation, setpoint, enable).
[TwinCAT3_EL7041_CleanBoot3.zip](https://github.com/user-attachments/files/16143765/TwinCAT3_EL7041_CleanBoot3.zip)

keithiscool, 2024-07-09:
first inspection after filtering the data - the TwinCAT3 working motor operation displays only 2 FPRD write with the mailbox setup (0x1C12/0x1C13 - 1 write to each register)
The SOEM project seems to write 65 times when filtered by mailbox - not sure why... it looks like it may be a problem with the struct elements, but I am not sure what I am doing wrong...
Source files are in the above post: https://github.com/OpenEtherCATsociety/SOEM/issues/822#issuecomment-2215099337

[image]
[image]

```c
int EL7031setup(uint16 slave){
int retval = 0;
printf("\n Configure slave EL7031 stepper motor drive \r\n");

printf("Configure PDO mapping----------------------------------\r\n");
//cannot write PDO mapping since they are read-only registers - page 191-195
// ////page 171 for PDO mappings

// //Velocity Control Compact (default)
// uint16 map_1c12[] = {0x1600, 0x1602, 0x1604};
// uint16 map_1c13[] = {0x1A00, 0x1A03};
//Velocity Control
uint16 map_1c12[] = {0x1601, 0x1602, 0x1604};
uint16 map_1c13[] = {0x1A01, 0x1A03};
// ////Position Control
// uint16 map_1c12[] = {0x1601, 0x1602, 0x1603};
// uint16 map_1c13[] = {0x1A01, 0x1A03};

// ////Positioning interface (Auto start) (errors - does not work)
// uint16 map_1c12[] = {0x1601, 0x1602, 0x1606, 0x1607};
// uint16 map_1c13[] = {0x1A01, 0x1A03, 0x1A06};
// ////Positioning interface (Auto start) with info (errors - does not work)
// uint16 map_1c12[] = {0x1601, 0x1602, 0x1606, 0x1607};
// uint16 map_1c13[] = {0x1A01, 0x1A03, 0x1A04, 0x1A06};

int MapSizeRX = sizeof(map_1c12);
int MapSizeTX = sizeof(map_1c13);
int MapNumRX = sizeof(map_1c12)/sizeof(uint16);
int MapNumTX = sizeof(map_1c13)/sizeof(uint16);

// Set PDO mapping using Complete Access
// Strange, writing CA works, reading CA doesn't
// This is a protocol error of the slave.

printf("write Zero to reset the PDO Map\r\n");
retval += ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof_uint8, &zeroByte, EC_TIMEOUTSAFE);
retval += ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof_uint8, &zeroByte, EC_TIMEOUTSAFE);
printf("write the PDO Map: MapSizeRX: %d, MapSizeTX: %d\r\n", MapSizeRX, MapSizeTX);
retval += ec_SDOwrite(slave, 0x1c12, 0x01, TRUE, MapSizeRX, &map_1c12, EC_TIMEOUTSAFE);
retval += ec_SDOwrite(slave, 0x1c13, 0x01, TRUE, MapSizeTX, &map_1c13, EC_TIMEOUTSAFE);
printf("write num of the PDO Map to confirm entries: MapNumRX: %d, MapNumTX: %d\r\n", MapNumRX, MapNumTX); 
retval += ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof_uint8, &MapNumRX, EC_TIMEOUTSAFE);
retval += ec_SDOwrite(slave, 0x1c13, 0x00, FALSE, sizeof_uint8, &MapNumTX, EC_TIMEOUTSAFE);

printf("Done Configuring PDO mapping----------------------------------\r\n");

// bug in EL7031 old firmware, CompleteAccess for reading is not supported even if the slave says it is.
ec_slave[slave].CoEdetails &= ~ECT_COEDET_SDOCA;

// set some motor parameters, just as example
uint16 u16val = 1800; // max motor current in mA
retval += ec_SDOwrite(slave, 0x8010, 0x01, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTSAFE);
// u16val = 150; // motor coil resistance in 0.01ohm
// retval += ec_SDOwrite(slave, 0x8010, 0x04, FALSE, sizeof(u16val), &u16val, EC_TIMEOUTSAFE);

/* Explicitly set flags that are (probably) invalid in EEPROM */
ec_slave[slave].SM[2].SMflags = 0x10024;

/* Explicitly disable sync managers that are activated by EEPROM */
// ec_slave[slave].SM[4].StartAddr = 0;
ec_slave[slave].SM[5].StartAddr = 0;

[... 58 lines cut ...]
uint32 counterVal		:	10;	//   [0x000C.0] 0x6000:0x11 0x20 UNSIGNED32   Counter value
uint32 latchVal			:	10;	//   [0x0010.0] 0x6000:0x12 0x20 UNSIGNED32   Latch value
// uint16 counterVal;				//   [0x000C.0] 0x6000:0x11 0x20 UNSIGNED32   Counter value
// uint16 latchVal;				//   [0x000E.0] 0x6000:0x12 0x20 UNSIGNED32   Latch value

uint16 readyEN			:	1;	//   [0x0014.0] 0x6010:0x01 0x01 BOOLEAN      Ready to enable
uint16 ready			:	1;	//   [0x0014.1] 0x6010:0x02 0x01 BOOLEAN      Ready
uint16 warning			:	1;	//   [0x0014.2] 0x6010:0x03 0x01 BOOLEAN      Warning
uint16 error			:	1;	//   [0x0014.3] 0x6010:0x04 0x01 BOOLEAN      Error
uint16 mvPostv			:	1;	//   [0x0014.4] 0x6010:0x05 0x01 BOOLEAN      Moving positive
uint16 mvNegv			:	1;	//   [0x0014.5] 0x6010:0x06 0x01 BOOLEAN      Moving negative
uint16 torqueReduced	:	1;	//   [0x0014.6] 0x6010:0x07 0x01 BOOLEAN      Torque reduced

uint16 empty4 			: 	4;	//   [0x0014.7] 0x0000:0x00 0x04

uint16 digInput1		:	1;	//   [0x0015.3] 0x6010:0x0C 0x01 BOOLEAN      Digital input 1
uint16 digInput2		:	1;	//   [0x0015.4] 0x6010:0x0D 0x01 BOOLEAN      Digital input 2
uint16 syncError2		:	1;	//   [0x0015.5] 0x1C32:0x20 0x01 BOOLEAN      Sync error

uint16 empty5			:	1;	//   [0x0015.6] 0x0000:0x00 0x01

uint16 TxPDOtoggle2		:	1;	//   [0x0015.7] 0x1803:0x09 0x01 BOOLEAN      TxPDO Toggle
//last bit might be Rx PDO toggle - mislabeled????
} in_DrivetVelocity;
```

keithiscool, 2024-07-09:
will look at timings here: https://github.com/OpenEtherCATsociety/SOEM/issues/712
There are for loops that sit on the processdata for 10k cycles (I assume to let the system gain stability and better stabilize the DC (distributed clock).
```c
//6.- Do 10,000 process data cycles in SAFE-OP state
        ec_slave[SLAVE_ID].state = EC_STATE_SAFE_OP;
        ec_writestate(0);
        ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE);

        for (int i =1; i<10000;i++)
        {
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTSTATE);
        }
```

will also try manually setting the dcsync0 again with modifying the clock tolerances
```c
//9.- Call ec_dcsync0() for the reference clock slave
 ec_dcsync01(1, TRUE, 5000000,0,0); //5millisecond poll rate
```

keithiscool, 2024-07-09:
Again, I am not getting clock errors or SM errors with the mailbox setup - I get into operation just fine. I am not getting valid PDO data - I try to send the commands without success in the ECATOperation.c file
```c
int setDrive2Velocity(int slave_index, int velSP) {
	// outPtr1->velocitySP = 3000 >> 6;
	// outPtr1->velocitySP = velSP<<4;
	// outPtr1->velocitySP = (int)1234 >> 4; - thought I heard something move
	// outPtr1->velocitySP = (int)1234 >> 3;
	// outPtr1->velocitySP = 333;
	outPtr1->velocitySP = velSP;
	printf("Set Velocity: %d\r\n", outPtr1->velocitySP);
	// outPtr1->velocitySP = (outPtr1->velocitySP)3000;

	// out_Drivet* SPvalue10bit = malloc(sizeof(out_Drivet->velocitySP));
	// SPvalue10bit = 3000;
	// outPtr1->velocitySP = SPvalue10bit;

	printf("Set Enables\r\n");
	// outPtr1->ENLatchCMD = 1;
	outPtr1->reset = 1;
	outPtr1->enable = 1;
	ec_send_processdata();
	ec_receive_processdata(EC_TIMEOUTRET);
	// osal_usleep(100);

	outPtr1->reset = 0;
	ec_send_processdata();
	ec_receive_processdata(EC_TIMEOUTRET);
	return 1;
}

int Drive2profileOperation(int slave_index) {
	printf("begin Drive2 Operation\r\n");
	//section 6.5.1 motor parameters
	//use section 6.6.4 , page 156 for internal state machine debugging
	outPtr1->reduceTorque = 1;
	outPtr1->reset = 1;
	outPtr1->velocitySP = 50;
	outPtr1->enable = 0;
	ec_send_processdata();
	ec_receive_processdata(EC_TIMEOUTRET);
	osal_usleep(200);
	printf("val123: %d, Rst: %d\r\n", outPtr1->velocitySP, outPtr1->reset);
	outPtr1->reduceTorque = 0;
	outPtr1->reset = 0;
	outPtr1->enable = 1;
	ec_send_processdata();
	ec_receive_processdata(EC_TIMEOUTRET);
	printf("val456: %d, Rst: %d\r\n", outPtr1->velocitySP, outPtr1->reset);

	wkc2 += setDrive2Velocity(slave_index, 1000);
	
	readRegistersPDO(10, slave_index);
	printf("PDO Read\r\n");

	if(inPtr1->error || inPtr1->syncError1 || inPtr1->syncError2){
		printf("Error:%d, Sync1:%d, Sync2:%d\r\n", inPtr1->error, inPtr1->syncError1, inPtr1->syncError2);
	}

	wkc2 += setDrive2Velocity(slave_index, 2000);

	// //if not ready, get the drive to readyEN
	// if(!inPtr1->readyEN){
	// 	readRegistersPDO(11, slave_index);
	// 	printf("readyEN: %d\r\n", inPtr1->readyEN);
	// 	// outPtr1->reset = 1;
	// 	outPtr1->enable = 1;
	// 	ec_send_processdata();
	// 	ec_receive_processdata(EC_TIMEOUTRET);
	// }

	readRegistersPDO(12, slave_index);

	wkc2 += setDrive2Velocity(slave_index, 3000);

	if(inPtr1->ready)
		printf("ready\r\n");

	if(inPtr1->syncError1)
		printf("syncError1\r\n");

	wkc2 += setDrive2Velocity(slave_index, 4000);

	// while (!checkDriveStatus(SPAck)) { //Blocking code to make sure desired state is reached
	// 	readRegistersPDO(7, slave_index);
	// 	osal_usleep(4000);
	// }
	
	readRegistersPDO(38, slave_index);
	// readRegistersSDO(slave_index);

	int16 intLoopCnt = 1;

	wkc2 += setDrive2Velocity(slave_index, intLoopCnt);

	// while(1) {
	// 	wkc2 += setDrive2Velocity(slave_index, intLoopCnt);
	// 	// osal_usleep(4000);

	// 	readRegistersPDO(12, slave_index);

	// 	if(intLoopCnt< 4000)
	// 		intLoopCnt += 1;
	// 	else
	// 		intLoopCnt = 1;
	// }

    return 1;
}
```

Here is the PDO read method to access the elements of the struct.
```c
void readRegistersPDO(int readNum, int slave_index) {
	ec_send_processdata();
	ec_receive_processdata(EC_TIMEOUTRET);

	// printf("PDORead%d: %04X, Pos: %d, Vel: %u, Mode: %d, wkc2: %d\r\n", readNum, inPtr1->rd_status, inPtr1->rd_position, inPtr1->rd_velocity, inPtr1->rd_mode, wkc2);
	printf("PDORead%d: Ready%d, ReadyEN%d, ENLatchC%d, Warning%d, Error%d, SetCountDone%d, CountVal:%u, TorqRed:%d, WKC:%d\r\n", 
				readNum, inPtr1->ready, inPtr1->readyEN, inPtr1->ENLatchC, inPtr1->warning, inPtr1->error, inPtr1->setCntrdone, inPtr1->counterVal, inPtr1->torqueReduced, wkc2);
	if (wkc2 == -1) {
		//ERROR HANDLER NEEDED
		printf("Error with wkc2\r\n");
		osal_usleep(100);
	}
}
```

Struct pointer declaration:
```c
	in_DrivetVelocity* inPtr1;
	out_DrivetVelocity* outPtr1;
```

PDO-IOMap pointer assignment:
NOTE: when checking the sizeof the used pointer struct, I get 8 bytes on both in and out structs
when checking the size of the struct within this method, I get 8 bytes for the RXPacked struct and 7 bytes for the TX packed struct
- note the struct declaration here is not used outside this method - just wanted to see if the packed structure was working
- it seems the packed structs around 7-8 bytes are correct in size - each variable is not a new uint16, but rather the struct is able to assemble the boolean, empty bits, and integers correctly.

```c
void setStructPointer(int slave_index){
	printf("-+-setStructPntr at index: %d\r\n", slave_index);
	#ifdef VELOCITY_PDO
		inPtr1 = (in_DrivetVelocity*)ec_slave[slave_index].inputs;
		outPtr1 = (out_DrivetVelocity*)ec_slave[slave_index].outputs;
	#elif POSITION_PDO
		inPtr1 = (in_DrivetPosition*)ec_slave[slave_index].inputs;
		outPtr1 = (out_DrivetPosition*)ec_slave[slave_index].outputs;
	#endif

	// int sizeOfMyPackedStruct = sizeof(PackedStruct);
	int sizeOfMyRXPackedStructPtr = sizeof(outPtr1);
	int sizeOfMyTXPackedStructPtr = sizeof(inPtr1);
	printf("sizeOfMyRXPackedStructPtr: %d\r\n", sizeOfMyRXPackedStructPtr);
	printf("sizeOfMyTXPackedStructPtr: %d\r\n", sizeOfMyTXPackedStructPtr);

	struct out_DrivetVelocity;
	struct in_DrivetVelocity;
	int sizeOfMyRXPackedStruct = sizeof(out_DrivetVelocity);
	int sizeOfMyTXPackedStruct = sizeof(in_DrivetVelocity);
	printf("sizeOfMyRXPackedStruct: %d\r\n", sizeOfMyRXPackedStruct);
	printf("sizeOfMyTXPackedStruct: %d\r\n", sizeOfMyTXPackedStruct);

	printf("-+-Done setStructPntr at index: %d\r\n", slave_index);
}
```

keithiscool, 2024-07-11:
Problems were the struct used decimal numbers for declaring size of variables - that was fixed
also, added for() loops as stated above that kept the PDO traffic going to avoid potential watchdog issues.
EL7041 requires resets and enable toggling to take the next setpoint; reset was used to permit further setpoints.
The attached example is for Raspberry Pi 5 with standard C - GCC compiler
The ECat2 file with the ECATOperation file were the main test that was used.
ECATOperation sends multiple velocity setpoints.
Note PDO map setup and struct pointer assignment are in Drive2::EL7031setup() and Drive2::setStructPointer()
compile using CMake and give adapter "eth0" to the program - sudo permissions
cd ./build
cmake --build .
sudo ./ECatTestLinux2App eth0

[20240711_SOEM_EL7041VelocityMode.zip](https://github.com/user-attachments/files/16179946/20240711_SOEM_EL7041VelocityMode.zip)

keithiscool, 2024-07-11:
closing - uploaded files above

closed 2024-07-11 by keithiscool

ArthurKetels, 2024-07-11:
Amazing job to get this working despite the many many conceptual mistakes. I would advise others to take this in consideration when using this code as a reference for their own projects.