OpenEtherCATsociety/SOEM #828: how to connect the SOEM to virtual network interface in Linux

monishekar518, 2024-06-24:
Hello has someone tried or know to do the following implementation on EtherCAT, please help!

SOEM master <--> virtual nic <--> physical nic <--> slave

i am able to initalize but the master is not able to read and configure the slave on the network

ArthurKetels, 2024-06-29:
It all depends on the implementation of the virtual nic. In principle it could work. But if the implementation relies on standard Ethernet traffic with tcp or udp or even MAC addresses for routing, it will not work. SOEM uses RAW Ethernet sockets.

monishekar518, 2024-06-29:
So I want a set up a network like shown below.
Libethercat master <--> virtual nic <--> physical nic <--> slave

i am able to initalize but the master is not able to read the slaves and configure the slave on the network.
virtual interface is created as below.
#sudo modprobe dummy
#sudo ip link add eth0 type dummy
#sudo ip link set eth0 up

the binding between virtual to physical nic is operated using raw sockets.
I have created 2 threads,
thread 1 > binding function with 2 sockets, source as virtual nic and destination as physical nic, then forward frames
thread 2 > binding function with 2 sockets, source as physical nic and destination as virtual nic, then forward frames

Physical connection is connected to Beckhoff slave.

Could you please tell me why the master doesn't detect slaves with above setup.

ArthurKetels, 2024-06-29:
As far as I know you can not use dummy interfaces to transmit anything. That is why they are called "dummy".

monishekar518, 2024-06-29:
But I checked using wireshark and was able to capture the frames from master on this interface. Also after transmitting frames from master with the above network configuration the slave coupler (EK1101) indicates with a led which has a higher frequency of blinking.