OpenEtherCATsociety/SOEM #830: Problem with using IOM ECT -- ec_SDOwrite failed

gzovermas, 2024-06-27:
We had use SOEM with Beckhoff IOs and servo drives sucessfully, but with IOM ECT we have problems.
We have 4 sub modules: 32DI, 32DO, 4AI, 4AO
[image]
the xml file of this module is below:
[IOM ECT-V1.2_xml.txt](https://github.com/user-attachments/files/16006875/IOM.ECT-V1.2_xml.txt)
We can config and use it with codesys
[image]
[image]
But when we change back to soem, it can't be auto config. the slave_info below
[slaveinfo_map.txt](https://github.com/user-attachments/files/16006914/slaveinfo_map.txt)
[slaveinfo_sdo.txt](https://github.com/user-attachments/files/16006915/slaveinfo_sdo.txt)
We had try to reconfig the pdo, but found that ec_SDOwrite() always return 0, the test code like below:

    u8val = 0;
    res = ec_SDOwrite(1, 0x1c13, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	printf("write 1c13.00 : %d\n", res);

We had try to trace the code in ethercatcod.c, and found that in <ecx_SDOwrite> function:
[image]
the judgement returns false, so abort.
the value of etohs(aSDOp->CANOpen)>> 12 is 0(should be 3)
the value of aSDOp->Index is 1c00(should be 1c13)

Is there any suggestion? Thx a lot

gzovermas, 2024-06-27:
The manufacture said their products doesn't support pdo assignment, and won't config pdo using sdo. How can I do then? Does SOEM can support this module?

toshisanro, 2024-06-28:
I've had the same issue before, the  slave's SM 2 and  SM3 are  empty  ,you should  config the SMs  manually.

gzovermas, 2024-06-28:
> I've had the same issue before, the slave's SM 2 and SM3 are empty ,you should config the SMs manually.

how can i config SMs? is there any example? thx a lot

ArthurKetels, 2024-06-29:
There are plenty examples here in the issues. Just use search. And as always, first read the documentation.