OpenEtherCATsociety/SOEM #834: ESTUN ED31 Servo-drive Communication Problem

rafaelkrause, 2024-07-10:
Dear all,

We are trying to establish communication between our SOEM system and a Servo-drive ESTUN ED3S (formerly known as ABD, a white-label product). 

When we attempt to communicate with the drive, we can successfully put it into OP mode; however, it immediately enters alarm E.75 (DC Sync Fault). Interestingly, when we run the Codesys controller, it runs without triggering the alarm. If we switch the cables from the Codesys controller to our SOEM controller, the drive runs normally without triggering the alarm on the Servo-drive.

We have checked the PDO configuration in both situations and found them to be identical. We have also analyzed Wireshark frames (ecat_mailbox) and confirmed that the PDO write sequence is the same in both cases.

Have any of you encountered similar issues with this or a similar drive?

We have reached out to Servo-drive sales, but they simply referred us to the manual and provided the generic response "Works with Codesys". Any help or insights would be greatly appreciated.

Note: We tested into a Realtime system and no realtime, getting same results.

Files:
ABD_ALTRA_BRASIL_DRIVE.xml: Drive configuration file;
summa_series: Drive Manual
CodesysWinSoftMotion.pcapng: Codesys WireShark (No Faults);
Soem.pcapng: Soem before codesys (Get faults);
SoemAfterCodesys.pcapng: SOEM after Codesys (Working OK);

[WireShark_ABD.zip](https://github.com/user-attachments/files/16167956/WireShark_ABD.zip)

nakarlsson, 2024-07-10:
Sounds like a DC issue, please read the following Issue comment. Eg if DC is required the timing must be precise.

https://github.com/OpenEtherCATsociety/SOEM/issues/487#issuecomment-786245585

nakarlsson, 2024-07-11:
Ok, that might be a clue than, you don’t activate DC but the drive signal error , DC Sync Fault.

DC is often ESC register writes of 0x9xx registers or 0x1C3x SDOs.

Read the manual on Drive requirements for synchronisation, FreeRun, SM-Sync or DC-sync.

ArthurKetels, 2024-07-11:
In line 375 of the Codesys startup the SYNC0 generation is started. The SOEM version does not do that. But because SOEM does not reset those registers it is possible to run the Codesys ->SOEM sequence. This is just a coincidence.

Also codesys does a burst of thousands of DC synchronizations. 

In the ESI file it states the slave can also handle freerun mode, but I guess this is not working. 

SO the fix would be to first sync master and slave and then enable SYNC0 with the proper parameters.

rafaelkrause, 2024-07-15:
Hi,

As @ArthurKetels and @nakarlsson noted, we had a DC Sync problem. More specific, we need to set the DC type, DC time, and perform time synchronization between the master and slave.

We conducted a test using Etherlab, which has an existing example, to verify your points, getting a successful comunication.

Next, we will try implementing this solution using the SOEM API. 
Thanks for your help. 

We can close this for now.

closed 2024-07-15 by ArthurKetels