OpenEtherCATsociety/SOEM #836: Writing boolean value to an output card

sigurdrb, 2024-07-15:
Hi. I got the simple_test up and running.

I also ran the ./slaveinfo program and got this printout

> SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on ens33 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:WAGO Series 750 Bus Coupler
 Output size: 48bits
 Input size: 192bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000021 ID: 07500354 Rev: 00010002
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:2400 L:  24 F:00010000 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:  24 Lsb:0 Leb:7 Ps:2400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0xF200:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag Disable
  [0x0000.1] 0xF200:0x02 0x01 BOOLEAN      Input Process Data Hold Request
  [0x0000.2] 0xF200:0x03 0x01 BOOLEAN      Output Process Data Hold Request
  [0x0000.3] 0xF200:0x04 0x01 BOOLEAN      Output Process Data Clear Request
  [0x0000.4] 0x0000:0x00 0x0C
  [0x0002.0] 0xF200:0x05 0x10 UNSIGNED16   Diagnostics Control Word
  [0x0004.0] 0x7010:0x01 0x01 BOOLEAN      Channel 1, Data 1
  [0x0004.1] 0x7010:0x02 0x01 BOOLEAN      Channel 2, Data 1
  [0x0004.2] 0x7010:0x03 0x01 BOOLEAN      Channel 3, Data 1
  [0x0004.3] 0x7010:0x04 0x01 BOOLEAN      Channel 4, Data 1
  [0x0004.4] 0x7010:0x05 0x01 BOOLEAN      Channel 5, Data 1
  [0x0004.5] 0x7010:0x06 0x01 BOOLEAN      Channel 6, Data 1
  [0x0004.6] 0x7010:0x07 0x01 BOOLEAN      Channel 7, Data 1
  [0x0004.7] 0x7010:0x08 0x01 BOOLEAN      Channel 8, Data 1
  [0x0005.0] 0x7010:0x09 0x01 BOOLEAN      Channel 9, Data 1
  [0x0005.1] 0x7010:0x0A 0x01 BOOLEAN      Channel 10, Data 1
  [0x0005.2] 0x7010:0x0B 0x01 BOOLEAN      Channel 11, Data 1
  [0x0005.3] 0x7010:0x0C 0x01 BOOLEAN      Channel 12, Data 1
  [0x0005.4] 0x7010:0x0D 0x01 BOOLEAN      Channel 13, Data 1
  [0x0005.5] 0x7010:0x0E 0x01 BOOLEAN      Channel 14, Data 1
  [0x0005.6] 0x7010:0x0F 0x01 BOOLEAN      Channel 15, Data 1
  [0x0005.7] 0x7010:0x10 0x01 BOOLEAN      Channel 16, Data 1
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0006.0] 0xF100:0x01 0x01 BOOLEAN      K-Bus Cycle Overrun Flag
  [0x0006.1] 0xF100:0x02 0x01 BOOLEAN      Input Process Data Hold Ack.
  [0x0006.2] 0xF100:0x03 0x01 BOOLEAN      Output Process Data Hold Ack.
  [0x0006.3] 0xF100:0x04 0x01 BOOLEAN      Output Process Data Clear Ack.
  [0x0006.4] 0x0000:0x00 0x0B
  [0x0007.7] 0x10F3:0x04 0x01 BOOLEAN      New Message Available
  [0x0008.0] 0xF100:0x05 0x10 UNSIGNED16   Diagnostics Status Word
  [0x000A.0] 0x6030:0x01 0x10 UNSIGNED16   Channel 1, Word 1
  [0x000C.0] 0x6030:0x02 0x10 UNSIGNED16   Channel 2, Word 1
  [0x000E.0] 0x6030:0x03 0x10 UNSIGNED16   Channel 3, Word 1
  [0x0010.0] 0x6030:0x04 0x10 UNSIGNED16   Channel 4, Word 1
  [0x0012.0] 0x6040:0x01 0x10 UNSIGNED16   Channel 1, Word 1
  [0x0014.0] 0x0000:0x00 0x00
  [0x0014.0] 0x6040:0x02 0x10 UNSIGNED16   Channel 2, Word 1
  [0x0016.0] 0x6040:0x04 0x10 UNSIGNED16   Channel 4, Word 1
  [0x0018.0] 0x6000:0x01 0x01 BOOLEAN      Channel 1, Data 1
  [0x0018.1] 0x6000:0x02 0x01 BOOLEAN      Channel 2, Data 1
  [0x0018.2] 0x6020:0x01 0x01 BOOLEAN      Channel 1, Data 1
  [0x0018.3] 0x6020:0x02 0x01 BOOLEAN      Channel 2, Data 1
  [0x0018.4] 0x6020:0x03 0x01 BOOLEAN      Channel 3, Data 1
  [0x0018.5] 0x6020:0x04 0x01 BOOLEAN      Channel 4, Data 1
  [0x0018.6] 0x6020:0x05 0x01 BOOLEAN      Channel 5, Data 1
  [0x0018.7] 0x6020:0x06 0x01 BOOLEAN      Channel 6, Data 1
  [0x0019.0] 0x6020:0x07 0x01 BOOLEAN      Channel 7, Data 1
  [0x0019.1] 0x6020:0x08 0x01 BOOLEAN      Channel 8, Data 1
  [0x0019.2] 0x6020:0x09 0x01 BOOLEAN      Channel 9, Data 1
  [0x0019.3] 0x6020:0x0A 0x01 BOOLEAN      Channel 10, Data 1
  [0x0019.4] 0x6020:0x0B 0x01 BOOLEAN      Channel 11, Data 1
  [0x0019.5] 0x6020:0x0C 0x01 BOOLEAN      Channel 12, Data 1
  [0x0019.6] 0x6020:0x0D 0x01 BOOLEAN      Channel 13, Data 1
  [0x0019.7] 0x6020:0x0E 0x01 BOOLEAN      Channel 14, Data 1
  [0x001A.0] 0x6020:0x0F 0x01 BOOLEAN      Channel 15, Data 1
  [0x001A.1] 0x6020:0x10 0x01 BOOLEAN      Channel 16, Data 1
  [0x001A.2] 0x0000:0x00 0x0E
End slaveinfo, close socket
End program

This matches the IO I have connected (WAGO 750-354 ethercat slave with some different IO attached).
The problem I'm facing is reading and writing to/from the different IO attached. I have tried some different commands with no luck.

I tried writing
`uint8 *data_ptr;
data_ptr = ec_slave[1].outputs;
*data_ptr++ = 1 & 0x18;
`
from the tutorial for writing the output with address 0x0018.0.
This is channel 1 on a 16 channel digital output card. I tried to use floating point (18.0 instead of 18) but got an error 

> hexadecimal floating constants require an exponent
  100 |             *data_ptr++ = 1 & 0x18.0;

I'm not sure if what I'm trying to do is correct, but I greatly appreciate any help I can get. I see that a similar set up was discussed here:
[Issue](https://github.com/OpenEtherCATsociety/SOEM/issues/272)

ArthurKetels, 2024-07-15:
Ehh, what you are doing is NOT like the examples in the tutorial. The most convenient way is to define a packed struct that overlaps the defined elements of the slave mapping 1:1. Assign that struct pointer to ec_slaves[1].outputs. Then manipulate the struct instead of the IOmap directly. This is also the way the tutorial text explains.

What you do is assign data_pointer to the first output mapping address of slave 1 (so far so good). 
Then increment the pointer by 1 (uint8 size) and put the value 1 & 0x0018 (evaluates to 0) to that address. This has exactly zero effect because that logical address is not used in the mapping above ([0x0000.4] 0x0000:0x00 0x0C).

I guess what you wanted to do is:
```
data_ptr += 0x0018; // increase pointer to logical address 0x0018
*data_ptr |= 0x01; // take the current value OR 1, ie set the first bit
```

sigurdrb, 2024-07-16:
Thank you for your quick answer. I really appreciate it.

I managed to write directly to the address as you suggested.
I tried to define a packed struct  so I could easily write to the different boolean channels as shown in the slaveinfo printout above.
I made a struct that looks like:
```
typedef struct PACKED
{
        bool ch1;
        bool ch2;
        bool ch3;
        bool ch4;
        bool ch5;
        bool ch6;
        bool ch7;
        bool ch8;
        bool ch9;
        bool ch10;
        bool ch11;
        bool ch12;
        bool ch13;
        bool ch14;
        bool ch15;
        bool ch16;
} out_WAGOt;

```
and writing a true boolean value to ch 5 (address 0x0004.4).

out_WAGOt    *out_WAGO;

out_WAGO = (out_WAGOt*) ec_slave[1].outputs;
out_WAGO += 0x0004;
out_WAGO->ch5 = true;   

I see on the slave device that no output channel is set high. Can you see where I'm wrong?

nakarlsson, 2024-07-16:
Bool in C doesn’t work that way. For that to work you’d need to make use of bitfields.

Google on C bitfields and make the booleans Chx bitfields in an uint16_t.

sigurdrb, 2024-07-18:
Ok! Thanks for the reply. I will check it out.
I will mark this question as completed since I managed to write to the slave with the IOmap feature.

closed 2024-07-18 by sigurdrb