OpenEtherCATsociety/SOEM #839: Still can't quite figure out how to do PDO mapping

BongoBill, 2024-07-19:
Hi, and apologies in advance for my ignorance,

Yes, I did RTFM, and searched thru the documentation (the readme.txt and this issues forum), but still not having any luck.

Most helpful was this answer:  https://github.com/OpenEtherCATsociety/SOEM/issues/400#issuecomment-626311283, but I notice it doesn't even evaluation the return integer of ec_SDOwrite.  Should I also be ignoring it?  Maybe it often/always returns 0?

I saw the warnings about doing this only in PREOP, so I #ifdef'd out my code that goes past PREOP, so I think that's not my issue.

Additionally, I will admit here more ignorance... it seems unintuitive to me that PDO mapping should have ec_SDOwrite's 2nd and 3rd parameter as the data being mapped, and the 5th and 6th parameters being the PDO mapping location, but I did as the posting suggested.  (I tried it the opposite way too, switching parameters 2 & 3 with 5 & 6, but that also returned 0, so my misunderstanding of what is "to" vs. "from" doesn't appear to be my problem anyway...  I just mention that misunderstanding for completeness/confession.

And before someone asks, no, I didn't read and understand all of ETG's documentation.  I don't have that many years left.  :)

The readme implies the cool configuration hook method, which I confess, I haven't tried yet.  I'm just trying to get my ultra-simple inline method of "take the bus up to PREOP; assign the PDO mapping hopefully getting 1's in response to ec_SDOwrite(...)" to succeed first.

I've scoured the Issues, but if there's one clear description of how to do PDO mapping in the simplest case, I'm all ears.  

Thanks for any advice.

BongoBill, 2024-07-20:
I guess you can close this.  I figured out that parameters 2 & 3 ARE the PDO Mapping and 5 & 6 WAS the data being mapped (not what I inferred from the link'ed comment).  What I also figured out, after I found someone else's code, is parameter 6 is a combination of the data's index, subindex AND length, arithmetically =  index * 0x10000 + subindex * 0x100 + length (in bits).  Maybe this is a feature of my device, and not something that even should be documented?  In any case, I guess I do have to read and understand the EtherCAT spec and the spec for the device to move forward.

nakarlsson, 2024-07-22:
> parameter 6 is a combination of the data's index, subindex AND length, arithmetically = index * 0x10000 + subindex * 0x100 + length (in bits).

Thats is described in ETG 1000.5 som or .6.

closed 2024-07-22 by nakarlsson