OpenEtherCATsociety/SOEM #844: Simultaneous SDO Read/Write

thomazach, 2024-08-02:
I have a use case that at full scale will have 144 axes being controlled and configured using CoE. I've been starting to build the infrastructure for this, but have run into an issue/limitation with the `ecx_SDOread` and `ecx_SDOwrite` functions. My understanding of the tutorial.txt and source code is that these functions take the index, sub index, and data, and then send it to the slave which ultimately returns the result. My current understanding of CoE is that a large number of slaves can be accessed in a single EtherCAT frame in separate datagrams, each of which has its own CANopen data with the proper index and sub index as described by the slave's documentation. 

My goal is to assemble the CANopen information for each slave and then send it in a single Ethernet frame. Is this functionality achievable with SOEM? Have I missed something fundamental about SDO in the CoE docs?

ArthurKetels, 2024-08-03:
CoE is in its core an asynchronous request/response protocol. It are the corner cases that make it hard to do simple optimizations. But the more you know about your specific use case the easier it is to come up with an acceptable approach. 

You can combine multiple datagrams in one frame but that is one of the last optimization steps you would take. The overhead of the Ethernet header is relatively small, so the gain of combining datagrams into one packet is not so big. 

The minimum size of an EtherCAT packet is 64bytes (same as Ethernet), a shorter packet will be automatically padded to 64 bytes by the NIC. So when you have to send multiple short datagrams, then yes, this is an easy optimization. But more than a handful is quickly not worth the trouble. Because there is one big disadvantage of multi datagrams in a packet. When there is one  bit fault in a packet, all datagrams become invalid. As there is only one CRC that protects the packet.

Suppose you have 4 datagrams for 4 different slaves and this packet fails its CRC then you have to redo all 4 slaves. And because CoE is multi-datagram sequence you could have to to quite a bit of work. Each slave could be in a different phase of the protocol and therefore require different recovery. 

The easy way of increasing the CoE throughput of SOEM is simply to spawn multiple tasks that each do their own CoE for a single slave. SOEM is multi-tasking safe. But beware of the the buffer limit (MAXBUF) that is default 16. You could increase this when necessary.

Another strategy that is used by other EtherCAT masters is to split the mailbox read or write in a top and bottom halve. A slave mailbox is relatively large (min 16, max 1500 or so bytes). So one datagram could occupy the whole MTU of Ethernet. Even when CoE only uses a few bytes (because we only read an object of one byte for example) we still have to transfer the whole mailbox in order to clear it for the next exchange. But actually it is only necessary to transfer the first and the last byte of the mailbox. 

So the strategy is to send a datagram with only the occupied bytes, and attached a second datagram with only the last byte of the mailbox. Of course only do this when it is a net gain, you have to take into account the extra overhead of a datagram. On average this will reduce the bandwidth on the wire quite a bit. 

Also think about it if you really want to have 144 axes connected to the same EtherCAT bus? You could also split the slaves over multiple physical NIC's. SOEM can handle that perfectly fine with multiple contexts.

1GuoHaiTong1, 2024-09-23:
Dear Arthur, i want to combine multiple coe datagrams in one frame when configure one slave. but the program always fails in ecx_mbxreceive(wkc <= 0).
Have I miss something important about SDO Writing? 

note: 1. each coe datagram datasize is <= 4 bytes;
         2. the whole frame is < 1518 bytes
         3. Attacked is the captured packet file
int ecx_SDOwritelist(ecx_contextt *context, uint16 Slave, uint16 Index[], uint8 SubIndex[],
                 int listsize, int psize[], void *p[], int Timeout)
{
   ec_SDOt *aSDOp,*SDOp;
   int wkc;
   ec_mbxbuft MbxIn, MbxOut[listsize];
   uint8 cnt;
   const uint8 *hp;

   ec_clearmbx(&MbxIn);
   /* Empty slave out mailbox if something is in. Timeout set to 0 */
   wkc = ecx_mbxreceive(context, Slave, (ec_mbxbuft *)&MbxIn, 0);
   aSDOp = (ec_SDOt *)&MbxIn;
   char* parr[listsize];
   for(int i = 0; i < listsize; i++)
   {
      ec_clearmbx(&MbxOut[i]);
      SDOp = (ec_SDOt *)(&MbxOut[i]);
      SDOp->MbxHeader.length = htoes(0x000a);
      SDOp->MbxHeader.address = htoes(0x0000);
      SDOp->MbxHeader.priority = 0x00;
      /* get new mailbox counter, used for session handle */
      cnt = ec_nextmbxcnt(context->slavelist[Slave].mbx_cnt);
      context->slavelist[Slave].mbx_cnt = cnt;
      SDOp->MbxHeader.mbxtype = ECT_MBXT_COE + MBX_HDR_SET_CNT(cnt); /* CoE */
      SDOp->CANOpen = htoes(0x000 + (ECT_COES_SDOREQ << 12)); /* number 9bits service upper 4 bits */
      SDOp->Command = ECT_SDO_DOWN_EXP | (((4 - psize[i]) << 2) & 0x0c); /* expedited SDO download transfer */
      SDOp->Index = htoes(Index[i]);
      SDOp->SubIndex = SubIndex[i];
      hp = p[i];
      /* copy parameter data to mailbox */
      memcpy(&SDOp->bdata[0], hp, psize[i]);
      parr[i] = (char*)MbxOut[i];
   }
   SDOp = (ec_SDOt *)(&MbxOut[0]);
   wkc = ecx_mbxsendlist(context, Slave,listsize, (void**)parr, EC_TIMEOUTTXM);
   if (wkc > 0)
   {
      ec_clearmbx(&MbxIn);
         /* read slave response */
      wkc = ecx_mbxreceive(context, Slave, (ec_mbxbuft *)&MbxIn, Timeout);// Does this function work in this case ?  
      /* response should be CoE, SDO response, correct index and subindex */
   } 
   return wkc;     
}

int ecx_mbxsendlist(ecx_contextt *context, uint16 slave, 
                 int listsize, void *p[], int timeout)
{
   uint16 mbxwo,mbxl,configadr;
   int wkc;

   wkc = 0;
   configadr = context->slavelist[slave].configadr;
   mbxl = context->slavelist[slave].mbx_l;
   if ((mbxl > 0) && (mbxl <= EC_MAXMBX))
   {
      if (ecx_mbxempty(context, slave, timeout))
      {
         mbxwo = context->slavelist[slave].mbx_wo;
         /* write slave in mailbox */
         printf("prepare send mbx offset: %d mbx write len: %d\n",mbxwo, mbxl);
         uint8 idx;
         idx = ecx_getindex(context->port);
         ecx_setupdatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FPWR, idx, configadr, mbxwo, mbxl,p[0]);
         int i = 0;
         for(i = 1; i < listsize-1; i++)
         {
            ecx_adddatagram(context->port, &(context->port->txbuf[idx]),EC_CMD_FPWR, idx, TRUE, configadr, mbxwo, mbxl, p[i]);
         }
         if(i < listsize)
         {
            ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FPWR, idx, FALSE, configadr, mbxwo, mbxl, p[i]);
         }
         wkc = ecx_srconfirm(context->port, idx, timeout);
         ecx_setbufstat(context->port, idx, EC_BUF_EMPTY);
         printf("slave %d,send coe frame retval: %d\n",slave,wkc);
         return wkc;
      }
      else
      {
         wkc = 0;
      }
[test.zip](https://github.com/user-attachments/files/17096163/test.zip)

   }

   return wkc;
}

1GuoHaiTong1, 2024-09-24:
Dear Arthur, i want to combine multiple coe datagrams in one frame when configure one slave. but the program always fails in ecx_mbxreceive(wkc <= 0).
Have I missed something important about SDO Writing? 

note:  1 Each coe datagram datasize is <= 4 bytes;
          2 The whole frame is < 1518 bytes
          3 Attached is the captured packet file
          
[test.zip](https://github.com/user-attachments/files/17106586/test.zip)

`int ecx_SDOwritelist(ecx_contextt *context, uint16 Slave, uint16 Index[], uint8 SubIndex[],
                 int listsize, int psize[], void *p[], int Timeout)
{
   ec_SDOt *aSDOp,*SDOp;
   int wkc;
   ec_mbxbuft MbxIn, MbxOut[listsize];
   uint8 cnt;
   const uint8 *hp;

   ec_clearmbx(&MbxIn);
   /* Empty slave out mailbox if something is in. Timeout set to 0 */
   wkc = ecx_mbxreceive(context, Slave, (ec_mbxbuft *)&MbxIn, 0);
   aSDOp = (ec_SDOt *)&MbxIn;
   char* parr[listsize];
   for(int i = 0; i < listsize; i++)
   {
      ec_clearmbx(&MbxOut[i]);
      SDOp = (ec_SDOt *)(&MbxOut[i]);
      SDOp->MbxHeader.length = htoes(0x000a);
      SDOp->MbxHeader.address = htoes(0x0000);
      SDOp->MbxHeader.priority = 0x00;
      /* get new mailbox counter, used for session handle */
      cnt = ec_nextmbxcnt(context->slavelist[Slave].mbx_cnt);
      context->slavelist[Slave].mbx_cnt = cnt;
      SDOp->MbxHeader.mbxtype = ECT_MBXT_COE + MBX_HDR_SET_CNT(cnt); /* CoE */
      SDOp->CANOpen = htoes(0x000 + (ECT_COES_SDOREQ << 12)); /* number 9bits service upper 4 bits */
      SDOp->Command = ECT_SDO_DOWN_EXP | (((4 - psize[i]) << 2) & 0x0c); /* expedited SDO download transfer */
      SDOp->Index = htoes(Index[i]);
      SDOp->SubIndex = SubIndex[i];
      hp = p[i];
      /* copy parameter data to mailbox */
      memcpy(&SDOp->bdata[0], hp, psize[i]);
      parr[i] = (char*)MbxOut[i];
   }
   SDOp = (ec_SDOt *)(&MbxOut[0]);
   wkc = ecx_mbxsendlist(context, Slave,listsize, (void**)parr, EC_TIMEOUTTXM);
   if (wkc > 0)
   {
      ec_clearmbx(&MbxIn);
         /* read slave response */
      wkc = ecx_mbxreceive(context, Slave, (ec_mbxbuft *)&MbxIn, Timeout);// Does this function work in this case ?  
      /* response should be CoE, SDO response, correct index and subindex */
   } 
   return wkc;     
}

int ecx_mbxsendlist(ecx_contextt *context, uint16 slave, 
                 int listsize, void *p[], int timeout)
{
   uint16 mbxwo,mbxl,configadr;
   int wkc;

   wkc = 0;
   configadr = context->slavelist[slave].configadr;
   mbxl = context->slavelist[slave].mbx_l;
   if ((mbxl > 0) && (mbxl <= EC_MAXMBX))
   {
      if (ecx_mbxempty(context, slave, timeout))
      {
         mbxwo = context->slavelist[slave].mbx_wo;
         /* write slave in mailbox */
         printf("prepare send mbx offset: %d mbx write len: %d\n",mbxwo, mbxl);
         uint8 idx;
         idx = ecx_getindex(context->port);
         ecx_setupdatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FPWR, idx, configadr, mbxwo, mbxl,p[0]);
         int i = 0;
         for(i = 1; i < listsize-1; i++)
         {
            ecx_adddatagram(context->port, &(context->port->txbuf[idx]),EC_CMD_FPWR, idx, TRUE, configadr, mbxwo, mbxl, p[i]);
         }
         if(i < listsize)
         {
            ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FPWR, idx, FALSE, configadr, mbxwo, mbxl, p[i]);
         }
         wkc = ecx_srconfirm(context->port, idx, timeout);
         ecx_setbufstat(context->port, idx, EC_BUF_EMPTY);
         printf("slave %d,send coe frame retval: %d\n",slave,wkc);
         return wkc;
      }
      else
      {
         wkc = 0;
      }
   }

   return wkc;
}
`

ArthurKetels, 2024-09-24:
@1GuoHaiTong1, please do not reuse an issue by another poster, make a new one. Second, some indentation would help readability.

Now to your question. What you want to do is not possible. A slave has only one input mailbox, and until a response is generated by the slave in the out-mailbox, the in-mailbox is not to be used. What you are doing is sending multiple in-mailboxes to the same slave in the same packet. Only the first datagram will succeed, then the in-mailbox is locked by the ESC until the slave firmware reads the contents.

Please read the EtherCAT documentation on how the CoE protocol and mailboxes actually work.

Your only option with one slave is to do the SDOwrite/read one by one.  If you have multiple slaves you can open multiple tasks and have each task handle a slave asynchronously. SOEM is build re-entrant safe.

If your issue is that the slave is too slow processing CoE request, then talk to the manufacturer, or if it is in your control, optimize the firmware. SOES is rocket fast in CoE responses.

1GuoHaiTong1, 2024-09-25:
OK,thanks for your advice.

closed 2024-09-25 by ArthurKetels