OpenEtherCATsociety/SOEM #846: How to get driver ready?

PhamTuTai, 2024-08-06:
Hi everyone.
I have OP status driver now.
Can anyone tell me which register I have to write value to for driver ready?
Thanks.

ArthurKetels, 2024-08-06:
First have a look at the tutorial. It explains the basic steps.

PhamTuTai, 2024-08-07:
> First have a look at the tutorial. It explains the basic steps.
When using PLC connected by EtherCAT. After configuring slave into OP mode. Then call Axis enable function to ready driver.
I don't know if there is any document describing this issue or not?
For example, A_PWR (Axis enable), what value will the master send to which register for slave?
[image]

ArthurKetels, 2024-08-07:
This looks like a question for the slave manufacturer, not for SOEM. It seems you have the manual, so that explains how to do it?

PhamTuTai, 2024-08-07:
These are PLC commands. And it is master to control slave.
This document is user manual for PLC Xinje. Connect to slave by EtherCAT standard.
So I want to ask if anyone here knows about this issue?
For example how will the A_PWR function be executed from master to slave. It will have common rules for all slave manuals.

closed 2024-10-29 by PhamTuTai