OpenEtherCATsociety/SOEM #849: support for virtual PDO in EL6695?

pingplug, 2024-08-21:
EL6695 has a virtual RxPDO (0x1608) and a virtual TxPDO (0x1A08).
Is there any plan for support virtual PDO?

ArthurKetels, 2024-08-21:
A virtual PDO has no entries. Support would also be "virtual". So what would be the point?

pingplug, 2024-08-22:
in EL6695 manual:
```
If no TwinCAT EtherCAT Master with [Load PDO info from device] is available on the other side, the PDO upload mechanism must be reproduced there:
- Read PDO assignment objects 0x1C12, 0x1C13
- then read out the PDO Mapping Objects 0x160x (outputs), 0x1A0x (inputs) and 
- read out the PDOs 0x60xx, 0x70xx accordingly
```
I find the PDO Mapping Objects becomes a mess after starting SOEM or `slaveinfo`. So I wonder if there is an option to disable writting PDO Mapping Objects?

ArthurKetels, 2024-08-24:
SOEM does exactly what you have copied above from the EL6695 manual. Can you describe what is a "mess"? And does the configuration succeed with SOEM?

pingplug, 2024-08-25:
I set 128 BYTE inputs and 128 BYTE outputs in EL6695, and run `slaveinfo` from another side (EL6695-0002)
the topology is: master -- EK1100 -- EL2008 --EL6695-0002
what I expected:
```
0x1C12: 0x1608
  0x1608: 0x7000:01:08 ... 0x0700:80:08
0x1C13: 0x1A01, 0x1A08
  0x1608: 0x0000:00:0C 0x1802:09:01 0x1802:07:01 0x0000:00:01 0x10F4:10:01
  0x1A08: 0x6000:01:08 ... 0x0600:80:08
```
the output form `slaveinfo` and wireshark capture is there:
[slaveinfo.zip](https://github.com/user-attachments/files/16740720/slaveinfo.zip)

ArthurKetels, 2024-08-26:
There is something funny going on with this slave. When reading the PDO mapping with Complete-Access mode the PDO items are as you expected, so we have 128 entries each with length 8 bits for outputs. But when we do the same in standard mode we get a different response from the slave and it outputs a compressed PDO as shown by slaveinfo. 

The the messing up happens in the slave and not with SOEM.

There is another issue that when requesting the pre-OP to safe-OP transition the slave responds with error 0x0003. And this code does not exists in the specification.

pingplug, 2024-08-31:
So is there a way to force SOEM to use CA in PDO items?

ArthurKetels, 2024-09-01:
SOEM does use CA when mapping the PDO items. You can see this in the Wireshark capture. It is only when slaveinfo prints out the PDO mapping it uses the single object read method. Both should return the same answer however. And this is a slave issue. 

If you look at the code of slaveinfo you see why it is easier to not use CA there. Secondly there multiple slaves out there that do not handle CA very well and crash. 

If your application needs the detailed PDO mapping information, there is nothing holding you back using CA to read the PDO objects. SOEM does give you all the tools you need.