OpenEtherCATsociety/SOEM #853: eRob motor control using EtherCAT

Jennifer-Mo-sys, 2024-09-09:
Hello,
I was reviewing the SOEM library issues and came across the one you opened [here](https://github.com/OpenEtherCATsociety/SOEM/issues/712). I am currently facing similar challenges and wanted to inquire if you have made any progress in resolving the issues mentioned.
The file I am using closely resembles the final version you uploaded, with only minor modifications that I have included, which can be found [here](https://github.com/VModugno/ether_motor_test). I am primarily testing your code and am attempting to send commands to my motor (an eRob motor). 

I apologize for reaching out in this manner, but any guidance or insights you could provide would be greatly appreciated as I am facing some difficulties.
Thank you in advance for your assistance.

Rohithossain007, 2024-10-11:
I am also trying to turn the motor on using etherCAT I know I don't have answer for your question but could you please tell me how did you approach? Turning on the motor?

jennhhw, 2024-10-16:
To turn on the motor I referred to the documentation. My motor turned on once I had plugged in the power supply. If you mean turning the motor on and off via EtherCAT, this wasn't an option with my motor.

ZeroErrControl, 2024-11-28:
You can refer to this link: [https://github.com/ZeroErrControl/eRob_SOEM_linux/blob/main/test/linux/simple_test/eRob_test.cpp](https://github.com/ZeroErrControl/eRob_SOEM_linux/blob/main/test/linux/simple_test/eRob_test.cpp). I hope it will be helpful to you.

closed 2024-11-28 by ArthurKetels