OpenEtherCATsociety/SOEM #857: How to synchronize on stm32?

PhamTuTai, 2024-09-20:
Hello everyone.
I have run the servo on linux based on the tutorial on the forum.
However, according to red_test. We need to calculate PI to calculate the time delay. But this value is calculated in ns.
Currently I am developing ethercat on stm32f7. I am not using RTOS.
I want to ask if anyone on this forum has solved this time compensation problem? And if anyone has any idea how to handle this problem please let me know.
Thanks everyone.

ArthurKetels, 2024-09-20:
There is no problem. You can do exactly the same on bare metal STM32f7 as on standard linux. The only thing you need to do is to adapt the OSAL and OSHW abstraction to your platform. Making a network driver is most of the work. 

I have successfully implemented SOEM on 8bit 16bit and 32bit micro controllers without much fuss. 

To be specific on your time keeping. EtherCAT has an internal time representation of 1ns. But the actual resolution is only 10ns. As long your hardware can support something with a few us resolution you are fine.

PhamTuTai, 2024-09-21:
Thanks for your feedback.
I have successfully deployed SOEM on stm32f7 to the state "Operational state reached for all slaves.". This is where the connection problem occurs.
I successfully call the function ec_dcsync0(1, TRUE, 1000000U, 0U). I continue to try the ecatthread function with different cycletimes. The driver still runs continuously.
However, in linux, there will be a part that calculates "toff" so that the cycletime is the most accurate. And that thread will sleep very accurately with the function "clock_nanosleep".
Because I use stm32 without rtos. What can I do to make the communication cycle between master and slave continuous? Is it possible to use Timer?
If you have deployed SOEM on the processor, can you help me with the keyword to handle the problem?

ArthurKetels, 2024-09-21:
The STM32 has a load of hardware timers that you can utilize. Pick one that is free and set the timer expiry to approximately the cycle time of your synchronous task. At timer expiry you generate an interrupt that will start your task. Rearm with the cycle time + toff. The feedback loop will take care of the correct adjustments to synchronize the timer expiry with the requested EtherCAT global time. This is basic PLL control that should be bread and butter for a bare metal micro-controller developer.

dchukhutin, 2024-10-13:
> Rearm with the cycle time + toff. The feedback loop will take care of the correct adjustments to synchronize the timer expiry with the requested EtherCAT global time. 

If the master is on a microcontroller and all EtherCAT transactions occur in the timer interrupt handler, is it necessary to adjust the timer period?
In a microcontroller, the timer interrupts will be absolutely accurate. No jitter.

ArthurKetels, 2024-10-13:
You have the wrong concept of time and clocks. No clock is absolutely precise. Even the most precise atomic clock has drift, offset, jitter, etc. 

For EtherCAT slaves a common requirement is the drift of the reference oscillator to be less than 20ppm. This is already regarded a precision clock. 

Let's assume your micro-controller has a similar oscillator. And just to calculate easy, assume the slave is perfect 0ppm error, your micro-controller clock is 1ppm too high. Both start perfectly at the correct time, then after 1e6 cycles your timer interrupt has made exactly one cycle extra compared to your slave. 1e6 cycles at 1 kHz is only 1000 seconds. 

The purpose of synchronous communication between master and slaves is not only to have the same number of cycles (drift control) but also to have a constant offset within one cycle (phase control). So the requirement is even more strict. 

The only way to have systems with independent clocks behave phase synchronous is to observe the clock differences and adjust. Who observes who, and who and how to adjust best, has been a hot debate amongst experts. The method chosen by EtherCAT is to distribute a reference clock so all others may observe and adjust. My example in the SOEM test programs is to use the first slave as reference and adjust the master time to that. TwinCAT uses the master time as reference and adjusts the slave to that. In other scenarios I have used an external GPS reference. It just depends on your application what is best.

dchukhutin, 2024-10-13:
> You have the wrong concept of time and clocks. No clock is absolutely precise. Even the most precise atomic clock has drift, offset, jitter, etc........

Thank you very much for your detailed answer !

closed 2024-10-22 by PhamTuTai