OpenEtherCATsociety/SOEM #858: EC_DO64 : Invalid input configuration

Vjinqu, 2024-09-20:
Hi,
I had use SOEM with Beckhoff IOs sucessfully, but with EC_DO64 I have problems. EC_DO64 is slave with only 64-bit output. however，I found the output IO in slaveinfo changed to the input IO，why?
I looked for many issues but didn't find the same problem. I tried similar issues, but it  always prompt : Invalid input configuration.

here is the slaveinfo imformation
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{C2F61E97-62B7-4935-8427-BBE4866515A2} succeeded.
1 slaves found and configured.
Calculated workcounter 1
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
   Name:EC-DO64
   Output size: 0bits
   Input size: 64bits
   State: 18
   Delay: 0[ns]
   Has DC: 1
   DCParentport:0
   Activeports:1.0.0.0
   Configured address: 1001
   Man: 00000af0 ID: 20240510 Rev: 20240510
   SM0 A:1000 L: 128 F:00010026 Type:1
   SM1 A:1080 L: 128 F:00010022 Type:2
   SM2 A:1100 L:   8 F:00010064 Type:4
   SM3 A:1400 L:   2 F:00010020 Type:3
   FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
   FMMUfunc 0:1 1:2 2:3 3:0
   MBX length wr: 128 rd: 128 MBX protocols : 04
   CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
   Ebus current: 0[mA]
   only LRD/LWR:0
  PDO mapping according to CoE :
    SM2 inputs
       addr b   index: sub bitl data_type    name
    [0x0000.0] 0x7000:0x01 0x10 UNSIGNED16   OUTPUT1
    [0x0002.0] 0x7000:0x02 0x10 UNSIGNED16   OUTPUT2
    [0x0004.0] 0x7000:0x03 0x10 UNSIGNED16   SubIndex 003
    [0x0006.0] 0x7000:0x04 0x10 UNSIGNED16   SubIndex 004
    SM3 outputs
       addr b   index: sub bitl data_type    name
  End slaveinfo, close socket
  End program

Here is the xml file、wireshark capturing and slaveinfo -sdo 
[EC_DO64.zip](https://github.com/user-attachments/files/17070842/EC_DO64.zip)

I know there may be a problem with the firmware, but can we solve it through software configuration, after all, twincat can make it work through xml.
Any suggestions for us, thank you.

ArthurKetels, 2024-09-20:
Your slave has a well know firmware bug where it mixes up the inputs/outputs with SM association. From above:
```
SM0 A:1000 L: 128 F:00010026 Type:1
SM1 A:1080 L: 128 F:00010022 Type:2
SM2 A:1100 L: 8 F:00010064 Type:4
SM3 A:1400 L: 2 F:00010020 Type:3
```
The types should actually be SM2 type 3 and SM3 type 4. SOEM just follows the description the slave reports.

You can circumvent this by manually modifying the response coming from that particular slave.

Of course using slaves with these kind of bugs do question the quality of the firmware. I would not support such a manufacturer if they do not promptly fix their bugs.

Vjinqu, 2024-09-21:
Thank you for your quick response, I do appreciate it.
I found the same problem in #423  and #224 yesterday, and followed to modify ethercatcoe file , and it worked.
But the problem ( Not all slaves reached safe operational state.  )remains unresolved. There seem to be other undiscovered problems.

the new slaveinfo imformation:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{C2F61E97-62B7-4935-8427-BBE4866515A2} succeeded.
1 slaves found and configured.
Calculated workcounter 2
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:EC-DO64
 Output size: 64bits
 Input size: 0bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000af0 ID: 20240510 Rev: 20240510
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   8 F:00010064 Type:3
 SM3 A:1400 L:   2 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0xD00E1BC0.0] 0x7000:0x01 0x10 UNSIGNED16   OUTPUT1
  [0xD00E1BC2.0] 0x7000:0x02 0x10 UNSIGNED16   OUTPUT2
  [0xD00E1BC4.0] 0x7000:0x03 0x10 UNSIGNED16   SubIndex 003
  [0xD00E1BC6.0] 0x7000:0x04 0x10 UNSIGNED16   SubIndex 004
  SM3 outputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program

new slaveinfo wireshark here:
[EC_DO64 modify ethercatcoe.zip](https://github.com/user-attachments/files/17082827/EC_DO64.modify.ethercatcoe.zip)

I don't know if it matters that this place is wrong.
[image]

I tried to modify 0x1c00 by using ec_slave[slc].PO2SOconfig = &ECDO64setup because I knew 0x1c00 should be 1234, but I got an error
the slaveinfo sdo 0x1c00
[image]
the error
[image]
my code
```
int ECDO64setup(uint16 slave)
{
    int retval;
    uint8 u8val;
    uint16 u16val;
    uint32 u32val;

    retval = 0;
    u8val = 0;
    retval += ec_SDOwrite(slave, 0x1c00, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
[... 166 lines cut ...]
```
simple_test wireshark here:
[simple_test_SDOwriteERROR.zip](https://github.com/user-attachments/files/17082934/simple_test_SDOwriteERROR.zip)

Next, I tried to change the CA parameter of ec_SDOwrite to TRUE, but there was a new ERROR
code:
```
    u8val = 0;
    retval += ec_SDOwrite(slave, 0x1c00, 0x00, TRUE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u8val = 1;
    retval += ec_SDOwrite(slave, 0x1c00, 0x01, TRUE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u8val = 2;
    retval += ec_SDOwrite(slave, 0x1c00, 0x02, TRUE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u8val = 3;
    retval += ec_SDOwrite(slave, 0x1c00, 0x03, TRUE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u8val = 4;
    retval += ec_SDOwrite(slave, 0x1c00, 0x04, TRUE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
    u8val = 4;
    retval += ec_SDOwrite(slave, 0x1c00, 0x00, TRUE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
```
error :
[image]

Status 12 indicates a malfunction during the PREOP->SAfeOP.
I am new to ethercat and can't analyze wireshark data,  I didn't know what to do next. can you help me.

ArthurKetels, 2024-09-21:
```
int ECDO64setup(uint16 slave)
{
    // force SM types to SM2 = outputs and SM3 = inputs
    ec_slave[slave].SMtype[2] = 3;
    ec_slave[slave].SMtype[3] = 4;
    // force SM3 to length 0 and not enabled.
    ec_slave[slave].SM[3].SMlength = 0;
    ec_slave[slave].SM[3].SMflags = 0;

    qDebug()<< "ECDO64 setup slave" << slave;
    return 1;
}
```
This should do it. 

It is no use writing to a read only object as you tried in your code. That is why you get the error messages.

The manufacturer has some bugs in the firmware SDO descriptions. They fixed it in the XML (sort of) so TwinCAT works. I regard this company as incompetent EtherCAT suppliers.

Vjinqu, 2024-09-24:
Wow, that's it. After setting up the SMlength and SMflags, the slave can output correctly !
My main problem is solved.
Thank you very much for your reply, [ArthurKetels](https://github.com/ArthurKetels).

Another, I see that other issues  can set 0x1c12 or other object(read only), but every time I try to appear the above error, is the hardware set read-only caused by unchangeable, or where not set well?

closed 2024-09-26 by Vjinqu