OpenEtherCATsociety/SOEM #859: DC activation

Edphoenixmax, 2024-09-24:
I wish all you good.
I have noticed that in SOEM ,the dcactivation only has 3 and 7 in register 0x981.Is that means Ethercat only supports these two dc activation mode?
Is there any device support other dc activation mode like write 5 in register 0x981 or other value?
Thanks for your answer

Edphoenixmax, 2024-09-25:
Another question,If I set value in register0x980 (DClatch signal) ,what is the usefulness of this register?

ArthurKetels, 2024-09-25:
Register 0x981 only has bit [0..2] used for most of the ESC chips. The HDL IP can use the other bits [3..9]. You can all read this in the ESC datasheet (beckhoff website).

Reg 0x980 is only used to transfer sync and latch control from master to slave (ECAT or PDI in ESC language).

Edphoenixmax, 2024-09-27:
OK I get it.So I understand why SOEM only wirte 3 and 7 in register 0x981,Is that means write value =5 in 0x981 is useless?
Thanks for your answer again

> Register 0x981 only has bit [0..2] used for most of the ESC chips. The HDL IP can use the other bits [3..9]. You can all read this in the ESC datasheet (beckhoff website).
> 
> Reg 0x980 is only used to transfer sync and latch control from master to slave (ECAT or PDI in ESC language).

ArthurKetels, 2024-09-27:
Vale = 5 would mean bit 2 and bit 0 set. This will activate only sync1. I do not know of any slave that does not use sync0 and does use sync1. So yeah, useless.