OpenEtherCATsociety/SOEM #861: Regarding IO Mapping like Sycon.net Utility

ANANTH5354, 2024-09-27:
Hello all,
I hope you all doing well !
              
SOEM master code I am here compiled like (slaveinfo fun and  simple test ) it working well.

========================================================================
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{F4271784-7547-493D-8A78-8EBAC65E2F75} succeeded.
5 slaves found and configured.
Calculated workcounter 7

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0

Slave:2
 Name:EL3602
 Output size: 0bits
 Input size: 96bits
 State: 4
 Delay: 140[ns]
 Has DC: 1
 DCParentport:1

Slave:3
 Name:EL5152
 Output size: 96bits
 Input size: 160bits
 State: 4
 Delay: 285[ns]
 Has DC: 1
 DCParentport:1

IOmap mapped successfully.
End slaveinfo, close socket
End program

[image]
========================================================================

Here, I want to construct the slave configuration channels (AI, AO, DI, ,DO and Data Address, Data Length) and the code address table (IO mapping) (e.g., Sycon.net utility Address table). Would you please help me, anyone?

final output like that I want here,
[Address table PMX.zip](https://github.com/user-attachments/files/17162199/Address.table.PMX.zip)

Thank you

ArthurKetels, 2024-09-27:
Try :  simple_test [network name] -map

ANANTH5354, 2024-09-27:
Okay.. I will check !!!

Could you provide me with some suggestions on how to formulate my questions.

Thank you