OpenEtherCATsociety/SOEM #862: When frequently switching the state perop to safeop, PO2SOconfig will be overwritten

yuanchanghong, 2024-09-29:
When frequently switching the state perop to safeop, PO2SOconfig will be overwritten。

Due to FMMUunused not being cleared, when FMMUc is greater than the maximum array length of FMMU
Execute the sentence `context ->slaveist [slave] FMMU[FMMUc]. LogStart = htoel(*LogAddr);` 
 Causing PO2SOconfig to be overwritten

how can i fix it?thanks!!

ArthurKetels, 2024-09-29:
Thanks for your research. There is a problem with not having a check for array bounds of FMMU[]. This needs to be addressed.

However you can not trigger this by switching the state from pre-OP to safe-OP (as many times as you like). 

This overflow can only happen when calling ecx_config_map_group() multiple times without first initializing the structs with ecx_config_init().

The best solution would be to return an error when trying to call the mapping function without first calling the init function.

yuanchanghong, 2024-09-30:
thanks,  your answer help me solve the problem