OpenEtherCATsociety/SOEM #863: Invalid Input Configuration

Photoheyler, 2024-09-29:
want to a configure  an Anedo S30e 
always when I configure it, it says there is an input missmatch

In Twincat the slave has 85 outputs and 143 inputs.
SOEM says it has Ibytes: 123, Obytes: 83

I'm not sure how to configure my slave correct.

So I get a missmatch.

This is my code snippet:

    if (ec_init(ifname))
    {
        printf("ec_init on %s succeeded.\n", ifname);
        readStateofAllSlaves();
        /* find and auto-config slaves */
        //AnedoS30Setup(4);
        if (ec_config_init(false) > 0)
        {
            printf("%d slaves found and configured.\n", ec_slavecount);
            readStateofAllSlaves();
           
            usedBytes = ec_config_map(&IOmap);
            
             /* Map I/O for each slave */
            if (usedBytes == 0)
            {
                printf("No PDOs mapped, check ESI file.\n");
                return -1;
            }

            for (int slc = 1; slc <= ec_slavecount; slc++)
            {
                printf("Slave %d Ibytes: %d, Obytes: %d\n", slc, ec_slave[slc].Ibytes, ec_slave[slc].Obytes);
            }
            printf("Mapped %d bytes of I/O data.\n", usedBytes);
            //setStateForAllSlaves(EC_STATE_PRE_OP);
            ec_send_processdata();
            wkc=ec_receive_processdata(2000);
            ec_writestate(0);
            ec_statecheck(0, EC_STATE_OPERATIONAL,  EC_TIMEOUTSTATE);
            
            
            
            This is my Esi file:
[open_io_S30e.txt](https://github.com/user-attachments/files/17179078/open_io_S30e.txt)

ArthurKetels, 2024-09-29:
Can you show the output of slaveinfo -map?
You are comparing number of inputs/outputs with bytes, these do not necessarily correspond one to one.

Photoheyler, 2024-09-30:
Do you mean the output of the Slaveinfo exe?
[image]
This is the Twin Cat config which works.

[image]

Photoheyler, 2024-09-30:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on /Device/NPF_{568F8484-72B8-4BE4-B37F-A877EC8EEAE2} succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:open:I/O Slaves
 Output size: 662bits
 Input size: 983bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000aed0 ID: 00000100 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:  83 F:00010064 Type:3
 SM3 A:1c00 L: 123 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  83 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000053 Ll: 123 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 01 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x00 0x10
  [0x0002.0] 0x7000:0x01 0x10
  [0x0004.0] 0x7000:0x02 0x08
  [0x0005.0] 0x7001:0x00 0x10
  [0x0007.0] 0x7001:0x01 0x10
  [0x0009.0] 0x7001:0x02 0x08
  [0x000A.0] 0x7002:0x00 0x10
  [0x000C.0] 0x7002:0x01 0x10
  [0x000E.0] 0x7002:0x02 0x08
  [0x000F.0] 0x7003:0x00 0x10
  [0x0011.0] 0x7003:0x01 0x10
  [0x0013.0] 0x7003:0x02 0x08
  [0x0014.0] 0x7004:0x00 0x10
  [0x0016.0] 0x7004:0x01 0x10
  [0x0018.0] 0x7004:0x02 0x08
  [0x0019.0] 0x7005:0x00 0x10
  [0x001B.0] 0x7005:0x01 0x10
  [0x001D.0] 0x7005:0x02 0x08
  [0x001E.0] 0x7006:0x00 0x10
  [0x0020.0] 0x7006:0x01 0x10
  [0x0022.0] 0x7006:0x02 0x08
  [0x0023.0] 0x7007:0x00 0x10
  [0x0025.0] 0x7007:0x01 0x10
  [0x0027.0] 0x7007:0x02 0x08
  [0x0028.0] 0x7008:0x00 0x10
  [0x002A.0] 0x7008:0x01 0x10
  [0x002C.0] 0x7008:0x02 0x08
  [0x002D.0] 0x7009:0x00 0x10
  [0x002F.0] 0x7009:0x01 0x10
  [0x0031.0] 0x7009:0x02 0x08
  [0x0032.0] 0x700A:0x00 0x10
  [0x0034.0] 0x700A:0x01 0x10
  [0x0036.0] 0x700A:0x02 0x01
  [0x0036.1] 0x700A:0x03 0x01
  [0x0036.2] 0x7016:0x00 0x10
  [0x0038.2] 0x7016:0x01 0x10
  [0x003A.2] 0x7016:0x02 0x01
  [0x003A.3] 0x7016:0x03 0x01
  [0x003A.4] 0x7018:0x00 0x08
  [0x003B.4] 0x7019:0x00 0x08
  [0x003C.4] 0x701A:0x00 0x08
  [0x003D.4] 0x701B:0x00 0x08
  [0x003E.4] 0x701C:0x00 0x08
  [0x003F.4] 0x701D:0x00 0x10
  [0x0041.4] 0x701D:0x01 0x08
  [0x0042.4] 0x701E:0x00 0x10
  [0x0044.4] 0x701E:0x01 0x08
  [0x0045.4] 0x701F:0x00 0x10
  [0x0047.4] 0x701F:0x01 0x08
  [0x0048.4] 0x7020:0x00 0x10
  [0x004A.4] 0x7020:0x01 0x08
  [0x004B.4] 0x7021:0x00 0x10
  [0x004D.4] 0x7021:0x01 0x08
  [0x004E.4] 0x7022:0x00 0x10
  [0x0050.4] 0x7022:0x01 0x10
  [0x0052.4] 0x7022:0x02 0x01
  [0x0052.5] 0x7022:0x03 0x01
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0053.0] 0x6000:0x00 0x10
  [0x0055.0] 0x6000:0x01 0x10
  [0x0057.0] 0x6000:0x02 0x01
  [0x0057.1] 0x6001:0x00 0x10
  [0x0059.1] 0x6001:0x01 0x10
  [0x005B.1] 0x6001:0x02 0x01
  [0x005B.2] 0x6002:0x00 0x10
  [0x005D.2] 0x6002:0x01 0x10
  [0x005F.2] 0x6002:0x02 0x01
  [0x005F.3] 0x6003:0x00 0x10
  [0x0061.3] 0x6003:0x01 0x10
  [0x0063.3] 0x6003:0x02 0x01
  [0x0063.4] 0x6004:0x00 0x10
  [0x0065.4] 0x6004:0x01 0x10
  [0x0067.4] 0x6004:0x02 0x01
  [0x0067.5] 0x6005:0x00 0x10
  [0x0069.5] 0x6005:0x01 0x10
  [0x006B.5] 0x6005:0x02 0x01
  [0x006B.6] 0x6006:0x00 0x10
  [0x006D.6] 0x6006:0x01 0x10
  [0x006F.6] 0x6006:0x02 0x01
  [0x006F.7] 0x6007:0x00 0x10
  [0x0071.7] 0x6007:0x01 0x10
  [0x0073.7] 0x6007:0x02 0x01
  [0x0074.0] 0x6008:0x00 0x10
  [0x0076.0] 0x6008:0x01 0x10
  [0x0078.0] 0x6008:0x02 0x01
  [0x0078.1] 0x6009:0x00 0x10
  [0x007A.1] 0x6009:0x01 0x10
  [0x007C.1] 0x6009:0x02 0x01
  [0x007C.2] 0x600A:0x00 0x10
  [0x007E.2] 0x600A:0x01 0x01
  [0x007E.3] 0x600C:0x00 0x20
  [0x0082.3] 0x600C:0x01 0x20
  [0x0086.3] 0x600C:0x02 0x01
  [0x0086.4] 0x600C:0x03 0x01
  [0x0086.5] 0x600C:0x04 0x06
  [0x0087.3] 0x600D:0x00 0x20
  [0x008B.3] 0x600D:0x01 0x20
  [0x008F.3] 0x600D:0x02 0x01
  [0x008F.4] 0x600D:0x03 0x01
  [0x008F.5] 0x600D:0x04 0x06
  [0x0090.3] 0x600E:0x00 0x20
  [0x0094.3] 0x600E:0x01 0x20
  [0x0098.3] 0x600E:0x02 0x01
  [0x0098.4] 0x600E:0x03 0x01
  [0x0098.5] 0x600E:0x04 0x06
  [0x0099.3] 0x600F:0x00 0x20
  [0x009D.3] 0x600F:0x01 0x20
  [0x00A1.3] 0x600F:0x02 0x01
  [0x00A1.4] 0x600F:0x03 0x01
  [0x00A1.5] 0x600F:0x04 0x06
  [0x00A2.3] 0x6010:0x00 0x20
  [0x00A6.3] 0x6010:0x01 0x20
  [0x00AA.3] 0x6010:0x02 0x01
  [0x00AA.4] 0x6010:0x03 0x01
  [0x00AA.5] 0x6010:0x04 0x06
  [0x00AB.3] 0x6011:0x00 0x20
  [0x00AF.3] 0x6011:0x01 0x20
  [0x00B3.3] 0x6011:0x02 0x01
  [0x00B3.4] 0x6011:0x03 0x01
  [0x00B3.5] 0x6011:0x04 0x06
  [0x00B4.3] 0x6012:0x00 0x20
  [0x00B8.3] 0x6012:0x01 0x20
  [0x00BC.3] 0x6012:0x02 0x01
  [0x00BC.4] 0x6012:0x03 0x01
  [0x00BC.5] 0x6012:0x04 0x06
  [0x00BD.3] 0x6013:0x00 0x20
  [0x00C1.3] 0x6013:0x01 0x20
  [0x00C5.3] 0x6013:0x02 0x01
  [0x00C5.4] 0x6013:0x03 0x01
  [0x00C5.5] 0x6013:0x04 0x06
  [0x00C6.3] 0x6014:0x00 0x08
  [0x00C7.3] 0x6015:0x00 0x08
  [0x00C8.3] 0x6016:0x00 0x10
  [0x00CA.3] 0x6016:0x01 0x01
  [0x00CA.4] 0x6018:0x00 0x01
  [0x00CA.5] 0x6019:0x00 0x01
  [0x00CA.6] 0x601A:0x00 0x01
  [0x00CA.7] 0x601B:0x00 0x01
  [0x00CB.0] 0x601C:0x00 0x01
  [0x00CB.1] 0x601D:0x00 0x01
  [0x00CB.2] 0x601E:0x00 0x01
  [0x00CB.3] 0x601F:0x00 0x01
  [0x00CB.4] 0x6020:0x00 0x01
  [0x00CB.5] 0x6021:0x00 0x01
  [0x00CB.6] 0x6022:0x00 0x10
  [0x00CD.6] 0x6022:0x01 0x01
End slaveinfo, close socket
End program
Drücken Sie eine beliebige Taste . . .

ArthurKetels, 2024-09-30:
Thanks again for the information. The issue is clear. In the XML file you supplied there are padding objects that will align byte and word size objects to a byte boundary. These objects are absent in the CoE mapping structure. 

An example from the above mapping list:
```
[0x0036.0] 0x700A:0x02 0x01
[0x0036.1] 0x700A:0x03 0x01
[0x0036.2] 0x7016:0x00 0x10
```
At byte 0x0036 there are two bit sized objects, that occupy bits 0 and 1. Then object 0x7016:00 follows with bit size 0x10 (2 bytes). This object is then located at 0x0036 bit 2 and further. This is obviously wrong, as that object should start at 0x0037.0 (at a byte boundary). 

In the XML file there is a padding object 0x0000:00 with size 6 bits between object 0x700A:03 and 0x7016:00. When we add all the padding the output and input size is the same as in the TwinCAT case.

So this is a bug in the firmware of the slave. The CoE mapping objects should have padding objects inserted at the right places. You should ask for a bug fix and update the firmware of the slave in question.

As an alternative stop-gap you can make a PO2SO hook function that overwrites the SM and FMMU sizes. SOEM will then do the correct thing, but for the wrong reasons.

Photoheyler, 2024-10-02:
Ok thank you very much!
I'll try to fix it like you described it.
Is there a ESI file Parser aviable or do I have to build it on my own?