OpenEtherCATsociety/SOEM #865: Cannot start EL6751 using SOEM because 0x1C32 and 0x1C33 not available

hernanda16, 2024-10-09:
I want to access EL6751 with SOEM using reference startup from TwinCAT XAE startup for EL6751
    
[image]

    ret_wr[0] = ec_SDOwrite(can_slave_ecNum, 0x1C32, 0x02, FALSE, sizeof(cycle_time), &cycle_time, EC_TIMEOUTRXM);
    ret_wr[1] = ec_SDOwrite(can_slave_ecNum, 0x1C33, 0x03, FALSE, sizeof(shift_time), &shift_time, EC_TIMEOUTRXM);

    ret_wr[2] = ec_SDOwrite(can_slave_ecNum, 0xF800, 0x00, TRUE, sizeof(_F800), &_F800, EC_TIMEOUTRXM);
    ret_wr[3] = ec_SDOwrite(can_slave_ecNum, 0x8000, 0x00, TRUE, sizeof(_8000), &_8000, EC_TIMEOUTRXM);
    ret_wr[4] = ec_SDOwrite(can_slave_ecNum, 0x8006, 0x00, TRUE, sizeof(_8006), &_8006, EC_TIMEOUTRXM);
    ret_wr[5] = ec_SDOwrite(can_slave_ecNum, 0x8008, 0x00, TRUE, sizeof(_8008), &_8008, EC_TIMEOUTRXM);
    ret_wr[6] = ec_SDOwrite(can_slave_ecNum, 0x8003, 0x00, TRUE, sizeof(_8003), &_8003, EC_TIMEOUTRXM);
    ret_wr[7] = ec_SDOwrite(can_slave_ecNum, 0x8010, 0x00, TRUE, sizeof(_8010), &_8010, EC_TIMEOUTRXM);
    ret_wr[8] = ec_SDOwrite(can_slave_ecNum, 0x8016, 0x00, TRUE, sizeof(_8016), &_8016, EC_TIMEOUTRXM);
    ret_wr[9] = ec_SDOwrite(can_slave_ecNum, 0x8018, 0x00, TRUE, sizeof(_8018), &_8018, EC_TIMEOUTRXM);
    ret_wr[10] = ec_SDOwrite(can_slave_ecNum, 0x8013, 0x00, TRUE, sizeof(_8013), &_8013, EC_TIMEOUTRXM);

    ret_wr[11] = ec_SDOwrite(can_slave_ecNum, 0x1C12, 0x00, TRUE, sizeof(_1C12), &_1C12, EC_TIMEOUTRXM);
    ret_wr[12] = ec_SDOwrite(can_slave_ecNum, 0x1C13, 0x00, TRUE, sizeof(_1C13), &_1C13, EC_TIMEOUTRXM);
    
    From this code return value of ret_wr[0], ret_wr[1], ret_wr[11], and ret_wr[12] is 0
    
    Is there any solution or program to access EL6751 using SOEM?

ArthurKetels, 2024-10-09:
I am sorry, but you do not supply enough data to find out the issue. 

It could be that the timeout values of the ec_SDowrite() are too short. Always check if there is any error returned, and if so print it out on screen. Another issue could be that you try to write parameters in the wrong state (pre-OP safe-OP OP). 

A wireshark capture of your initialization sequence is also very helpful. 

The last screen-shot is in -offline- mode, so there is no way to check if these parameters are actually available in the online slave at the current state.

azzamwildan462, 2024-10-10:
@ArthurKetels sorry for late answer, i'm Azzam as a @hernanda16's friend, currently we develop Robot using Elmo driver motor (CANopen) that connect to Beckhoff EK1101 - EL6751, We plan to use SOEM to interract with beckhoff, from now on, we only have a knowledge about CANopen but only have a few knowledge about EtherCAT.   

I've follow the instruction from EL6751 Manual, it says going to PRE-OP than configure than going to safe-OP than OP. 

below is my Tshark show 
2257 0.255859608 Private_01:01:01 → Broadcast    ECAT 586 'FPRD': Len: 558, Adp 0x1004, Ado 0x1a2e, Wc 0
2258 0.256009763 03:01:01:01:01:01 → Broadcast    ECAT 586 'FPRD': Len: 558, Adp 0x1004, Ado 0x1a2e, Wc 1  Mbx(CoE SDO Res: Scs 2)
2259 0.256023332 Private_01:01:01 → Broadcast    ECAT 30 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 0
2260 0.256124929 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 1
2261 0.256129649 Private_01:01:01 → Broadcast    ECAT 29 'FPRD': Len: 1, Adp 0x1004, Ado 0x805, Wc 0
2262 0.256229560 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 1, Adp 0x1004, Ado 0x805, Wc 1
2263 0.256233780 Private_01:01:01 → Broadcast    ECAT 586 'FPWR': Len: 558, Adp 0x1004, Ado 0x1800, Wc 0  Mbx(CoE SDO Req : 'Initiate Download' (1) Idx=0x1c12 Sub=0)
2264 0.256383598 03:01:01:01:01:01 → Broadcast    ECAT 586 'FPWR': Len: 558, Adp 0x1004, Ado 0x1800, Wc 1  Mbx(CoE SDO Req : 'Initiate Download' (1) Idx=0x1c12 Sub=0)
2265 0.256388088 Private_01:01:01 → Broadcast    ECAT 30 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 0
2266 0.256490520 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 1
2267 0.256775853 Private_01:01:01 → Broadcast    ECAT 30 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 0
2268 0.256887397 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 1
2269 0.256903799 Private_01:01:01 → Broadcast    ECAT 586 'FPRD': Len: 558, Adp 0x1004, Ado 0x1a2e, Wc 0
2270 0.257054665 03:01:01:01:01:01 → Broadcast    ECAT 586 'FPRD': Len: 558, Adp 0x1004, Ado 0x1a2e, Wc 1  Mbx(CoE SDO Req : 'Abort Transfer' (4))
2271 0.257060615 Private_01:01:01 → Broadcast    ECAT 30 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 0
2272 0.257175494 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 1
2273 0.257179598 Private_01:01:01 → Broadcast    ECAT 29 'FPRD': Len: 1, Adp 0x1004, Ado 0x805, Wc 0
2274 0.257284018 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 1, Adp 0x1004, Ado 0x805, Wc 1
2275 0.257287610 Private_01:01:01 → Broadcast    ECAT 586 'FPWR': Len: 558, Adp 0x1004, Ado 0x1800, Wc 0  Mbx(CoE SDO Req : 'Initiate Download' (1) Idx=0x1c13 Sub=0)
2276 0.257437973 03:01:01:01:01:01 → Broadcast    ECAT 586 'FPWR': Len: 558, Adp 0x1004, Ado 0x1800, Wc 1  Mbx(CoE SDO Req : 'Initiate Download' (1) Idx=0x1c13 Sub=0)
2277 0.257442232 Private_01:01:01 → Broadcast    ECAT 30 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 0
2278 0.257545495 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 1
2279 0.257801572 Private_01:01:01 → Broadcast    ECAT 30 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 0
2280 0.257908804 03:01:01:01:01:01 → Broadcast    ECAT 60 'FPRD': Len: 2, Adp 0x1004, Ado 0x80d, Wc 1
2281 0.257916578 Private_01:01:01 → Broadcast    ECAT 586 'FPRD': Len: 558, Adp 0x1004, Ado 0x1a2e, Wc 0
2282 0.258068130 03:01:01:01:01:01 → Broadcast    ECAT 586 'FPRD': Len: 558, Adp 0x1004, Ado 0x1a2e, Wc 1  Mbx(CoE SDO Req : 'Abort Transfer' (4))
2283 0.258087775 Private_01:01:01 → Broadcast    ECAT 34 'FPRD': Len: 6, Adp 0x1004, Ado 0x130, Wc 0
2284

sorry for our little knowledge about SOEM, i hope you can help

azzamwildan462, 2024-10-10:
@ArthurKetels  Another question, why my EL6751 got "Can err with 1hz flashes" (indicate that At least one box state is not equal to 0 (e.g. device not found, wrong configuration, device in error state). But if i program and run it from TwinCAT, that led is off. 

Even i run it again by using my program using SOEM, everytime after i run it from TwinCAT the led is off. 

If i power-off my Beckhoff, the led flashes 1hz again. 

i thought it was TwinCAT wrote some value that my SOEM didn't. That values is not gone even the Beckhoff is re-INIT, but that values will be gone if the power is off. 

sorry, i and my friend so confused now :)

azzamwildan462, 2024-10-10:
update: now the return value of my ec_SDOwrite is 1 for all of my commands, but i can't change my EL6751 state from pre-op to safe-op

ArthurKetels, 2024-10-10:
When you want to post a tshark capture, please use the -w option to write the packets to a file in raw format. Zip it and post it here as attached file. Then I can analyse the packets in detail.

> can't change my EL6751 state from pre-op to safe-op

If you wait a few seconds, do you get an error on the EL6751?  Use something like below snippet:

```
ec_readstate();
for(i = 1; i<=ec_slavecount ; i++)
{
        printf("Slave %d State=0x%2.2x StatusCode=0x%4.4x : %s\n",
            i, ec_slave[i].state, ec_slave[i].ALstatuscode, ec_ALstatuscode2string(ec_slave[i].ALstatuscode));
}
```

> i thought it was TwinCAT wrote some value that my SOEM didn't.

Most likely. Use wireshark to capture the TwinCAT traffic to the slave in a minimal configuration. Do not forget to put TwinCAT in promiscuous mode, otherwise noting will be captured.

With bridge slaves like the EL6751 Beckhoff does some special configuration tricks that are not in the manual. They can be very difficult to get working correctly. If you search on that topic in this forum you will find examples.

ArthurKetels, 2024-10-10:
> update: now the return value of my ec_SDOwrite is 1 for all of my commands

Please explain what you did to solve this part of the issue. Then others, that read this thread later, also know how to do it.

azzamwildan462, 2024-10-10:
there is my error: 
Slave 4 State=0x12 StatusCode=0x001e : Invalid input configuration

how do i config my el6751? i was copy-paste similarly from the boot startup from TwinCAT about my EL6751

ArthurKetels, 2024-10-10:
Please use the search function as I requested. You will quickly find people that had the same issue and solved it.
I did it for you, and came up with #488

azzamwildan462, 2024-10-10:
update: 
i;ve successful communicate with my driver, i used the node scanning (refer to EL6751 manual on address 0xf002). 

but i have another question, how do i do SDO communication to my driver? Can i do the SDO from my EL6751? or can i write and send a raw CAN frame to my driver?

hernanda16, 2024-12-22:
I've done this problem by scanning configuration from TwinCat then without turning off the driver and beckhoff, using SOEM to save current configuration loaded from TwinCat to EEPROM.

closed 2024-12-22 by hernanda16