OpenEtherCATsociety/SOEM #867: SDO and PDO parallel execution

nkt17, 2024-10-25:
Good day everyone.
I need periodically request not mappable object by SDO. So the question occured. Can we parallel execute next actions?

1.
ec_send_processdata();
ec_receive_processdata(EC_TIMEOUTRET);

and
2.
ec_SDOread(1, index, subindex, FALSE, &size1, &v1, EC_TIMEOUTSAFE);

nkt17, 2024-10-25:
Also when i'm trying to request object by SDO sometimes there is incorrect data
[image]

ArthurKetels, 2024-10-25:
Yes, PDO and SDO can function perfectly next to each other. But make sure they are running in different threads. Also check the return value of any SDO function, they might fail. And when it fails check the error message that is stored in the error list.

nkt17, 2024-10-27:
> Yes, PDO and SDO can function perfectly next to each other. But make sure they are running in different threads. Also check the return value of any SDO function, they might fail. And when it fails check the error message that is stored in the error list.

Good day. Thanks for response.
I removed lock_guard:
[image]

I use this method to request parameters on the next photo (right values).
[image]

On removing lock_guard all the parameters became wrong (result on the next photo)
[image]

ArthurKetels, 2024-10-27:
OK, but did you do it the way I instructed? You do not show any of your code or a wireshark trace.