OpenEtherCATsociety/SOEM #868: Unable to transition to OPERATIONAL state

hshah44, 2024-10-28:
Hello - 

I have been at this issue for a few days not and appreciate all or any help. I am unable to transition the drive I am using to OPERATIONAL state. My goal (ATM) is to "rotate" the servo motor connected to the Servo drive.

I went through the suggestions in various issue like [#224](https://github.com/OpenEtherCATsociety/SOEM/issues/224).

Here's the info -

**Hardware**
Servo Drive: Inovance SV660NS1R61-FS-INT
Servo Motor: Inovance MS1H1-10B30CB-A330Z-INT

**slaveinfo output**
`SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on en7 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:InoSV660N
 Output size: 96bits
 Input size: 224bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00100000 ID: 000c010d Rev: 00010000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000c Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 06
 CoE details: 0d FoE details: 00 EoE details: 02 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program`

**Here's my main**

`if (ec_init(IFNAME)) {
    printf("Socket connection: %s\n", IFNAME);
} else {
    printf("No socket connection: %s\n", IFNAME);
    return 1;
}
    
if (ec_config_init(FALSE) > 0) {
    printf("EtherCAT slaves detected: %d\n", ec_slavecount);

    ec_config_map(&IOmap);
    ec_configdc();

    ec_slave[1].state = EC_STATE_OPERATIONAL;
    ec_writestate(1);
    ec_statecheck(1, EC_STATE_OPERATIONAL, EC_TIMEOUTSTATE);

    ec_close();

} else {
    printf("No EtherCAT slaves detected\n");
    return 1;
}`

Appreciate any pointers. Thank you!

ArthurKetels, 2024-10-28:
> Slave 1 State=0x14 StatusCode=0x0027 : Freerun not supported

I think this error message is very clear. You need to activate SYNC0 at the appropriate frequency, make a real-time thread that synchronizes to the reference slave. Just like in the red_test example. You can also read about synchronized PDO in many issues posted here.

Also I would encourage you to read some more about the workings and principles of EtherCAT. I guess you are missing some basics.

hshah44, 2024-10-28:
Thank you @ArthurKetels.

Could you suggest a clear and concise resource on the workings and principles of EtherCAT?

closed 2024-10-28 by hshah44

bao1ong, 2024-12-13:
Hello, may I ask if you have solved this problem? The problem I encountered was also with slave state=0x14 code=0x27. At first, I thought it was a problem with my pdo configuration that couldn't put the servo into op state. After reading many posts, I thought it might be a problem with sync0, but I am not sure how to fix and adjust it. Can you reply to me when you see this message? I have been stuck with this problem for a week now. Thank you