OpenEtherCATsociety/SOEM #871: 64-bit issue (malformed packages)

Frank-1964, 2024-11-06:
Hello,

I have a Beckhoff CX2030 with some EL-Cards and use SOEM 1.4.0 with this cards. This works fine with a 32-bit operating system (Linux with Kernel 3.18.140). I use the ec_bhf.ko kernel module.

When I switch to a 64-bit operating system (same hardware, Ubuntu 22.04 with 5.17.0 kernel), I can see some malformed packages (captured with tcpdump) and this results in further errors.

Here are the two tcpdump captures:
[tcpdump.zip](https://github.com/user-attachments/files/17646715/tcpdump.zip)

Any ideas?

If you need more infos, please let me know.

Thanks.

ArthurKetels, 2024-11-06:
The problem is that your driver strips the last 8 bytes of a received packet. 

If you have short packets (<=52 bytes) then they get padded by hardware to 60 bytes. This is the minimum size Ethernet packet. When the driver then strips the last 8 bytes SOEM does not notice because only padding bytes are stripped. But when there are more than 52 bytes in the packet then real bytes are stripped and the packet is invalid.

This stripping happens only in the 64bit kernel. The 32 bit version has the correct packet length.

Frank-1964, 2024-11-07:
@ArthurKetels Thanks for pushing me in the right direction.

I found in the kernel sources ec_bhf.c from the 3.18 kernel a correction for the calculation of the package size:

                int pkt_size = (le16_to_cpu(desc->header.len) &
//                             RXHDR_LEN_MASK) - sizeof(struct rx_header) - 4;
                               RXHDR_LEN_MASK) - sizeof(struct rx_header) + 4;

So I assume this is the cause of missing the 8 bytes.

I will take this change to the 5.17 kernel and I am shure my problem is gone.

Because this issue is obviously not soem caused I close this issue.

closed 2024-11-07 by Frank-1964