OpenEtherCATsociety/SOEM #872: red_test冗余测试ecx_waitinframe_red函数条件判断不完全

wmyqcy, 2024-11-08:
您好，我现在使用SOEM 1.4 遇见的问题跟你一样，我在linux系统上运行 linux\red_test.c，程序提示说没有找到从站，程序就停止了。我自己参考red_test.c编写了一个冗余程序，可以扫描到2个从站，电机也可以旋转，当我断开EtherCAT主站主网口或者从网口和从站的网线时，电机就停止旋转了，当我断开2个从站的网线，第1个从站就停止旋转了，第二个从站正常旋转，有时候现象不唯一，有可能第二个电机停止，是不是冗余功能有BUG？

我后面检测到
nicdrv.c文件ecx_waitinframe_red这个函数里面主网口数据和从网口数据的条件判断会不会缺少，
1、从网口断开，可以不用处理，使用主网口的数据即可。
2、主网口断开，从网口重新发送数据，将从网口接收的数据复制到主网口里面。
3、主网口接收到主网口数据，从网口接收到从网口数据，为什么需要将主网口的接收数据复制到主网口的发送数据里面？
我也尝试修改上面的逻辑，可是冗余功能还是存在问题。
static int ecx_waitinframe_red(ecx_portt *port, uint8 idx, osal_timert *timer)
{
	osal_timert timer2;
	int wkc = EC_NOFRAME;
	int wkc2 = EC_NOFRAME;
	int primrx, secrx;

	/* if not in redundant mode then always assume secondary is OK */
	if (port->redstate == ECT_RED_NONE)
		wkc2 = 0;
	do
	{
		/* only read frame if not already in */
		if (wkc <= EC_NOFRAME)
			wkc = ecx_inframe(port, idx, 0);
		/* only try secondary if in redundant mode */
		if (port->redstate != ECT_RED_NONE)
		{
			/* only read frame if not already in */
			if (wkc2 <= EC_NOFRAME)
				wkc2 = ecx_inframe(port, idx, 1);
		}
		/* wait for both frames to arrive or timeout */
	} while (((wkc <= EC_NOFRAME) || (wkc2 <= EC_NOFRAME)) && !osal_timer_is_expired(timer));
	/* only do redundant functions when in redundant mode */
	if (port->redstate != ECT_RED_NONE)
	{
		/* primrx if the received MAC source on primary socket */
		primrx = 0;
		if (wkc > EC_NOFRAME)
			primrx = port->rxsa[idx];
		/* secrx if the received MAC source on psecondary socket */
		secrx = 0;
		if (wkc2 > EC_NOFRAME)
			secrx = port->redport->rxsa[idx];

		/* primary socket got secondary frame and secondary socket got primary frame */
		/* normal situation in redundant mode */
		if (((primrx == RX_SEC) && (secrx == RX_PRIM)))
		{
			/* copy secondary buffer to primary */
			memcpy(&(port->rxbuf[idx]), &(port->redport->rxbuf[idx]), port->txbuflength[idx] - ETH_HEADERSIZE);
			wkc = wkc2;
		}
		/* primary socket got nothing or primary frame, and secondary socket got secondary frame */
		/* we need to resend TX packet */
		if (((primrx == 0) && (secrx == RX_SEC)) ||
			((primrx == RX_PRIM) && (secrx == RX_SEC)))
		{
			/* If both primary and secondary have partial connection retransmit the primary received
			 * frame over the secondary socket. The result from the secondary received frame is a combined
			 * frame that traversed all slaves in standard order. */
			if ((primrx == RX_PRIM) && (secrx == RX_SEC))
			{
				/* copy primary rx to tx buffer */
				memcpy(&(port->txbuf[idx][ETH_HEADERSIZE]), &(port->rxbuf[idx]), port->txbuflength[idx] - ETH_HEADERSIZE);
			}
			osal_timer_start(&timer2, EC_TIMEOUTRET);
			/* resend secondary tx */
			ecx_outframe(port, idx, 1);
			do
			{
				/* retrieve frame */
				wkc2 = ecx_inframe(port, idx, 1);
			} while ((wkc2 <= EC_NOFRAME) && !osal_timer_is_expired(&timer2));
			if (wkc2 > EC_NOFRAME)
			{
				/* copy secondary result to primary rx buffer */
				memcpy(&(port->rxbuf[idx]), &(port->redport->rxbuf[idx]), port->txbuflength[idx] - ETH_HEADERSIZE);
				wkc = wkc2;
			}
		}
	}

	/* return WKC or EC_NOFRAME */
	return wkc;
}

ArthurKetels, 2024-11-09:
Dear wmyqcy, this forum is english language only. Please repost accordingly.

toshisanro, 2024-11-13:
you can contact me  , I had the same problem and solved it. toshisanro@163.com

closed 2024-11-15 by ArthurKetels

wmyqcy, 2024-11-19:
> 你可以联系我，我遇到了同样的问题并解决了它。[toshisanro@163.com](mailto:toshisanro@163.com)

您好，我给您发邮箱了，付费咨询，里面有我的联系方式，麻烦看一下邮箱

HsinKaiLiao, 2025-02-04:
您好，請問可否請教您怎麼解決這個問題的，謝謝您