OpenEtherCATsociety/SOEM #874: Not going directly into operational mode

Rohithossain007, 2024-11-20:
Hi first of all I am new to this. I am using 2 motors Qrob70. Currently trying to get the motors into operational mode. However it is taking some time initially. Like lets say 5 or 10 secs to get the motor from safe op to operational (I am also using EtherCAT not CAN). Is there any solve to that meaning I want it directly to go to operational mode every time I start the code. Currently I am using this code: 

void transition_to_operational() {
    char IOmap[4096];

    if (ec_init("enp2s0")) { 
        printf("EtherCAT initialized on interface 'eth0'.\n");

        if (ec_config_init(FALSE) > 0) {
            printf("%d slaves found and configured.\n", ec_slavecount);

            if (ec_config_map(IOmap) > 0) {
                printf("Process data mapped successfully.\n");
            } else {
                printf("Failed to map process data.\n");
                ec_close();
                return;
            }

            ec_configdc();

            ec_slave[0].state = EC_STATE_PRE_OP;
            ec_writestate(0);
            ec_statecheck(0, EC_STATE_PRE_OP, EC_TIMEOUTSTATE);
            printf("Slave moved to PRE-OPERATIONAL state.\n");

            //ec_slave[0].state = EC_STATE_SAFE_OP;
            //ec_writestate(0);
            //ec_statecheck(0, EC_STATE_SAFE_OP, EC_TIMEOUTSTATE);
            //printf("Slave moved to SAFE-OPERATIONAL state.\n");

            printf("Switching on the motor...\n");
            set_control_word(0x0006, 0);
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            usleep(10000);

            set_control_word(0x0007, 0);
            ec_send_processdata();
            ec_receive_processdata(EC_TIMEOUTRET);
            usleep(10000);

            ec_slave[0].state = EC_STATE_OPERATIONAL;
            ec_writestate(0);
            ec_statecheck(0, EC_STATE_OPERATIONAL, EC_TIMEOUTSTATE);

            if (ec_slave[0].state == EC_STATE_OPERATIONAL) {
                printf("Slave is now in OPERATIONAL state.\n");
            } else {
                printf("Failed to move slave to OPERATIONAL state.\n");
            }
        } else {
            printf("No slaves found!\n");
        }
        ec_close();
    } else {
        printf("Failed to initialize EtherCAT.\n");
    }
}

Any advises would be appreciated.

ZeroErrControl, 2024-11-28:
You can refer to this link: [https://github.com/ZeroErrControl/eRob_SOEM_linux/blob/main/test/linux/simple_test/eRob_test.cpp](https://github.com/ZeroErrControl/eRob_SOEM_linux/blob/main/test/linux/simple_test/eRob_test.cpp). I hope it will be helpful to you.

Rohithossain007, 2024-12-02:
> You can refer to this link: https://github.com/ZeroErrControl/eRob_SOEM_linux/blob/main/test/linux/simple_test/eRob_test.cpp. I hope it will be helpful to you.

this one does not start the motor. Do you have anything that will start the motor in torque or position mode? I am using qrob 70.

ZeroErrControl, 2024-12-02:
> qrob 70

Sorry, Qrob is not our product; our product is eRob. If this program cannot resolve the issue with starting the motor, you may need to contact the manufacturer or provide more detailed error information to effectively address the problem. Wishing you success!

Rohithossain007, 2024-12-02:
> > qrob 70
> 
> Sorry, Qrob is not our product; our product is eRob. If this program cannot resolve the issue with starting the motor, you may need to contact the manufacturer or provide more detailed error information to effectively address the problem. Wishing you success!

my apologies I meant Erob 70

ZeroErrControl, 2024-12-02:
> > > qrob 70
> > 
> > 
> > Sorry, Qrob is not our product; our product is eRob. If this program cannot resolve the issue with starting the motor, you may need to contact the manufacturer or provide more detailed error information to effectively address the problem. Wishing you success!
> 
> my apologies I meant Erob 70

Thank you very much for your feedback! This program has been tested and can successfully start eRob. However, you may need to add sudo to run it with administrator privileges each time before starting the program. Additionally, you need to modify the relevant network interface name variable in the program. If the initialization fails and gets stuck at step 8, try closing and rerunning the program. Once it successfully starts, it should run smoothly every time thereafter.

If the issue persists, please provide more detailed error logs in the GitHub project repository, and we will do our best to address the problem based on your feedback.

Rohithossain007, 2024-12-02:
> Thank you very much for your feedback! This program has been tested and can successfully start eRob. However, you may need to add sudo to run it with administrator privileges each time before [...]

Thank you very much

Rohithossain007, 2024-12-04:
closing this one as the code provided by zeroerrcontrol solved the issue.

closed 2024-12-04 by Rohithossain007