OpenEtherCATsociety/SOEM #876: Understanding the output of `slaveinfo -map` and `slaveinfo -sdo`

brunogold, 2024-12-13:
I’ve been working on getting  ECS to the OPERATIONAL state for a couple of months now. I believe I have a good enough understanding of the fundamentals, but I’m struggling to make sense of the output below. It feels like it could be a firmware issue, though I’m not certain. Any suggestions on what steps I should take next? Here's the output for couple flags from slaveinfo.

Output of sudo ./slaveinfo en0 -map

`SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enx50a0300e0736 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:InoSV660N
 Output size: 96bits
 Input size: 224bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00100000 ID: 000c010d Rev: 00010000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000c Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 06
 CoE details: 0d FoE details: 00 EoE details: 02 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6040:0x00 0x10
  [0x0002.0] 0x607A:0x00 0x20
  [0x0006.0] 0x60B8:0x00 0x10
  [0x0008.0] 0x60FE:0x01 0x20
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x000C.0] 0x603F:0x00 0x10
  [0x000E.0] 0x6041:0x00 0x10
  [0x0010.0] 0x6064:0x00 0x20
  [0x0014.0] 0x0000:0x00 0x00
  [0x0014.0] 0x6077:0x00 0x10
  [0x0016.0] 0x0000:0x00 0x00
  [0x0016.0] 0x60B9:0x00 0x10
  [0x0018.0] 0x0000:0x00 0x00
  [0x0018.0] 0x60BC:0x00 0x20
End slaveinfo, close socket
End program
`

Output of sudo ./slaveinfo en0 -sdo

`SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enx50a0300e0736 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:InoSV660N
 Output size: 96bits
 Input size: 224bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00100000 ID: 000c010d Rev: 00010000
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:  12 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000000c Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 06
 CoE details: 0d FoE details: 00 EoE details: 02 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
Time:1734075398.040 MBX slave:1 error:0004 The service is no supported
Time:1734075398.041 MBX slave:1 error:0004 The service is no supported
Time:1734075398.042 MBX slave:1 error:0004 The service is no supported
Time:1734075398.043 MBX slave:1 error:0004 The service is no supported
Time:1734075398.044 MBX slave:1 error:0004 The service is no supported
Time:1734075398.045 MBX slave:1 error:0004 The service is no supported
Time:1734075398.046 MBX slave:1 error:0004 The service is no supported
Time:1734075398.047 MBX slave:1 error:0004 The service is no supported
Time:1734075398.048 MBX slave:1 error:0004 The service is no supported
Time:1734075398.049 MBX slave:1 error:0004 The service is no supported
Time:1734075398.050 MBX slave:1 error:0004 The service is no supported
Time:1734075398.051 MBX slave:1 error:0004 The service is no supported
Time:1734075398.052 MBX slave:1 error:0004 The service is no supported
Time:1734075398.053 MBX slave:1 error:0004 The service is no supported
Time:1734075398.054 MBX slave:1 error:0004 The service is no supported
Time:1734075398.055 MBX slave:1 error:0004 The service is no supported
Time:1734075398.056 MBX slave:1 error:0004 The service is no supported
Time:1734075398.057 MBX slave:1 error:0004 The service is no supported
Time:1734075398.058 MBX slave:1 error:0004 The service is no supported
Time:1734075398.059 MBX slave:1 error:0004 The service is no supported
Time:1734075398.060 MBX slave:1 error:0004 The service is no supported
Time:1734075398.061 MBX slave:1 error:0004 The service is no supported
Time:1734075398.062 MBX slave:1 error:0004 The service is no supported
Time:1734075398.063 MBX slave:1 error:0004 The service is no supported
Time:1734075398.064 MBX slave:1 error:0004 The service is no supported
Time:1734075398.065 MBX slave:1 error:0004 The service is no supported
Time:1734075398.066 MBX slave:1 error:0004 The service is no supported
Time:1734075398.067 MBX slave:1 error:0004 The service is no supported
Time:1734075398.068 MBX slave:1 error:0004 The service is no supported
Time:1734075398.069 MBX slave:1 error:0004 The service is no supported
Time:1734075398.070 MBX slave:1 error:0004 The service is no supported
Time:1734075398.071 MBX slave:1 error:0004 The service is no supported
Time:1734075398.072 MBX slave:1 error:0004 The service is no supported
Time:1734075398.073 MBX slave:1 error:0004 The service is no supported
Time:1734075398.074 MBX slave:1 error:0004 The service is no supported
Time:1734075398.075 MBX slave:1 error:0004 The service is no supported
Time:1734075398.076 MBX slave:1 error:0004 The service is no supported
Time:1734075398.077 MBX slave:1 error:0004 The service is no supported
Time:1734075398.078 MBX slave:1 error:0004 The service is no supported
Time:1734075398.079 MBX slave:1 error:0004 The service is no supported
Time:1734075398.080 MBX slave:1 error:0004 The service is no supported
Time:1734075398.081 MBX slave:1 error:0004 The service is no supported
Time:1734075398.082 MBX slave:1 error:0004 The service is no supported
Time:1734075398.083 MBX slave:1 error:0004 The service is no supported
Time:1734075398.084 MBX slave:1 error:0004 The service is no supported
Time:1734075398.085 MBX slave:1 error:0004 The service is no supported
Time:1734075398.086 MBX slave:1 error:0004 The service is no supported
Time:1734075398.087 MBX slave:1 error:0004 The service is no supported
Time:1734075398.088 MBX slave:1 error:0004 The service is no supported
Time:1734075398.089 MBX slave:1 error:0004 The service is no supported
Time:1734075398.090 MBX slave:1 error:0004 The service is no supported
Time:1734075398.091 MBX slave:1 error:0004 The service is no supported
Time:1734075398.092 MBX slave:1 error:0004 The service is no supported
Time:1734075398.093 MBX slave:1 error:0004 The service is no supported
Time:1734075398.094 MBX slave:1 error:0004 The service is no supported
Time:1734075398.095 MBX slave:1 error:0004 The service is no supported
Time:1734075398.096 MBX slave:1 error:0004 The service is no supported
Time:1734075398.097 MBX slave:1 error:0004 The service is no supported
Time:1734075398.098 MBX slave:1 error:0004 The service is no supported
Time:1734075398.099 MBX slave:1 error:0004 The service is no supported
Time:1734075398.100 MBX slave:1 error:0004 The service is no supported
Time:1734075398.101 MBX slave:1 error:0004 The service is no supported
Time:1734075398.102 MBX slave:1 error:0004 The service is no supported
Time:1734075398.103 MBX slave:1 error:0004 The service is no supported
End slaveinfo, close socket
End program
`

dennislandre, 2025-01-03:
I have the same issue but with ..-sdo:

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{19F186BC-DFB1-4547-9DB6-BEFD9DA7FA4E} succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:mcDSA-E4-EtherCAT
 Output size: 16bits
 Input size: 16bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000554 ID: 23240031 Rev: 00000001
 SM0 A:1200 L:  32 F:00010026 Type:1
 SM1 A:1220 L:  32 F:00010022 Type:2
 SM2 A:1000 L:   2 F:00010064 Type:3
 SM3 A:1100 L:   2 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000002 Ll:   2 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 32 rd: 32 MBX protocols : 04
 CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
Time:1735904107.932 MBX slave:1 error:0004 The service is no supported
Time:1735904107.933 MBX slave:1 error:0004 The service is no supported
Time:1735904107.933 MBX slave:1 error:0004 The service is no supported
Time:1735904107.934 MBX slave:1 error:0004 The service is no supported
Time:1735904107.934 MBX slave:1 error:0004 The service is no supported
Time:1735904107.935 MBX slave:1 error:0004 The service is no supported
Time:1735904107.936 MBX slave:1 error:0004 The service is no supported
Time:1735904107.937 MBX slave:1 error:0004 The service is no supported
Time:1735904107.937 MBX slave:1 error:0004 The service is no supported
Time:1735904107.939 MBX slave:1 error:0004 The service is no supported
Time:1735904107.940 MBX slave:1 error:0004 The service is no supported
Time:1735904107.940 MBX slave:1 error:0004 The service is no supported
Time:1735904107.941 MBX slave:1 error:0004 The service is no supported
Time:1735904107.942 MBX slave:1 error:0004 The service is no supported
Time:1735904107.943 MBX slave:1 error:0004 The service is no supported
Time:1735904107.943 MBX slave:1 error:0004 The service is no supported
Time:1735904107.944 MBX slave:1 error:0004 The service is no supported
Time:1735904107.945 MBX slave:1 error:0004 The service is no supported
Time:1735904107.946 MBX slave:1 error:0004 The service is no supported
Time:1735904107.946 MBX slave:1 error:0004 The service is no supported
Time:1735904107.947 MBX slave:1 error:0004 The service is no supported
Time:1735904107.948 MBX slave:1 error:0004 The service is no supported
Time:1735904107.948 MBX slave:1 error:0004 The service is no supported
Time:1735904107.949 MBX slave:1 error:0004 The service is no supported
Time:1735904107.949 MBX slave:1 error:0004 The service is no supported
Time:1735904107.950 MBX slave:1 error:0004 The service is no supported
Time:1735904107.951 MBX slave:1 error:0004 The service is no supported
Time:1735904107.951 MBX slave:1 error:0004 The service is no supported
Time:1735904107.952 MBX slave:1 error:0004 The service is no supported
Time:1735904107.953 MBX slave:1 error:0004 The service is no supported
Time:1735904107.953 MBX slave:1 error:0004 The service is no supported
Time:1735904107.954 MBX slave:1 error:0004 The service is no supported
Time:1735904107.955 MBX slave:1 error:0004 The service is no supported
Time:1735904107.956 MBX slave:1 error:0004 The service is no supported
Time:1735904107.957 MBX slave:1 error:0004 The service is no supported
Time:1735904107.958 MBX slave:1 error:0004 The service is no supported
Time:1735904107.958 MBX slave:1 error:0004 The service is no supported
Time:1735904107.959 MBX slave:1 error:0004 The service is no supported
Time:1735904107.960 MBX slave:1 error:0004 The service is no supported
Time:1735904107.961 MBX slave:1 error:0004 The service is no supported
Time:1735904107.961 MBX slave:1 error:0004 The service is no supported
Time:1735904107.962 MBX slave:1 error:0004 The service is no supported
Time:1735904107.963 MBX slave:1 error:0004 The service is no supported
Time:1735904107.964 MBX slave:1 error:0004 The service is no supported
Time:1735904107.964 MBX slave:1 error:0004 The service is no supported
Time:1735904107.965 MBX slave:1 error:0004 The service is no supported
Time:1735904107.966 MBX slave:1 error:0004 The service is no supported
Time:1735904107.967 MBX slave:1 error:0004 The service is no supported
Time:1735904107.968 MBX slave:1 error:0004 The service is no supported
Time:1735904107.969 MBX slave:1 error:0004 The service is no supported
Time:1735904107.970 MBX slave:1 error:0004 The service is no supported
Time:1735904107.970 MBX slave:1 error:0004 The service is no supported
Time:1735904107.971 MBX slave:1 error:0004 The service is no supported
Time:1735904107.972 MBX slave:1 error:0004 The service is no supported
Time:1735904107.973 MBX slave:1 error:0004 The service is no supported
Time:1735904107.974 MBX slave:1 error:0004 The service is no supported
Time:1735904107.975 MBX slave:1 error:0004 The service is no supported
Time:1735904107.975 MBX slave:1 error:0004 The service is no supported
Time:1735904107.976 MBX slave:1 error:0004 The service is no supported
Time:1735904107.977 MBX slave:1 error:0004 The service is no supported
Time:1735904107.978 MBX slave:1 error:0004 The service is no supported
Time:1735904107.978 MBX slave:1 error:0004 The service is no supported
Time:1735904107.979 MBX slave:1 error:0004 The service is no supported
Time:1735904107.980 MBX slave:1 error:0004 The service is no supported
End slaveinfo, close socket
End program

ArthurKetels, 2025-01-06:
From the data of your slave:
`CoE details: 0d`

Bit 1 of CoE details is not set. This means that SDO info is not supported. In other words there is no online SDO dictionary available. The result is that when SOEM tries to access the dictionary to show a meaningful text of the mapped PDO objects it fails.

This is also what the error message says : 

> error:0004 The service is no supported

So talk to your slave vendor if you want to have this feature implemented. This is nothing SOEM can help with.