OpenEtherCATsociety/SOEM #878: Use of SM/FMMU 6&7 for Process Data

Guistac, 2024-12-15:
Hello,

I'm encountering an issue with a Beckhoff EtherCat device (ELM7231-9016)
This device calls for the mandatory use of SM 6&7 for transmission of FSOE PDUs (as per the .xml datasheet)
Though its regular process data is still expected to be exchanged through SM 2&3.

By selecting my PDOs through objects x1C12 x1C13 x1C16 x1C17 and also setting the correct Sync0 and Sync1 times, I am able to bring my device into Operational State. SOEM correctly reports the configured process data length to be the sum of all four SyncManager sizes, and process data is written/read successfully for SM 2&3. But process data sent to SM 6&7 is not received by the device, and the data received from the slave seems to be misaligned or faulty (though it is consistent and not random)

My understanding of how SOEM works is still a bit limited, but I have the impression that the library does not configure FMMUs above the first 4, which would cause process data for SM6&7 to not be mapped to any particular slave memory.

Could you give me some pointers on how and where I could implement a workaround ?

Thank you !

Guistac, 2024-12-15:
After some digging I found out that SOEM actually tries to configure FMMU[1,3] for use with SM[6,7].
The data lengths are correct, but it seems that the Physical Start Addresses are mismatched.

[image]

After looking at the ESI file again, SOEM actually takes default values for SM[4,5] instead of those from SM[6,7].
FMMU[1,3] get configured with physical Start Addresses 0x1A00 and 0x1C00 instead of 0x1E00 and 0x1F00.
This is surprising since SM[4,5] should not be enabled by default, but it would explain the misaligned data.

Is there some patch I can apply to the library to fix this or is it better to somehow manually override this in my startup sequence?

ArthurKetels, 2024-12-17:
Could you show the output of slaveinfo -map to see what the information given by the slave is?

Guistac, 2024-12-18:
Hi ! Of course, here's the output after a power cycle, with none of my configuration of the process data in SM6,7

> Slave:2
 Name:ELM7231-9016
 Output size: 48bits
 Input size: 80bits
 State: 18
 Delay: 145[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 502275f9 Rev: 00122338
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1100 L: 256 F:00010022 Type:2
 SM2 A:1200 L:   6 F:00010024 Type:3
 SM3 A:1600 L:  10 F:00010020 Type:4
 SM4 A:1a00 L:   0 F:00000024 Type:3
 SM5 A:1c00 L:   0 F:00000020 Type:4
 SM6 A:1e00 L:  10 F:00010024 Type:3
 SM7 A:1f00 L:  10 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1200 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000015 Ll:  10 Lsb:0 Leb:7 Ps:1600 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:1
 MBX length wr: 256 rd: 256 MBX protocols : 0c
 CoE details: 2f FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 160[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7010:0x01 0x10 UNSIGNED16   Controlword
  [0x0002.0] 0x7010:0x05 0x20 UNSIGNED32   Target position
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0015.0] 0x6000:0x11 0x20 UNSIGNED32   Position
  [0x0019.0] 0x6010:0x01 0x10 UNSIGNED16   Statusword
  [0x001B.0] 0x6010:0x06 0x20 INTEGER32    Following error actual value
  SM4 outputs
     addr b   index: sub bitl data_type    name
  SM5 inputs
     addr b   index: sub bitl data_type    name
  SM6 outputs
     addr b   index: sub bitl data_type    name
  SM7 inputs
     addr b   index: sub bitl data_type    name

Guistac, 2025-02-25:
Hi,

I managed to get my device working by correcting the FMMU parameters manually after faulty some configuration.
I'm running the following after executing ec_config_map()
The device now accepts and replies with the correct FSOE data.

```
void ELM7231_9016::fixFMMUs(){
    //correct FMMU[1,3] physical start address for SM[6,7]
    uint16_t FMMU1_physicalStartAddress = 0x1E00;
    uint16_t FMMU3_physicalStartAddress = 0x1F00;
    ec_FPWR(identity->configadr, 0x618, 2, &FMMU1_physicalStartAddress, EC_TIMEOUTSAFE);
    ec_FPWR(identity->configadr, 0x638, 2, &FMMU3_physicalStartAddress, EC_TIMEOUTSAFE);
}
```

closed 2025-11-05 by Guistac