OpenEtherCATsociety/SOEM #883: May I ask if the SOEM main station supports configuring different DC time

brentonchris, 2025-02-11:
May I ask if the SOEM main station supports configuring different DC time?

ArthurKetels, 2025-02-13:
Yes and no. EtherCAT protocol assumes that all slaves on the same link have the same clock reference. That is why there is the read-multiple-write command that reads the clock of the reference slave and writes it to all following slaves (and thus  synchronizes the slaves). This is also the default method used by SOEM. 

But, because you are not bound by the default solution of SOEM, and can build your own synchronization mechanism using the SOEM tools, you can in fact have arbitrary clock synchronization for each slave (and master). It works less optimal, but is doable. I have build proof-of-concept code in the past that shows this working.

If you send clock synchronization packets from the master to a specific slave it will correct its internal clock. Just make sure that the timestamp of the master packet is enough in the past or future in relation to the slave internal clock.

brentonchris, 2025-02-14:
if there are any examples on SOEM that support different DC clocks for reference?

ArthurKetels, 2025-02-14:
Not that I know of. Those solutions are so specific that they would be next to useless for the general case.

It would help if you can explain what is your actual problem you like to solve.

brentonchris, 2025-02-24:
Thank you very much, using two groups can solve my problem.

closed 2025-02-24 by brentonchris