OpenEtherCATsociety/SOEM #884: Unable to transition to SAFEOP state

JianMX, 2025-02-13:
Hi,
This is my first time using SOEM, and I hope I can get some help here.
My slave is unable to transition to the SAFEOP state. Debugging shows that **Slave 1 State = 12 and StatusCode = 70: Unknown.**
There is a strange phenomenon: if I first run it with TwinCAT, it works fine with SOEM afterward. However, if I restart with a breakpoint, it no longer works.
I have attached the Wireshark file and XML file.

https://drive.google.com/file/d/14HknlOFZlrpKW_pcMZ3e_Qr82K-sA3Re/view?usp=drive_link

https://drive.google.com/file/d/1pj5xYn4Gdsg1_9req-btaycTt-ubqpNw/view?usp=drive_link

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{F02B4208-BC62-453C-BF06-521939C7A538} succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  70 : Unknown

Slave:1
 Name:EX-1100
 Output size: 64bits
 Input size: 144bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00444543 ID: 00000001 Rev: 00010001
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1200 L: 512 F:00010022 Type:2
 SM2 A:1400 L:   8 F:00010064 Type:3
 SM3 A:1a00 L:  18 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:  18 Lsb:0 Leb:7 Ps:1a00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Channel 1
  [0x0000.1] 0x7000:0x02 0x01 BOOLEAN      Channel 2
  [0x0000.2] 0x7000:0x03 0x01 BOOLEAN      Channel 3
  [0x0000.3] 0x7000:0x04 0x01 BOOLEAN      Channel 4
  [0x0000.4] 0x7000:0x05 0x01 BOOLEAN      Channel 5
  [0x0000.5] 0x7000:0x06 0x01 BOOLEAN      Channel 6
  [0x0000.6] 0x7000:0x07 0x01 BOOLEAN      Channel 7
  [0x0000.7] 0x7000:0x08 0x01 BOOLEAN      Channel 8
  [0x0001.0] 0x7000:0x09 0x01 BOOLEAN      Channel 9
  [0x0001.1] 0x7000:0x0A 0x01 BOOLEAN      Channel 10
  [0x0001.2] 0x7000:0x0B 0x01 BOOLEAN      Channel 11
  [0x0001.3] 0x7000:0x0C 0x01 BOOLEAN      Channel 12
  [0x0001.4] 0x7000:0x0D 0x01 BOOLEAN      Channel 13
  [0x0001.5] 0x7000:0x0E 0x01 BOOLEAN      Channel 14
  [0x0001.6] 0x7000:0x0F 0x01 BOOLEAN      Channel 15
  [0x0001.7] 0x7000:0x10 0x01 BOOLEAN      Channel 16
  [0x0002.0] 0x7020:0x01 0x01 BOOLEAN      Clear Count Value Of Channel 1
  [0x0002.1] 0x7020:0x02 0x01 BOOLEAN      Disable Z Phase Of Channel 1
  [0x0002.2] 0x7020:0x03 0x01 BOOLEAN      Count Direction Of Channel 1
  [0x0002.3] 0x7020:0x04 0x01 BOOLEAN      Enable Channel 1
  [0x0002.4] 0x7020:0x05 0x01 BOOLEAN      Enable Latch Of Channel 1
  [0x0002.5] 0x7021:0x01 0x20 UNSIGNED32   Initial Count Value
  [0x0006.5] 0x0000:0x00 0x0B
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0008.0] 0x6002:0x01 0x01 BOOLEAN      Temperature Alarm
  [0x0008.1] 0x6002:0x02 0x01 BOOLEAN      Voltage Alarm
  [0x0008.2] 0x6002:0x03 0x0E dt:0x0000 (14) SubIndex 003
  [0x000A.0] 0x6010:0x01 0x01 BOOLEAN      Channel 1
  [0x000A.1] 0x6010:0x02 0x01 BOOLEAN      Channel 2
  [0x000A.2] 0x6010:0x03 0x01 BOOLEAN      Channel 3
  [0x000A.3] 0x6010:0x04 0x01 BOOLEAN      Channel 4
  [0x000A.4] 0x6010:0x05 0x01 BOOLEAN      Channel 5
  [0x000A.5] 0x6010:0x06 0x01 BOOLEAN      Channel 6
  [0x000A.6] 0x6010:0x07 0x01 BOOLEAN      Channel 7
  [0x000A.7] 0x6010:0x08 0x01 BOOLEAN      Channel 8
  [0x000B.0] 0x6010:0x09 0x01 BOOLEAN      Channel 9
  [0x000B.1] 0x6010:0x0A 0x01 BOOLEAN      Channel 10
  [0x000B.2] 0x6010:0x0B 0x01 BOOLEAN      Channel 11
  [0x000B.3] 0x6010:0x0C 0x01 BOOLEAN      Channel 12
  [0x000B.4] 0x6010:0x0D 0x01 BOOLEAN      Channel 13
  [0x000B.5] 0x6010:0x0E 0x01 BOOLEAN      Channel 14
  [0x000B.6] 0x6010:0x0F 0x01 BOOLEAN      Channel 15
  [0x000B.7] 0x6010:0x10 0x01 BOOLEAN      Channel 16
  [0x000C.0] 0x6012:0x11 0x01
  [0x000C.1] 0x6012:0x12 0x01
  [0x000C.2] 0x6012:0x03 0x0E dt:0x0000 (14) SubIndex 003
  [0x000E.0] 0x6020:0x01 0x01 BOOLEAN      Latch Signal Of Channel 1
  [0x000E.1] 0x6020:0x02 0x0F dt:0x0000 (15) SubIndex 002
  [0x0010.0] 0x6021:0x03 0x20
  [0x0014.0] 0x6021:0x04 0x20
  [0x0018.0] 0x6022:0x05 0x01
  [0x0018.1] 0x6022:0x06 0x01
  [0x0018.2] 0x6022:0x03 0x0E dt:0x0000 (14) SubIndex 003
End slaveinfo, close socket
End program

ArthurKetels, 2025-02-13:
Sorry, I can not access the google drive. From the slave info text output I can not find anything wrong.

JianMX, 2025-02-13:
> 抱歉，我无法访问 google drive。从 slave info 文本输出中，我找不到任何错误。

The slave’s supplier gave me the following information, but I don’t know how to handle it in SOEM.

Detected Module Ident List (0xF030) and Configured Module Ident List (0xF050) do not match

This involves the Slot parameters. On the master side, check if the 0xF030 index is not configured.

I'm not sure how the master is handling this. After powering on the slave, these two indices should be filled with the corresponding module IDs. As for how the master handles these two indices, we need to look at the master’s code or capture packets with TwinCAT to compare. This will help you check if your upper-level machine is missing steps related to these two indices.

zhuhuzhenjing, 2026-08-19:
会不会是从站index-subindex的字节超过了从站所能容纳的上限，我也遇到了这个问题