OpenEtherCATsociety/SOEM #885: How to perform both PDO and SDO communication simultaneously without affecting the PDO communication thread cycle

Deft-C, 2025-02-17:
Hi,
I am currently using SOEM for ecat communication testing, and my requirement is for SOEM to run with a 1ms cycle. It performed well in the initial tests, but I have a thread that needs to periodically read and write some SDOs. After adding this thread, the ecat thread gets blocked.I noticed that SDO read and write operations are blocking. Do you have a proper method to prevent the ecat thread from being blocked due to SDO operations?

Deft-C, 2025-02-17:
Currently, it seems that the blocking is caused by ecx_waitinframe_red. Both PDO and SDO use this function to wait for data to arrive.

ArthurKetels, 2025-02-17:
SOEM supports multiple threads. So you should be able to run PDO in one thread and SDO in another, while not blocking each other. However this depends on the performance of scheduler. What OS are you using, and what is the scheduler policy? SOEM works best with real time scheduler, and the PDO thread should have a higher priority than the SDO thread..

Deft-C, 2025-02-17:
> SOEM supports multiple threads. So you should be able to run PDO in one thread and SDO in another, while not blocking each other. However this depends on the performance of scheduler. What OS are [...]

Thank you for your detailed reply. I am using a Linux system with a real-time kernel, and I created a real-time thread for PDO communication using osal_thread_create_rt. I also modified the sched_priority within this function to a higher value. For the SDO communication thread, I used osal_thread_create to create it.

I checked the thread priorities, and the PDO thread's priority is clearly higher than the SDO thread. From the Wireshark data, I can see that when there is jitter in the PDO communication, the first few data frames are related to SDO. Additionally, from the system call interrupt information, it seems that the SDO thread interrupts are affecting the PDO thread.

Deft-C, 2025-02-18:
I think this issue is similar to https://github.com/OpenEtherCATsociety/SOEM/issues/604
Make the SDO write/read non-blocking would be very useful
Has there been any progress on this work so far?

Kind regards