OpenEtherCATsociety/SOEM #886: Does SOEM support gigabit networks?

wmyqcy, 2025-02-18:
Does SOEM support gigabit networks?
EtherCAT official website has EtherCAT G, but I'm not sure if SOEM supports EtherCAT G. If it doesn't, what modifications are needed to support EtherCAT G?

ArthurKetels, 2025-02-18:
SOEM is agnostic for the physical transport layer. So using EtherCAT protocol over 1Gb to 10Gb lines is possible. However I have not tried this yet because lack of suitable devices to test.

wmyqcy, 2025-02-19:
> SOEM 与物理传输层无关。因此，可以在 1Gb 到 10Gb 线路上使用 EtherCAT 协议。但是，我还没有尝试过，因为缺乏合适的设备来测试。

I have also asked many manufacturers, and EtherCAT supports 100 Mbps networks.

I'm not sure if any manufacturers' EtherCAT slaves support Gigabit or 10 Gigabit.

wmyqcy, 2025-02-19:
Does anyone know which manufacturers' EtherCAT slaves support Gigabit or 10 Gigabit?

ArthurKetels, 2025-02-19:
Beckhoff has multiple EtherCAT products with 1Gb ports; EK1400, CU1411, FB1400, etc. Just look at their website.

ArthurKetels, 2025-02-19:
I do have a question for you, what is the reason you need 1Gb speeds? 

I have done many projects with EtherCAT, and never had issues with bandwidth. For industrial applications there are also serious disadvantages using higher bandwidth communications. At Beckhoff their main use case seems to be to have a fast back-bone splitting to groups of standard speed EtherCAT branches.