OpenEtherCATsociety/SOEM #888: DC Sync operation still not quite clear

madhu-dhanasekaran, 2025-02-20:
Dear SOEM community,

I am trying to setup an EtherCAT network with SOEM as master and a couple of slaves (EasyCAT PRO in my case with LAN 9252 ESC). The slaves are to be synchronized with the master and perform some operation (ex: digitally trigger a camera) at a specific EtherCAT timestamp. Inversely, be also able to timestamp when the camera captures an image and send it to the master using the LATCH0 signal.
 
For achieving this, I split my tasks this way:
NOTE: For simplicity, I am starting with just one slave for now
1. Setup an EtherCAT network with master and slave to communicate with each other (exchanging inputs and outputs) - Verified
2. Perform DC sync to achieve synchronization between master and slaves - In Progress
3. Use SYNC to enable camera trigger 
4. Use LATCH to timestamp when a camera is triggered

Now that in my step 2, I am trying to figure out how this is exactly done. I spent a few days to actually go through the issues to make out some sense about the DC synchronization and tried to use red_test.c application as it is intended to be by looking at different related issues. But, still I am missing some part of it or I feel the application is not yet fully helping to achieve DC sync. Please correct me if I am wrong.

So, let me put here, what I have tried so far. 
1. Configuring and mapping the slave (ec_config)
2. Configuring DC (ec_configdc)
3. Transition to SAFE_OP
4. Do state check - Wait for all states to reach OP
5a. Call ec_dcsync on the slave (in the main thread)
5b. Check for difference timer to verify if clocks are synchronized. I used FP_RD function for this and I always get ret val to be <= 0. So, could not clearly verify if the sync is happening.
6. Perfom PDO transfer (happens in parallel in the ethercat thread)

Can you maybe already spot what I am doing wrong here? For some information about the slave, please find the slave_info output here:
[slave_info.txt](https://github.com/user-attachments/files/18891396/slave_info.txt)

I am also trying to use U-Phy with STM32 Nucleo F446ZE as an EtherCAT slave. If there are already any examples on how to perform DC sync on this device those solutions also should be handy. 

Looking forward to your reply. Thank you!

Best regards
Madhu

madhu-dhanasekaran, 2025-02-26:
Please find the Wireshark captures here:
[Capture.zip](https://github.com/user-attachments/files/18982239/Capture.zip)

ArthurKetels, 2025-02-27:
Your wireshark trace stops after your issue the ec_dcsync0() command. I do not know if this is intentional. Up to that point everything seem to be ok. Slave is synchronized and the sync0 unit is programmed and activated. 

But did you mean to have a sync0 cycle time of 8us? That is very fast.

madhu-dhanasekaran, 2025-02-27:
Hi @ArthurKetels

Thanks for your response. Firstly, I want to verify if my DC sync is working fine. So that I can raise a signal on Sync0 to control my digital outputs. Can you please tell me how can I verify it? Because, reading the SYS TIME DIFF register did not help in my case.

Regarding your question on sync0 cycle time, I just followed the red_test example values. I can reduce it, my purpose is just to trigger digital output at a specific time and then turn it off.

ArthurKetels, 2025-02-27:
If you want to check sync pulse generation you have to measure the output on the sync pin directly with an oscilloscope, The pulse length is defined in the eeprom setting of the ESC, so you have to check that first.

I think you are confused. The cycle time (repetition period) you have set is only 8 micro seconds. So every 8 micro seconds a new pulse is generated. And if the pulse length is longer than 8us then you would get a constant on signal.

madhu-dhanasekaran, 2025-02-27:
Hi @ArthurKetels 

[image]

I could connect Sync0 pin of my first slave to the oscilloscope and could see trigger pulses generated as shown. I have increased the cycle time to 1ms now. I hope the new pulse is generated for every 1 ms. By pulse length you mean the duty cycle of the signal can be varied in the eeprom setting?

ArthurKetels, 2025-02-27:
`By pulse length you mean the duty cycle of the signal can be varied in the eeprom setting?`
Correct

madhu-dhanasekaran, 2025-02-27:
Ok. thanks.

[Capture_2slaves.zip](https://github.com/user-attachments/files/19011319/Capture_2slaves.zip)

Also, I have one more doubt. Now, I want to implement the same feature to my second slave as well. So, is it enough if I call ec_dcsync0 with slave 2 like **ec_dcsync0(2, TRUE, SYNC0TIME, 0);** immediately after calling sync on slave 1? However, I tried this approach and could not see signal on the second slave. I have attached the Captures for your reference.

I see some strange behavior though:
1. FPWR on Cyclic Unit Control (0x980) reg do not return WC 1
2. FPWR on DC StartTime (0x990) reg also do not return WC 1
Rest of the frames are paired with WC 1. Could this be the problem?

Also, may I know why are we seeing 2 frames for each FPWR/FPRD with Wc once with 0 and again with 1? 

Thank you again for your support.