OpenEtherCATsociety/SOEM #889: Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog

jukeboge, 2025-02-24:
Hello, based on the SOEM/test/linux/red_test/red_test.c test, the slave device is always in the "safeop+err" state. I have also referred to other issues and references.

[eRob_SOEM_linux/demo/eRob_CSP.cpp at main · ZeroErrControl/eRob_SOEM_linux](https://github.com/ZeroErrControl/eRob_SOEM_linux/blob/main/demo/eRob_CSP.cpp) 

[Fixed PDO mapped io module SyncManager watchdog error when moving to OP state · Issue #472 · OpenEtherCATsociety/SOEM](https://github.com/OpenEtherCATsociety/SOEM/issues/472)

[Cannot reach OPERATIONAL with SAFE OP + ERROR · Issue #456 · OpenEtherCATsociety/SOEM](https://github.com/OpenEtherCATsociety/SOEM/issues/456)

[LAN9252 slave cannot go OP under DC sync · Issue #824 · OpenEtherCATsociety/SOEM](https://github.com/OpenEtherCATsociety/SOEM/issues/824)

My code：

```c
void redtest(char *ifname, char *ifname2)
 {
    int cnt, i, j, oloop, iloop,chk;
 
    printf("Starting Redundant test\n");
 
    /* initialise SOEM, bind socket to ifname */
    (void) ifname2;
    //   if (ec_init_redundant(ifname, ifname2))
    if (ec_init(ifname))
    {
       printf("ec_init on %s succeeded.\n", ifname);
       /* find and auto-config slaves */
       if (ec_config_init(FALSE) > 0) // int->preop
       {
          printf("%d slaves found and configured.\n", ec_slavecount);
          // Check if the slave is ready to map
          ec_readstate(); // Read the state of the slaves

          for (int i = 1; i <= ec_slavecount; i++)
          { // Loop through each slave
             if (ec_slave[i].state != EC_STATE_PRE_OP)
             { // If the slave is not in PRE-OP state
                // Print the current state and status code of the slave
                printf("Slave %d State=0x%2.2x StatusCode=0x%4.4x : %s\n",
                       i, ec_slave[i].state, ec_slave[i].ALstatuscode, ec_ALstatuscode2string(ec_slave[i].ALstatuscode));
                printf("\nRequest init state for slave %d\n", i); // Request to change the state to INIT
                ec_slave[i].state = EC_STATE_INIT;                // Set the slave state to INIT
                printf("___________________________________________\n");
             }
             else
             {                                       // If the slave is in PRE-OP state
                ec_slave[0].state = EC_STATE_PRE_OP; // Set the first slave to PRE-OP state
                /* Request EC_STATE_PRE_OP state for all slaves */
                ec_writestate(0); // Write the state change to the slave
                /* Wait for all slaves to reach the PRE-OP state */
                if ((ec_statecheck(0, EC_STATE_PRE_OP, 3 * EC_TIMEOUTSTATE)) == EC_STATE_PRE_OP)
                {
                   printf("State changed to EC_STATE_PRE_OP: %d \n", EC_STATE_PRE_OP);
                   printf("___________________________________________\n");
                }
                else
                {
                   printf("State EC_STATE_PRE_OP cannot be changed in step 2\n");
                  //  return -1; // Return error if state change fails
                }
             }
          }

          ecx_context.manualstatechange = 1; //Disable automatic state change
          osal_usleep(1e6); //Sleep for 1 second
          for (int i = 1; i <= ec_slavecount; i++) {
            // (void)ecx_FPWR(ecx_context.port, i, ECT_REG_DCSYNCACT, sizeof(WA), &WA, 5 * EC_TIMEOUTRET);
             printf("Name: %s\n", ec_slave[i].name); //Print the name of the slave
             printf("Slave %d: Type %d, Address 0x%02x, State Machine actual %d, required %d\n", 
                    i, ec_slave[i].eep_id, ec_slave[i].configadr, ec_slave[i].state, EC_STATE_INIT);
             printf("___________________________________________\n");
[... 291 lines cut ...]
     addr b   index: sub bitl data_type    name
  [0x0010.0] 0x6040:0x00 0x10 UNSIGNED16   control word
  [0x0012.0] 0x607A:0x00 0x20 INTEGER32    target position
  [0x0016.0] 0x60FF:0x00 0x20 INTEGER32    target velocity
  [0x001A.0] 0x6071:0x00 0x10 INTEGER16    Target Torque
  [0x001C.0] 0x6072:0x00 0x10 UNSIGNED16   Max Torque
  [0x001E.0] 0x6060:0x00 0x08 INTEGER8     modes of operation
  [0x001F.0] 0x5FFE:0x00 0x08
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6041:0x00 0x10 UNSIGNED16   status word
  [0x0002.0] 0x6064:0x00 0x20 INTEGER32    position actual value
  [0x0006.0] 0x606C:0x00 0x20 INTEGER32    velocity actual value
  [0x000A.0] 0x6077:0x00 0x10 INTEGER16    torque actual value
  [0x000C.0] 0x603F:0x00 0x10 UNSIGNED16   error code
  [0x000E.0] 0x6061:0x00 0x08 INTEGER8     modes of operation display
  [0x000F.0] 0x5FFE:0x00 0x08
End slaveinfo, close socket
End program

```

wireshark file：

ESI/XML file：

jukeboge, 2025-02-24:
ESI/XML file：

[MT-Device.xml.txt](https://github.com/user-attachments/files/18937861/MT-Device.xml.txt)

jukeboge, 2025-02-24:
I have fix it,refer to https://github.com/OpenEtherCATsociety/SOEM/issues/224 patch of ethercatcoe.c

closed 2025-02-24 by jukeboge

HJ0804, 2025-07-01:
I'm also using the same MT_device, and I'm experiencing the same issue with the watchdog being triggered.
I'm currently using it with the following configuration:
ec_dcsyn0(slave, 1, 1000000, 0);

If I use the patched version of ethercatcoe.c, will that prevent the watchdog from being triggered?
I’d appreciate your advice on where exactly I should make the modification.