OpenEtherCATsociety/SOEM #890: Issue getting slaves (StaubliTX260L UniVAL drive) into OP state

tommachiels, 2025-02-26:
Hello

We are currently trying to interface with our robot (Staubli TX2 60L running UniVAL drive) via CoE SOEM. However, we are new to the world of EtherCAT and SOEM and are experiencing a lot of issues because of it. 
First, we configured the robot in TwinCAT as a verification, and are able to get it into OP state as expected. However in SOEM, we only manage to reach SAFE_OP. When trying to set it to OP:

```
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error
````

The wireshark captures of both TwinCAT and SOEM are given in the accompanied zip file.
We also made sure our IOmap rxpdo and txpdo structs match the data given by slave_info -map (also provided in the zip together with the -sdo output).

Could anyone point us in the right direction of what we might be missing here?

Thank you in advance

[data.zip](https://github.com/user-attachments/files/18988286/data.zip)

ArthurKetels, 2025-02-27:
Do you have the ESI file of the robot? There you could find the supported operational modes. It is very well possible you need sync0 activated for synchronous operation. It looks like the slave is simply waiting for it before going to OP mode.

tommachiels, 2025-02-27:
Thanks for your assistance! I think that’s very likely our issue. I’ve attached the ESI file of our robot.

[unival_coe_cs9 - Copy.txt](https://github.com/user-attachments/files/19012463/unival_coe_cs9.-.Copy.txt)

ArthurKetels, 2025-02-28:
Yes, confirmed. Only DC synchronous is supported. Also you need to manually set some SDO objects before pre-OP to safe-OP transition.

```
				<Mailbox DataLinkLayer="1">
					<EoE IP="1" MAC="1"></EoE>
					<CoE SdoInfo="1" CompleteAccess="0" PdoAssign="0" PdoConfig="1" PdoUpload="0" DiagHistory="1">
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x60da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 0 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x68da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 1 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x70da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 2 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x78da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 3 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x80da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 4 Synch Function Settings</Comment>
						</InitCmd>
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x88da</Index>
							<SubIndex>0</SubIndex>
							<Data>05000000</Data>
							<Comment>Ch 5 Synch Function Settings</Comment>
						</InitCmd>
					</CoE>
				</Mailbox>
				<Dc UnknownFRMW="0" Unknown64Bit="0">
					<OpMode>
						<Name>DC-synchronized</Name>
						<Desc>DC for synchronization</Desc>
						<AssignActivate>#x0300</AssignActivate>
						<CycleTimeSync0 Factor="1">0</CycleTimeSync0>
						<ShiftTimeSync0>0</ShiftTimeSync0>
					</OpMode>
				</Dc>
```

tommachiels, 2025-03-20:
Hi Arthur

Thank you very much for you help. It turns out that we were actually very close to the solution. 
It turns out we were setting up the DC sync incorrectly. 
After some further searching, we found the solution in commit abc0b63. We were missing the PO2SOconfig hook for the pre-OP to safe-OP transition.

Now we can get the robot into OP mode, and have verified by enabling, disabling its ValveIO via SOEM.

closed 2025-03-20 by tommachiels