OpenEtherCATsociety/SOEM #902: Reset PHY via SOEM

IceTea2, 2025-03-30:
Hello!

At first i want to say, SOEM works great! Its an amazing tool! It really is and i have no problems with it. I have tested it on many different slaves now, from many different Beckhoff Terminals to many different CiA402 Drives.

But there is one problem is do have and i wanted to see if someone of you guys could help me with that.

We have 9 AZD KED Drivers from Oriental Motor (Thos are really well implemented ethercat slaves). These work well with SOEM and i have no issues with them other than that sometimes i do get package loss. (about 1-10 packages per minute) The networks runs at 1 khz and the real time performance is "good" on linux with preempt patch and further actions to get it to have better RT performance. The really weird thing is, the package loss comes out of nowhere... Sometimes the network runs flawless with not a single lost package in even 2 hours, while doing heavy lifting on other CPU cores... EMI should not be a problem here, as only the control power supply was used in a mostly static eviorment with shielding and whatelse...

The loss i get is always on register 0x300 (RX-Error) and mostly only on one slave of those drivers (but often a different one)... What i dont get is, when i completely unplug both RJ45 ports of this driver, where the loss is happening, its gone when i reboot the network...

And this gets me to my question:
Do you guys know if its possible to reset the PHY of the ESC via SOEM? The chip is a "R7S910115" or an Renesas R-IN... i thought i'd try FPWR on registers 0x0510 to 0x0514... but no success here... the PHY doesnt reset... and each FPWR returns 0.

If needed i could post a wireshark file here, although i guess it wont be very helpful when the package gets lost by a slave deeper in the network.
Thanks in advance!

ArthurKetels, 2025-04-01:
It is possible to control a slave PHY via ECAT. However the protocol is very finicky.  I have done it once with an ET1100 ESC. No idea if it works with Renesas. 

On the other hand, this is the wrong way of handling the problem. The slave PHY's should be robust. And mostly an RX-error in the ESC is a result of improper slave HW design. Trying to fix this with a PHY hack through SOEM is not going to give you good results. 

Either you can cope with the lost packets or not. And if not you have to improve the hardware. And this is an Oriental motor issue.

If have experienced these situations multiple times (both with own slave designs and customer designs). Sometimes it can indeed be the RJ45 cables. Sometimes it is the RJ45 socket on the slave. Sometimes it is ground loops. Sometimes it is slave PCB design. One time it was even an out of spec oscillator. Oscilloscopes and spectrum analysers are your tools to go here.

For ground loop issues on the ECAT cables it is easy to check with non-shielded cables. If they work and shielded do not, then you know what is the problem.

IceTea2, 2025-04-01:
Hello Arthur!

Thanks for you advice. I gotta admit our Schneider Electric slave, which is in the same network, is never losing packages... i'll definitely try your advice with the unshielded cables... (and our ethernet cable pile grows once more, nice!)

Hacking the phy seems somewhat doable though, as the registerset of the renesas seems to be compatible to the ET1100 set.. But yes, i will contact oriental motor about this for sure. 

One further question though:
You mention that this seems (the package loss) to be a problem with some slaves... May i ask how common is this?
Is it a viable solution to just live with the package loss? Case depended i know, but how common is this?

IceTea2, 2025-04-05:
Hey Arthur,

Thanks again for your reply and your time!

I’ve now also tried using unshielded cables, but unfortunately the problem is still present. It’s getting quite frustrating…

As I mentioned before, this time slave 6 showed all the RX errors. I unplugged both RJ45 cables, plugged them back in, and the errors were gone.
Next time, slave 5 started showing the same RX errors. Same procedure – unplug, replug – and the issue disappeared again.

The network has now been running for about 30 minutes without errors, while normally they appear within the first 1–2 minutes.

What’s really strange is that when I power off all slaves and restart the network, the errors come back immediately – although not always on the same slave. So it doesn’t seem to be tied to a specific device.

I’m still waiting for a response from Oriental Motor, but to be honest, I don’t have too much hope at this point…
It’s a bit painful, especially considering these drivers weren’t exactly cheap to begin with.

closed 2025-04-05 by IceTea2