OpenEtherCATsociety/SOEM #906: Different WKC in slaveinfo and ec_receive_processdata()

NevolinAlexander, 2025-04-13:
I got another "pretty" Chinese servodrive (Innovance). As usual for China, it does not work "out of the box" - gets stuck in safe_op state, not going to operational. All previous hacks for other chinese drives does not help.

But I noticed strange thing: when I check this drive with slaveinfo, it says that WKC is 3 (that is correct). But when I make few cycles for syncing before going to operational state - function ec_receive_processdata returns 2. I don't understand how it can be. Maybe this is the way to understand what this drive needs.

nakarlsson, 2025-04-13:
It has not written any data to the SM3 yet, hence no inputs can be successfully read.

Educated guess

NevolinAlexander, 2025-04-30:
I found only one parameter to be written in ESI file:

				<Mailbox DataLinkLayer="1">
					<EoE IP="false" MAC="true" TimeStamp="false"/>
					<CoE SdoInfo="0" PdoAssign="1" PdoConfig="1" CompleteAccess="0">
						<InitCmd>
							<Transition>PS</Transition>
							<Index>#x6060</Index>
							<SubIndex>0</SubIndex>
							<Data>08</Data>
							<Comment>Modes of operation</Comment>
						</InitCmd>
					</CoE>
				</Mailbox>

Is it what I need?

I wrote it between pre_op and safe_op, while making PDO assign:
TRY(SDOwrite8(s, 0x6060, 0, 8, EC_TIMEOUTTXM), SDOMaxErrors, return FALSE;);

But actually it does not help, problem remains.

NevolinAlexander, 2025-04-30:
[SV660_1Axis_v9.15.xml.zip](https://github.com/user-attachments/files/19985557/SV660_1Axis_v9.15.xml.zip)

UPDATE: by disabling transition to safe_op during config_map I have solved problem with wkc, 

After going to safe_op I make 1000 cycles in synchronous mode, first 200-300 of them go with wkc=2, and than it changes to 3. 

But servo still does not allow to go to operational mode (does not switch in it without any errors).

In ESI file I also noticed this:

<Dc>
					<OpMode>
						<Name>DC</Name>
						<Desc>DC-Synchron</Desc>
						<AssignActivate>#x300</AssignActivate>
						<CycleTimeSync0 Factor="1">0</CycleTimeSync0>
						<ShiftTimeSync0>0</ShiftTimeSync0>
						<CycleTimeSync1 Factor="1">0</CycleTimeSync1>
					</OpMode>
				</Dc>

Do I need to write something to drive to activate DC?

NevolinAlexander, 2025-05-10:
I found solution. Actually Inovance drives require a bit more synchronous cycles before going to operational state. I used 25 cycles and it was not enough. Everything went fine when I increased cycles count.

Same thing was for going to "servo on" state.

closed 2025-05-10 by NevolinAlexander