OpenEtherCATsociety/SOEM #910: SOEM controls the EL6601, but the IP setting fails and the ping is unreachable. Does this indicate a flaw in the EOE functionality?

wmyqcy, 2025-04-16:
Hello, I found that setting the IP address failed and returned -5 while testing the EOE module. The module is EK1100 connected to EL6601. Is there a design flaw in the EOE functionality itself?

I connected the EL6601's Ethernet port to my laptop, but the ping is unreachable. Could it be due to a failed IP setting?

By capturing packets with Wireshark (laptop connected to EL6601), I can capture the Ethernet packets from the EL6601's Ethernet port.

[image]

wmyqcy, 2025-04-16:
Here are the packets I captured with Wireshark (switch connecting master station, laptop, and slave):

https://github.com/wmyqcy/test/blob/main/EOE.rar

nakarlsson, 2025-04-16:
The EL6601 act as an switch, the IP you set is for a device connected to the EL6601.
If I remember correct the EL6601 require a complete setr of IP data as DNS...
And I'm not sure it support GetIP either.

The best way to test an EL6601 is to connect a device with a static IP macthing your Laptop subnet.
You should see ARPs coming in an so on,

nakarlsson, 2025-04-16:
```
    eoe_param_t ipsettings;
    memset(&ipsettings, 0, sizeof(ipsettings));
 
    ipsettings.mac_set = 1;
    ipsettings.ip_set = 1;
    ipsettings.subnet_set = 1;
    ipsettings.default_gateway_set = 1;
    ipsettings.dns_ip_set = 1;
    ipsettings.dns_name_set = 1;

    ipsettings.mac.addr[0] = 0x0d;
    ipsettings.mac.addr[1] = 0xfa;
    ipsettings.mac.addr[2] = 0xf5;
    ipsettings.mac.addr[3] = 0xe4;
    ipsettings.mac.addr[4] = 0xb7;
    ipsettings.mac.addr[5] = 0x02;
    memset(ipsettings.dns_name, 0, sizeof(ipsettings.dns_name));
    EOE_IP4_ADDR_TO_U32(&ipsettings.ip, 192, 168, 127, 200);
    EOE_IP4_ADDR_TO_U32(&ipsettings.subnet, 255, 255, 255, 0);
    EOE_IP4_ADDR_TO_U32(&ipsettings.default_gateway, 192, 168, 127, 150);
    EOE_IP4_ADDR_TO_U32(&ipsettings.dns_ip, 0, 0, 0, 0);

    /* Send a set IP request */
    ecx_EOEsetIp(context, EOE_SLAVE, 0, &ipsettings, EC_TIMEOUTRXM);
```

wmyqcy, 2025-04-17:
@nakarlsson  
I tried, an industrial computer running the main station, two laptops connected to the EL6601 via the switch, and no new TAP adapter appeared.
Could you tell me more about your test method?

wmyqcy, 2025-04-17:
> ```
>     eoe_param_t ipsettings;
>     memset(&ipsettings, 0, sizeof(ipsettings));
>
>     ipsettings.mac_set = 1;
>     ipsettings.ip_set = 1;
>     ipsettings.subnet_set = 1; [...]

The IP address also fails to be set.  It is not clear why ecx_mbxreceive (SMstat & 0x8) is less than or equal to 0.  The return value is -5.

nakarlsson, 2025-04-17:
I used: 

Laptop with installed <TAP-Windows> interface 192,168,127,200
FileZilla FTP client for test
SOEM  Application using TAP ReadFile ->ecx_EOEsend(EL6601) / eoe_hook(EL6601) -> TAP WriteFile
< ECAT >  
EL6601 
< TCP/IP> 
Laptop FileZilla FTP server 192.168.127.1

I read/write files over FTP with speed ~300-500 KB/s
I used PING to test ping of large pakets > 1500

wmyqcy, 2025-04-18:
I tested EOE on Beckhoff TwinCAT, and the network is functioning normally — it can be pinged.
What you mentioned requires the network to be reachable via ping before proceeding. Right now, I still can't ping the network.

nakarlsson, 2025-04-18:
As a start make sure you receive drakes from the network connected to the EL6601 via the eoe_hook. 
Should see ARPs and simulator frames.

wmyqcy, 2025-04-18:
Do I need to install TAP to run my SOEM on ubuntu18?

nakarlsson, 2025-04-18:
Make use of Linux tun/tap interface.

Read from interface and whiteboarden to EL6601 ecx_EOESend.

Read from eoe_hook and write to interface egen frame is complete.

wmyqcy, 2025-04-21:
@nakarlsson  
I added a TAP interface in my program using "/dev/net/tun" and assigned it an IP address, but I still can't ping it. Could you please share your test case or example?

nakarlsson, 2025-04-21:
You can share your code?

I”d expect to see code that read/write the TAP interface using the EoE API and hook.