OpenEtherCATsociety/SOEM #916: While testing the redundancy mode, I found that unplugging and reconnecting the main network port cable results in the loss of one packet.

wmyqcy, 2025-05-27:
Hello, while testing the redundancy mode, I found that unplugging and reconnecting the network cable results in the loss of one packet.

Is this behavior normal?

The same phenomenon occurs with both 1ms and 4ms cycles — unplugging and reconnecting the main network port cable causes the loss of one packet.
Could this be a bug in the redundancy mode handling?

When the main network port's cable is disconnected, everything functions normally. However, once the main port's cable is reconnected, an issue arises: the idx parameter in the ecx_inframe function is always 1 greater than the idxf value inside that same function.

I also store the idx that meets the condition (valid_wkc == 0) in the ecx_receive_processdata_group function. The idxf value in ecx_inframe is the same as the idx in ecx_receive_processdata_group.

I stored ecx_inframe  in_idx and idxf at this location. “OSEE_PRINT("ecx_inframe(): WARNING: strange things happened\n");
” 
if (CNT_idx_idxf < 10)
{
    buf_in_idx[CNT_idx_idxf] = idx;
    buf_idxf[CNT_idx_idxf] = idxf;
    CNT_idx_idxf++;
    printf("ERR stacknumber=%d \n", stacknumber);
}

Below is the log/print information:
ecx_inframe：
cnt_0 in_idx = 9 idxf=8 
cnt_1 in_idx = 2 idxf=1 
cnt_2 in_idx = 0 idxf=15 
cnt_3 in_idx = 14 idxf=13 
cnt_4 in_idx = 14 idxf=13 
cnt_5 in_idx = 14 idxf=13 
cnt_6 in_idx = 4 idxf=3 
cnt_7 in_idx = 15 idxf=14 
cnt_8 in_idx = 1 idxf=0 
cnt_9 in_idx = 1 idxf=0 

ecx_receive_processdata_group ：
cnt_0 idx = 8 
cnt_1 idx = 1 
cnt_2 idx = 15 
cnt_3 idx = 13 
cnt_4 idx = 13 
cnt_5 idx = 13 
cnt_6 idx = 3 
cnt_7 idx = 14 
cnt_8 idx = 0 
cnt_9 idx = 0

wmyqcy, 2025-05-27:
During redundancy mode testing, unplugging and reconnecting the secondary (slave) port cable behaves normally.
Only the main network port shows this behavior—unplugging and reconnecting its cable causes the loss of one packet.

banbingbing, 2025-07-03:
Hello, I also encountered the same problem when testing the redundancy function. When disconnecting the main network port, packet loss occurs. Could I ask you for some advice?