OpenEtherCATsociety/SOEM #918: Clarification on EC_STATE_SAFE_OP + EC_STATE_ACK behavior in ec_statecheck()

open04, 2025-06-27:
Hi, I'm encountering an error when calling ec_statecheck() with the flags EC_STATE_SAFE_OP + EC_STATE_ACK.  
Upon checking the current state after setting the NMT state, it only shows SAFE_OP, and I don't see the ACK state applied. 

Can you clarify why this is happening? Is there a specific reason for this behavior, or is this a potential bug in the state-checking logic? Any guidance would be appreciated!

ArthurKetels, 2025-06-27:
What you observed is normal and expected behaviour. The slave state machine is precisely described in the standard.
The EC_STATE_ACK is send to the slave to acknowledge an error state. If accepted by the slave it returns the state without the EC_STATE_ERR flag. It just so happens that the bit that is defined for the EC_STATE_ACK in the command state register (0x120) is the same as EC_STATE_ERR in the status register (0x130).

So ec_statecheck() reads the EC_STATE_ERR bit and not the EC_STATE_ACK bit.

open04, 2025-06-30:
Thank you so much for clarifying, Im happy with the answer.

closed 2025-06-30 by open04