OpenEtherCATsociety/SOEM #922: Leadshine R3 commissioning issue

thetooth, 2025-07-17:
Hi, I am trying to get a Leadshine R3 bus coupler working under SOEM but have run into a wall getting PDO mapping to happen. The devices use CoE mailboxes much like the EK1100 where you write SDO parameters during pre_op to tell the coupler what the layout of the physical modules are.

The stride for the module placement is 0x10 starting from 0 and 0x8 stride of PDOs for fixed mapping modules, there of course being special modules like the analog input requiring normal dynamic mapping.

Here is the output slaveinfo after a cold boot:
```
Slave:2
 Name:EM06AX-E4
 Output size: 40bits
 Input size: 56bits
 State: 4
 Delay: 780[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: 00004321 ID: 61400025 Rev: 24060501
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   5 F:00010064 Type:3
 SM3 A:1c00 L:   7 F:00010020 Type:4
 FMMU0 Ls:00000006 Ll:   5 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000016 Ll:   7 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
 CoE Object Description found, 13 entries.
0x1000      "Device type"                                 [VAR]
    0x00      "Device type"                                 [UNSIGNED32       R_R_R_]      0x00001389 / 5001
0x1008      "Device name"                                 [VAR]
    0x00      "Device name"                                 [VISIBLE_STR(160) R_R_R_]      "R3EC"
0x1009      "Hardware version"                            [VAR]
    0x00      "Hardware version"                            [UNSIGNED32       R_R_R_]      0x24050701 / 604309249
0x100a      "Software version"                            [RECORD  maxsub(0x00 / 0)]
    0x00      "Software version"                            [UNSIGNED32       R_R_R_]      Time:1752711062.071 PACKET slave:2 index:100a.00 error:3

 -
0x1018      "Number of entries"                           [RECORD  maxsub(0x04 / 4)]
    0x00      "Number of entries"                           [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "Vendor ID"                                   [UNSIGNED32       R_R_R_]      0x00004321 / 17185
    0x02      "Product code"                                [UNSIGNED32       R_R_R_]      0x61400025 / 1631584293
    0x03      "Revision"                                    [UNSIGNED32       R_R_R_]      0x24060501 / 604374273
    0x04      "Serial number"                               [UNSIGNED32       R_R_R_]      0x00000001 / 1
0x1c00      "Sync manager type"                           [ARRAY  maxsub(0x04 / 4)]
    0x00      "Number of entries"                           [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "MailboxOut"                                  [UNSIGNED8        R_R_R_]      0x01 / 1
    0x02      "MailboxIn"                                   [UNSIGNED8        R_R_R_]      0x02 / 2
    0x03      "Outputs"                                     [UNSIGNED8        R_R_R_]      0x03 / 3
    0x04      "Inputs"                                      [UNSIGNED8        R_R_R_]      0x04 / 4
0x1c12      "SM2(Outputs) PDO Assignment"                 [ARRAY  maxsub(0x64 / 100)]
    0x00      "SubIndex 0"                                  [UNSIGNED8        RWR_R_]      0x00 / 0
0x1c13      "SM3(Inputs) PDO Assignment"                  [ARRAY  maxsub(0x64 / 100)]
    0x00      "SubIndex 0"                                  [UNSIGNED8        RWR_R_]      0x00 / 0
0xf000      "module device profile"                       [RECORD  maxsub(0x02 / 2)]
    0x00      "Number of entries"                           [UNSIGNED8        R_R_R_]      0x02 / 2
    0x01      "SubIndex 1"                                  [UNSIGNED16       R_R_R_]      0x0010 / 16
    0x02      "SubIndex 2"                                  [UNSIGNED16       R_R_R_]      0x0040 / 64
0xf030      "module ident list"                           [ARRAY  maxsub(0x40 / 64)]
    0x00      "SubIndex 0"                                  [UNSIGNED8        RWR_R_]      0x00 / 0
0xf050      "module ident detected"                       [ARRAY  maxsub(0x40 / 64)]
    0x00      "SubIndex 0"                                  [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "SubIndex 1"                                  [UNSIGNED32       R_R_R_]      0x61100026 / 1628438566
    0x02      "SubIndex 2"                                  [UNSIGNED32       R_R_R_]      0x61000026 / 1627389990
    0x03      "SubIndex 3"                                  [UNSIGNED32       R_R_R_]      0x61110206 / 1628504582
    0x04      "SubIndex 4"                                  [UNSIGNED32       R_R_R_]      0x61900106 / 1636827398
0x1bfa      "device status"                               [RECORD  maxsub(0x02 / 2)]
    0x00      "Number of entries"                           [UNSIGNED8        R_R_R_]      0x02 / 2
    0x01      "SubIndex 1"                                  [UNSIGNED32       R_R_R_]      0xf1000120 / 4043309344
[... 68 lines cut ...]
0xf030      "module ident list"                           [ARRAY  maxsub(0x40 / 64)]
    0x00      "SubIndex 0"                                  [UNSIGNED8        RWR_R_]      0x00 / 0
0xf050      "module ident detected"                       [ARRAY  maxsub(0x40 / 64)]
    0x00      "SubIndex 0"                                  [UNSIGNED8        R_R_R_]      0x04 / 4
    0x01      "SubIndex 1"                                  [UNSIGNED32       R_R_R_]      0x61100026 / 1628438566
    0x02      "SubIndex 2"                                  [UNSIGNED32       R_R_R_]      0x61000026 / 1627389990
    0x03      "SubIndex 3"                                  [UNSIGNED32       R_R_R_]      0x61110206 / 1628504582
    0x04      "SubIndex 4"                                  [UNSIGNED32       R_R_R_]      0x61900106 / 1636827398
0x1bfa      "device status"                               [RECORD  maxsub(0x02 / 2)]
    0x00      "Number of entries"                           [UNSIGNED8        R_R_R_]      0x02 / 2
    0x01      "SubIndex 1"                                  [UNSIGNED32       R_R_R_]      0xf1000120 / 4043309344
    0x02      "SubIndex 1"                                  [UNSIGNED32       R_R_R_]      0xf1000220 / 4043309600
0xf100      "device status"                               [RECORD  maxsub(0x02 / 2)]
    0x00      "Number of entries"                           [UNSIGNED8        R_R_R_]      0x02 / 2
    0x01      "SubIndex 1"                                  [UNSIGNED32       R_R_R_]      0x00000000 / 0
    0x02      "SubIndex 2"                                  [UNSIGNED32       R_R_R_]      0x00000000 / 0
End slaveinfo, close socket
End program
```

What's strange is that the output above is identical to when I use the device under codesys, Obits is zero yet codesys is able to write to the outputs PDO. In both cases SOEM/codesys I have full access to the two input modules. Only SOEM cannot do outputs with the same configuration.

Another detail that could help is if I don't perform any dynamic mapping, that is ONLY writing 0x1a00 to SM2 then the outputs DO get mapped, which makes absolutely no sense lol

[c2.pcapng.gz](https://github.com/user-attachments/files/21269042/c2.pcapng.gz)

nakarlsson, 2025-07-17:
SOEM does not program the FMMU for Obits Zero. Hence no outputs will be written

I”m not in a position were I can check the Wireshark. 
So, try using Complete Access see if it makes a difference.
Or
If using singel index access, make sure to trigger PDO reconfig by start with setting PDO objects Subindex 0 = 0, program PDO, finalize by setting correct Subindex 0 = n.

thetooth, 2025-07-18:
Complete access is denied outright by the device. And yeah the SDO writes are structured such that subindex 0 is zeroed then N subindexes followed by the count. I have even tried inserting a delay between programming the mapping indexes and the sync managers to give the device a moment to apply everything, no difference.

My suspicion is that other EtherCAT masters are ignoring the Obit value and just send data anyway. This is the only device I've seen with this behaviour so it's def not standard, I'm using SOEM with servos, IO and even some serial gateway stuff and it's always worked exactly as expected given the right startup config.

Maybe you could point me in the right direction for bypassing the the Obit check/forcing the FMMU programming just to rule out a commissioning issue since it will surely blow up if the config is incorrect? I'll email the manufacture and see if I can get anything out of them, they don't have firmware on their website but who knows maybe I'm running an old version that's cooked.

thetooth, 2025-07-20:
```
[  info  ] : Runtime initialized
[  info  ] : 2 slaves found
[ debug  ] : PRE_OP Slave 1: EasyIO found, vendor: 0x79a, id: 0xdebacca, rev: 0x2a
[ debug  ] : PRE_OP Slave 2: EM06AX-E4 found, vendor: 0x4321, id: 0x61400025, rev: 0x24060501
SDO write 0x8000, 3, 0xA
SDO write 0x8000, 4, 0xA
SDO write 0x8020, 1, 0x0
SDO write 0x8030, 1, 0x0
SDO write 0x8010, 1, 0x2
SDO write 0x8010, 2, 0x2
SDO write 0x8010, 3, 0x2
SDO write 0x8010, 4, 0x2
SDO write 0x8011, 1, 0x4
SDO write 0x8011, 2, 0x4
SDO write 0x8011, 3, 0x4
SDO write 0x8011, 4, 0x4
SDO write 0x8013, 1, 0x1
SDO write 0x8013, 2, 0x1
SDO write 0x8013, 3, 0x1
SDO write 0x8013, 4, 0x1
SDO write 0xF030, 0, 0x0
SDO write 0xF030, 1, 0x61100026
SDO write 0xF030, 2, 0x61000026
SDO write 0xF030, 3, 0x61110206
SDO write 0xF030, 4, 0x61900106
SDO write 0xF030, 0, 0x4
SDO write 0x1C13, 0, 0
SDO write 0x1C13, 1, 0x1BFA
SDO write 0x1C13, 2, 0x1A00
SDO write 0x1A08, 0, 0
SDO write 0x1A08, 1, 0x60100110
SDO write 0x1A08, 2, 0x60100210
SDO write 0x1A08, 3, 0x60100310
SDO write 0x1A08, 4, 0x60100410
SDO write 0x1A08, 0, 4
SDO write 0x1C13, 3, 0x1A08
SDO write 0x1A09, 0, 0
SDO write 0x1A09, 1, 0xA0100108
SDO write 0x1A09, 2, 0xA0100208
SDO write 0x1A09, 3, 0xA0100308
SDO write 0x1A09, 4, 0xA0100408
SDO write 0x1A09, 0, 4
SDO write 0x1C13, 4, 0x1A09
SDO write 0x1C13, 0, 4
SDO write 0x1C12, 0, 0
SDO write 0x1C12, 1, 0x1610
SDO write 0x1C12, 2, 0x1618
SDO write 0x1C12, 0, 2
[  info  ] : Slaves mapped and DC synchronization started
[ debug  ] : Slave 2 added to cyclic mailbox handler
[  info  ] : All slaves in SAFE_OP state
[ debug  ] : DC capable: yes
[ debug  ] : SAFE_OP Slave 1: EasyIO found, vendor: 0x79a, id: 0xdebacca, rev: 0x2a
[ debug  ] : SAFE_OP Slave 2: EM06AX-E4 found, vendor: 0x4321, id: 0x61400025, rev: 0x24060501
[ debug  ] : Slave 1 'EasyIO': state=0x08, Obits=48, Ibits=88
[ debug  ] : AL status code: 0x0000 (No error)
[ debug  ] : SM2 start=0, length=0, SM3 start=0, length=0
[ debug  ] : Slave 2 'EM06AX-E4': state=0x08, Obits=0, Ibits=192
[ debug  ] : AL status code: 0x0000 (No error)
[ debug  ] : SM2 start=6144, length=5, SM3 start=7168, length=24
[ debug  ] : Expected WKC 4, supervisor starting...
[warning ] : Stored Error List: Time:1752821193.508 SDO slave:2 index:1610.00 error:08000000 General error
[  info  ] : Subscribed to topic ctrl.start
[  info  ] : Subscribed to topic ctrl.stop
[  info  ] : Subscribed to topic ml.detection
[warning ] : Stored Error List: Time:1752821193.509 SDO slave:2 index:1618.00 error:08000000 General error
[  info  ] : Operational state reached for all slaves
[warning ] : Stored Error List: Time:1752821193.512 SDO slave:2 index:1a00.00 error:08000000 General error
```

More or less identical, those errors at the bottom come from SOEM reading back those addresses not the actual writes.

nakarlsson, 2025-07-20:
I would try to add x1610 and x1618 with the correct x70xx objects. That are essential for SOEM to be able to detect Obits.

Ex x1610:01-> x7020 01 01 for Bit0 Size 1 Bit
The ESI define that for 0x1600

As Done for 0x1A08